kriswiner/MPU9250 #391: pitch and roll locked

Papierkor654, 2019-10-02:
I've got a question:
When I rotate the IMU around the z axis, the x and y axis stay in place and have always the same orientation relative to the outside world. Is there some ways to convert this in such a way, that pitch and roll always stay in the same place relative to the IMU?
I've done it using a rotation matrix, but this is only somewhat accurate if the pitch and roll angle is below 45°. I don't really know which words to use to google for this problem.

I'm using the library from sparkfun, but since their code is based on this library and since the community is more active here, I'm asking the question here.

kriswiner, 2019-10-02:
Confused.

If you rotate about the z-axis, isn't it the z-axis that is stationary? How
can the x- and y-axes "stay in place"?

Papierkor654, 2019-10-02:
Exactly that is my problem, if the x axis is parallel to a line drawn from the north to the south pole and then the IMU is rotated around the z axis, the x axis was still parallel to the line mentioned before. So the x and y axis don't rotate with the z axis.

kriswiner, 2019-10-02:
Sounds like either:

0) you are not configuring your sensor properly (are you seeing reasonable
accel, gyro, and mag values? How do you know?)

1) your sensor is broken (do all sensor axes respond reasonably?)

2) your sensor is not calibrated (How did you calibrate it?)

3) You are not loading the sensor dat into the sensor fusion filter properly

4) all of the above.

3)

Papierkor654, 2019-10-02:
The angles are correct as far as I can see. With the transformation mentioned above I've even managed to stabilize a drone. But because this transformation causes the angles to be wrong beyond 45° the drone crashes if it's to much angled.

How would I calibrate the IMU? I didn't do this. I thought only the magnetometer has to be calibrated due to the magnetic declination.

kriswiner, 2019-10-02:
All the sensors need to be calibrated. Otherwise they will not be accurate.

Papierkor654, 2019-10-02:
Ok, thanks for your help ^^
I probably should to this to make sure to have to highest accuracy possible. 
But in my case everything is accurate at the very least to +-5° except for the fact that the x and y axis stay still. I think this can't really be an accuracy issue.

kriswiner, 2019-10-02:
Likely a software error, I thought you said your UAV flies...it shouldn't.

Are the sensor values reasonable? if you change orientation, do you see
gravity on all thee accel axes, for example, etc?