kriswiner/MPU9250 #392: Calibration before final mount

Rsambeek, 2019-10-10:
Hi,

I am trying to use the mpu9250 to get orientation and compass data. Its all good in my test setup but eventually I have to mount this in a plane and I don't think it will be easy to rotate the plane to calibrate the compass. Is there another way I can calibrate it beforehand and apply some sort of offset to get the somewhat correct calibration. 

Thanks for all the work you have already put in this it has helped me a lot with this project.

kriswiner, 2019-10-10:
You can calibrate the sensors before hand.

This should work pretty well.

But yes, the plane environment will affect the offset bias and different
locations will cause the magnetic declination to change. The latter can be
adjusted with a lookup table and GPS. The former requires some kind of
magnetic anomaly detection/correction. The algorithm should detect offset
(without rotating the plane) and correct them automatically. We have such
an algorithm but it is not open source.

Rsambeek, 2019-10-11:
Thanks for the response. I was thinking of such an algorithm and was hoping for an easier solution but it seems like there isn't an easy fix. I guess I should start programming that then.

On that second part tho "We have such an algorithm but it is not open source." does that imply that you have a closed source one someone can buy or something. Just figuring out my possibilities.

Rsambeek, 2019-10-25:
I'm still curious to this algorithm you were talking about

kriswiner, 2019-10-25:
Hi Rick,

I sent a message to Greg who has this, maybe send me an e-mail at
tleracorp@gmail.com in case github doesn't do forwarding correctly...

Kris

gregtomasch, 2019-10-26:
Hi Rick,

It would be best to e-mail me at poirot@earthlink.net and we can discuss..

Best,
Greg