kriswiner/MPU9250 #393: Could not connect to MPU9250: 0xFF

bradBot5, 2019-10-14:
Thanks for the code.
I am using a sparkfun pro mini 3.3v with the MPU 9250 (the one labeled MPU 92/65 on top)
I cannot connect. Could it be related to the resistors you refer to in the code comments? If so what value for the resistors would you reccomend? Is there a schematic that show the correct connections?

MPU9250
9-DOF 16-bit
motion sensor
60 ug LSB
MPU9250 I AM FF I should be 71
MPU9250
I AM
FF
I Should Be
71
Could not connect to MPU9250: 0xFF

Thanks,
Brad

bradBot5, 2019-10-14:
also there are no pins on mine labeled vdd or vddi. I took them to mean I should connect vcc to vcc. I am I missing something? Did they change the names of the pins between versions?

kriswiner, 2019-10-14:
When you run an I2C scan on the board, what addresses show up?

bradBot5, 2019-10-15:
Thanks for the swift response!

all an I2C scan gets back is:
I⸮
I⸮
I⸮

I suppose this means either something is not wired right or my IMU is junk.

If I disconnet the IMU I get:

I2C Scanner
Scanning...
No I2C devices found

...which seems to rule out an MCU issue

bradBot5, 2019-10-15:
had a stupid wiring mistake, ground off by a pin on breadboard.

now I'm getting:

09:31:30.929 -> 
09:31:30.929 -> I2C Address Scanner. Scanning ...
09:31:30.929 -> Found i2c Device Address: 104 (0x68)
09:31:30.929 -> Done.
09:31:30.929 -> Found 1 device(s).

bradBot5, 2019-10-15:
IMU online!
Thank you Mr Winer!