kriswiner/MPU9250 #395: strange yaw output from AHRS

bradBot5, 2019-10-21:
Mr. Kriswiner,
Hello again. I already owe you a few beers, but i was hoping you could help with one last issue. My AHRS output is working great on pitch and roll, however the yaw output is quite weird. I suspect there is something more than just magnetic interference at work here: 

https://vimeo.com/367785242

Have I missed something in the setup or calibration of the magnetometer that would give results such as those visible in the linked video?

Thanks again for the awesome work on this and your time tracking issues.

kriswiner, 2019-10-21:
The sensors need to be calibrated properly. The sensor data need to be fed
into the fusion filter in the appropriate order.

bradBot5, 2019-10-23:
I haven't yet figured out the calibration process, or how to input the results, so I have put my mpu9250 on the shelf for now and got a BNO055 working instead, and ordered myself a SENtral. Have you tried the myAHRS+ yet? Thanks again.

closed 2019-10-23 by bradBot5