kriswiner/MPU9250 #396: porting to desktop for higher fusion rate and faster convergence

shomedas, 2019-10-21:
I had found that there is a delay(a few seconds) in the convergence of the heading quaternion when the IMU is moved randomly or fast. So I have ported the sensor fusion part of your code(the for loop having  the madgwick update) to my desktop to get higher sensor fusion rates. I grab the accelerometer, gyroscope and the magnetometer readings from the MPU9250 using an ESP32 using the remaining part of your code.  I calibrate the data in my desktop.

My question is **what parameters do I need to change to attain faster convergence** when I move around the IMU. I read from your git forum that accelerometer and gyroscope need to be sampled at 1Khz with 100 Hz Low Pass Filter, while magnetometer is to be left at 100Hz. I made the following changes in initMPU9250 function of your code

SMPLRT_DIV : changed it to 0x00 from 0x04 to set the accelerometer and gyroscope sampling rate at 1Khz
writeByte(MPU9250_ADDRESS, SMPLRT_DIV, 0x02)

DLPF_CFG changed it to 0x02 from 0x03 to set the gyro bandwidth to 92Hz
writeByte(MPU9250_ADDRESS, CONFIG, 0x02)

c=c|0x02 ; // changed it to 0x02 from 0x03 to set the gyro bandwidth to 92Hz
writeByte(MPU9250_ADDRESS, CONFIG, ACCEL_CONFIG2,c)

What other changes do I need to do? Do I need to change the i2c frequency to 1Mhz. Do I need to change the i2c resistance for operating at 1Mhz
 What about the baud rate between the ESP32 and the desktop?

kriswiner, 2019-10-21:
This is not what I would have done. Sensor fusion is best done on the
embedded MCU, and the ESP32 is more than capable to provide fast enough
fusion rates to get good results. I suspect either the sensors are not
calibrated sufficiently or you are not feeding the sensor data to the
fusion filter in the proper order.

I have done testing with the Madgwick fusion filter and MPU9250 that shows
pretty low latency response using similar MCUs (like Teensy or
STM32L4-based MCUs). You should be able to get the same results with an
ESP32. Here is some results using a Teensy 3.2 I believe:

https://github.com/kriswiner/MPU6050/wiki/9-DoF-Motion-Sensor-Bakeoff

Here is the sketch I used for the ESP32 to get similar results:

https://github.com/kriswiner/ESP32/tree/master/MPU9250_MS5637

shomedas, 2019-10-22:
kindly tell me if the variables I have changed are correct.  I would try to run with the modified parameters on the esp32. Can I run 1Mhz i2c on esp32 for better results. 

I get reliable heading estimate from esp32 under static condition or in slow movement with the IMU. If I move faster it takes time ( two or three seconds)to converge. I get sensor fusion rate of only 2khz on the esp32. So I thought I am restricted to 250hz sampling on accelerometer and gyroscope. Can I try higher sampling rates.

My magnetometer calibration is a perfect sphere. I have calibrated magnetometer using ellipsoid method. I have calibrated accelerometer using least squares method from AN4399 application note from freescale. Accelerometer calibration also should be nearly ok.

I don't have access to stm32 or tindie devices immediately. So I have to manage with esp32 or the desktop.

Kindly help.

kriswiner, 2019-10-22:
Settings are fine. You might try 500 Hz or 1 kHz gyro/accel rate. 2 kHz
fusion rate seems a bit slower than what I usually see, but i don't
remember exactly. Still, if you are feeding the sensor data properly to the
fusion filter and your sensors are calibrated  then you should see much
lower latency that a few seconds to converge.

shomedas, 2019-10-22:
Thank you for your help, Kris

shomedas, 2019-10-22:
Kris, what low pass frequency filter should I use with 500hz sampling rate. Is 92Hz ok?

kriswiner, 2019-10-22:
Yes.

The ESP32 is supposed to have a FPU so you should be seeing much higher
fusion rates than 2 kHz with this.

I ran this sketch <https://github.com/kriswiner/ESP8285/tree/master/MPU9250>
with the MP9250 on the ESP8285 and got 2 kHz fusion rates. So something is
wrong with your sketch perhaps.

Could be a buggy I2C API I suppose, but if you are using the data ready
interrupt  and aren't bogging the main loop down with lots of extraneous
stuff you should be well over 10 kHz fusion rate with the ESP32.

In the ESP32 sketch I linked to before, I am only iterating the Madgwick
filter 10 times because the fusion rate was way to fast, such that I was
simply wasting cycles on iterating. Meaning the rate was >> 2 kHz when I
ran my sketch. I suspect there is something wrong with yours, like you are
polling for data ready or something else like this. Not good...

shomedas, 2019-10-22:
Will the 8285 code link given in your post above work with the esp32?

Then I would try again with it

kriswiner, 2019-10-22:
Maybe, why not just use the sketch I linked to yesterday specifically for
the ESP32?

shomedas, 2019-10-22:
Sorry I thought you were referring to the link in your post above.

I have been using the esp32 code(with some modifications) as you mentioned in your post yesterday. I will remove any unnecessary part and try again.

Thanks for your he

kriswiner, 2019-10-22:
Are you using the data ready interrupt and 400 kHz I2C?

shomedas, 2019-10-22:
I am using i2c an 400Kh.I am using data ready interrupt as below. In fact, if the interrupt pin is disconnected, I get constant quaternion of (1,0,0,0).

void setup()
{
....
  attachInterrupt(digitalPinToInterrupt(intPin), myinthandler, RISING);  // define interrupt for INT pin output of MPU9250
}

void loop{
   // If intPin goes high, all data registers have new data
   if(newData == true) {  // On interrupt, read data
     newData = false;  // reset newData flag
     readMPU9250Data(MPU9250Data); // INT cleared on any read
...
}

kriswiner, 2019-10-22:
Then your fusion rate should be >> 2 kHz.

shomedas, 2019-10-22:
I am going to run your original sketch again and report back.

shomedas, 2019-10-22:
No kris i still get ~2Kz fusion rate. i have used your code at https://github.com/kriswiner/ESP32/tree/master/MPU9250_MS5637. I have removed only those parts pertaining to 5637 sensor. Everything else is the same. I dont have  accurate calibration, but that should not change the sensor fusion rate.

I get 
q0 = 0.38 qx = 0.09 qy = 0.90 qz = -0.19
Yaw, Pitch, Roll: 192.70, 45.95, -156.87
Grav_x, Grav_y, Grav_z: 273.15, 718.68, -639.44 mg
Lin_ax, Lin_ay, Lin_az: -279.74, -695.43, 1637.67 mg
sumCount = 1010
sum = 0.51
**rate = 1981.34 Hz**

shomedas, 2019-10-22:
I am attaching herewith your original code from [link](https://github.com/kriswiner/ESP32/tree/master/MPU9250_MS5637) without the 5637 part. if you can spare some time please check if you get >> 2KHz fusion rate. 
```

/* MPU9250_MS5637_ESP32 Basic Example Code
 by: Kris Winer
 date: December 14, 2016
 license: Beerware - Use this code however you'd like. If you 
 find it useful you can buy me a beer some time.
 
 Demonstrate basic MPU-9250 functionality including parameterizing the register addresses, initializing the sensor, 
 getting properly scaled accelerometer, gyroscope, and magnetometer data out. Added display functions to 
 allow display to on breadboard monitor. Addition of 9 DoF sensor fusion using open source Madgwick and 
 Mahony filter algorithms. Sketch runs on the 3.3 V 8 MHz Pro Mini and the Teensy 3.1.
 
 This sketch is intended specifically for the MPU9250+MS5637 Add-on shield.
 It uses SDA/SCL on pins 21/22, respectively, and it uses the Wire library.
 The MS5637 is a simple but high resolution pressure sensor, which can be used in its high resolution
 mode but with power consumption of 20 microAmp, or in a lower resolution mode with power consumption of
 only 1 microAmp. The choice will depend on the application.
 
 SDA and SCL should have external pull-up resistors (to 3.3V).
 4K7 resistors are on the MPU9250+MS5637 breakout board.
 
 Hardware setup:
 MPU9250 Breakout --------- ESP32
 VDD ---------------------- 3.3V
 SDA ----------------------- 21
 SCL ----------------------- 22
 GND ---------------------- GND
 
 */
#include "Wire.h"   

// See also MPU-9250 Register Map and Descriptions, Revision 4.0, RM-MPU-9250A-00, Rev. 1.4, 9/9/2013 for registers not listed in 
// above document; the MPU9250 and MPU9150 are virtually identical but the latter has a different register map
//
//Magnetometer Registers
#define AK8963_ADDRESS   0x0C
#define WHO_AM_I_AK8963  0x00 // should return 0x48
#define INFO             0x01
#define AK8963_ST1       0x02  // data ready status bit 0
#define AK8963_XOUT_L	   0x03  // data
#define AK8963_XOUT_H	   0x04
#define AK8963_YOUT_L	   0x05
#define AK8963_YOUT_H	   0x06
#define AK8963_ZOUT_L	   0x07
#define AK8963_ZOUT_H	   0x08
#define AK8963_ST2       0x09  // Data overflow bit 3 and data read error status bit 2
#define AK8963_CNTL      0x0A  // Power down (0000), single-measurement (0001), self-test (1000) and Fuse ROM (1111) modes on bits 3:0
#define AK8963_ASTC      0x0C  // Self test control
#define AK8963_I2CDIS    0x0F  // I2C disable
#define AK8963_ASAX      0x10  // Fuse ROM x-axis sensitivity adjustment value
#define AK8963_ASAY      0x11  // Fuse ROM y-axis sensitivity adjustment value
#define AK8963_ASAZ      0x12  // Fuse ROM z-axis sensitivity adjustment value

#define SELF_TEST_X_GYRO 0x00                  
#define SELF_TEST_Y_GYRO 0x01                                                                          
#define SELF_TEST_Z_GYRO 0x02

/*#define X_FINE_GAIN      0x03 // [7:0] fine gain
#define Y_FINE_GAIN      0x04
#define Z_FINE_GAIN      0x05
#define XA_OFFSET_H      0x06 // User-defined trim values for accelerometer
#define XA_OFFSET_L_TC   0x07
#define YA_OFFSET_H      0x08
#define YA_OFFSET_L_TC   0x09
#define ZA_OFFSET_H      0x0A
#define ZA_OFFSET_L_TC   0x0B */

#define SELF_TEST_X_ACCEL 0x0D
[... 961 lines cut ...]
  Wire.endTransmission();           // Send the Tx buffer
}

  uint8_t readByte(uint8_t address, uint8_t subAddress)
{
  uint8_t data = 0;                        // `data` will store the register data   
  Wire.beginTransmission(address);         // Initialize the Tx buffer
  Wire.write(subAddress);                  // Put slave register address in Tx buffer
  Wire.endTransmission(false);             // Send the Tx buffer, but send a restart to keep connection alive
  Wire.requestFrom(address, 1);            // Read two bytes from slave register address on MPU9250 
  data = Wire.read();                      // Fill Rx buffer with result
  return data;                             // Return data read from slave register
}

  void readBytes(uint8_t address, uint8_t subAddress, uint8_t count, uint8_t * dest)
{  
  Wire.beginTransmission(address);   // Initialize the Tx buffer
  Wire.write(subAddress);            // Put slave register address in Tx buffer
  Wire.endTransmission(false);       // Send the Tx buffer, but send a restart to keep connection alive
  uint8_t i = 0;
  Wire.requestFrom(address, count);  // Read bytes from slave register address 
  while (Wire.available()) {
        dest[i++] = Wire.read(); }         // Put read results in the Rx buffer
}
```

kriswiner, 2019-10-22:
The sketch is designed to throttle the sensor fusion rate since the MCU is
so fast. In other words, it is designed to provide ten iterations of the
sensor fusion filter for every new gyro data set.

for(*uint8_t i = 0; i < 10; i++*) { // iterate a fixed number of times
per data read cycle
    Now = micros();
    deltat = ((Now - lastUpdate)/1000000.0f); // set integration time
by time elapsed since last filter update
    lastUpdate = Now;

    sum += deltat; // sum for averaging filter update rate
    sumCount++;

    MadgwickQuaternionUpdate(-ax, ay, az, gx*pi/180.0f, -gy*pi/180.0f,
-gz*pi/180.0f,  my,  -mx, mz);
    }

Since the gyro data rate is 200 Hz, the filter rate will be 10 x 200
Hz = 2 kHz, as you see.

If you want to use a faster fusion rate, this really means more
iterations. So change the ten in the above loop to 20, 30, or even 50
and see what you get. There will be a limit.

I would set the gyro/accel rate to 500 Hz or 1 kHz and experiment with
10, 20, 30, etc iterations to see if you can get what you want. In my
experience 20 is plenty, but YMMV.

shomedas, 2019-10-22:
I get the idea. Increase gyro sampling rate to get better resolution w.r.t. time. Then complete as many madgwick iterations as possible in the time interval between two consecutive sets of gyro data.
But how do we ensure that no gyro data set is skipped during the execution of the madgwick for loop?

kriswiner, 2019-10-22:
testing. At 1 kHz gyro sample rate you have 1 msec to do all the
iterations. So if the mcu can run at 50kHz rate as the STM32L4 can, you
could complete 50 iterations.

shomedas, 2019-10-23:
I measured the time taken between two consecutive set of data packets arriving. Strangely I found that new data arrives every I microsecond, not 1 milisecond

```

void loop()
{  
    if(newData == true) {  // On interrupt, read data
   
     newData = false;  // reset newData flag

    Now = micros();
    deltat = (Now - lastUpdate); 
    Serial.print("time between two data sets= "); Serial.println(deltat);
    lastUpdate = micros();          
    }
}

```
Output : 

```
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
X-Axis sensitivity adjustment value 1.18
Y-Axis sensitivity adjustment value 1.18
Z-Axis sensitivity adjustment value 1.14
time between consecutive data packets in microsec= 13466409.00
time between consecutive data packets in microsec= 4936.00
time between consecutive data packets in microsec= 4942.00
time between consecutive data packets in microsec= 1.00
time between consecutive data packets in microsec= 1.00
time between consecutive data packets in microsec= 1.00
time between consecutive data packets in microsec= 1.00

```

kriswiner, 2019-10-23:
Not possible, sorry..something in your sketch is wrong.

shomedas, 2019-10-23:
For the above timing I have used your ESP32 code from [https://github.com/kriswiner/ESP32/tree/master/MPU9250_MS5637](https://github.com/kriswiner/ESP32/tree/master/MPU9250_MS5637). I have removed 5637 related code.  Only other change is 

 writeByte(MPU9250_ADDRESS, SMPLRT_DIV, 0x00);  // Set sample rate to 1 kHz

kriswiner, 2019-10-23:
My sketch does not calculate time between interrupts. If you change the
sketch to do this you should see about 1/1000 Hz ~ 1 millisecond. It is
simply not possible for the interrupt to trigger every microsecond.
Something is not correct in your sketch.

Use an oscilloscope and verify that the interrupt is triggering at ~1 kHz
please...

shomedas, 2019-10-23:
Ok. I will use the oscilloscope to check

kriswiner, 2019-10-23:
If this is your main loop:

void loop()
{
    if(newData == true) {  // On interrupt, read data

     newData = false;  // reset newData flag

    Now = micros();
    deltat = (Now - lastUpdate);
    Serial.print("time between two data sets= "); Serial.println(deltat);
    lastUpdate = micros();
    }
}

How are you clearing the interrupt?

shomedas, 2019-10-23:
To clear interrupt I added 
readMPU9250Data(MPU9250Data); // INT cleared on any read

But the timing results Now report 500microseconds, not 1000microseconds

The loop is now 

```
void loop()
{  
    if(newData == true) {  // On interrupt, read data
   
     newData = false;  // reset newData flag
     readMPU9250Data(MPU9250Data); // INT cleared on any read

    Now = micros();
    deltat = ((Now - lastUpdate)); // set integration time by time elapsed since last filter update
    Serial.print("time between consecutive data packets in microsec= "); Serial.println(deltat);
    lastUpdate = micros();          
    }
}
```

shomedas, 2019-10-23:
I tried modifying SMPLRT_DIV to 0x00, 0x01. But the interval at which the data arrives stays at 5000microsecs which implies a 200Hz data update rate. How to modify the registers to be able to get 1Khz sampling rate i.e. to get interrupts at 1Khz rate.
SMPLRT_DIV works depending on FCHOICE and FCHOICE_b. They need to be set correctly. However the default scaling should be 1 i.e. no scaling, which would keep the sampling frequency at 1Khz itself even without modifying the SMPLRT_DIV

kriswiner, 2019-10-23:
Please check the data sheet.

shomedas, 2019-10-23:
yes doing that now.

kriswiner, 2019-10-23:
You might try increasing the bandwidth since setting a low bandwidth
amounts to internal averaging of the data so this might be changing the
effective simple rate. Although I seem to recall measuring the interrupt
rate at 200 Hz sample rate and 41 Hz bandwidth and still seeing 200 Hz or
so on the interrupt rate. So unlikely the problem is with the sensor. Can
you verify that the interrupt triggers at 200 Hz with an O scope when you
have the sample rate set to 1 kHz?