kriswiner/MPU9250 #397: i2c waveform for MPU9250 ESP32 interface

shomedas, 2019-10-25:
Hardware setup:
 MPU9250 Breakout --------- ESP32
 VDD ---------------------- 3.3V
 SDA ----------------------- 21
 SCL ----------------------- 22
 GND ---------------------- GND
SDAL, SCL pull up to 3.3V is 4.5K

I2C bus speed is set 400KHz.
SMPLRT_DIV is set at 0x01 so that the sensor sampling frequency is at 500Hz. 

I captured the SDA & SCL waveforms which are as below. They dont look right to me, they look noisy. 

What do you say kris?

SCL stays high for 2ms and then there is square wave with 400KhZ frequency. This is repeated as seen from the following two images. Should not the SCL be  just a square wave with 400KhZ frequency

SCL signal:
[image]

SCL zoomed in:
[image]

SDA is a square wave of frequency 500Hz(2ms time period). However it has a noisy, jittery region as shown below in the two Images

SDA signal:
[image]

SDA zoomed in
[image]

kriswiner, 2019-10-25:
I2C wave form looks pretty good to me.

Did you verify that the interrupt is triggering at 500 Hz?

shomedas, 2019-10-25:
i checked SDA waveform which had periodicity of 2ms ie the frequency was 500hz. is there any other way to check it.

Why is the SCL line high for significant time(2ms), then the 400Khz square wave(2.5us periodicity) occurs only for a short period. Isnt it supposed to be a just a square wave with periodicity of 2.5us?

kriswiner, 2019-10-25:
Check the interrupt pin.

shomedas, 2019-10-25:
Interrupt pin waveform. Looks ok

[image]

shomedas, 2019-10-25:
However when I time through the code,  i get 63.5 ms between capture of two interuupts
My loop to check interrupt timing is

```
void loop()
{  
    if(newData == true) {  // On interrupt, read data

     Now = micros();
     newData = false;  // reset newData flag
     readMPU9250Data(MPU9250Data); // INT cleared on any read

    deltat = ((Now - lastUpdate));
    Serial.print("time between consecutive data packets in microsec= "); Serial.println(deltat);
    lastUpdate = Now;          
    }
}

```

kriswiner, 2019-10-25:
Interrupt looks like it is firing at 1 kHz, is this right?

shomedas, 2019-10-25:
No. its 500Hz. between two interrupts is 2ms. I checked on the CRO. The image is over exposed

shomedas, 2019-10-25:
I don't understand why the timing shows 63.5 msec

kriswiner, 2019-10-25:
OK, so you set the divide to 1 so you are getting 1 kHz/2 output data rate,
and the interrupt is firing t 500 Hz as it should?

Then, you are actually seeing 63 ms instead of 2 ms between data ready in
your loop?

kriswiner, 2019-10-25:
Me either, it should show 2 ms, plus maybe a bit for I2C transactions...
What is the bandwidth? Try using very high bandwidth or no bandwidth

shomedas, 2019-10-25:
which bandwidth are you referring to? The low pass filter bandwidth?

kriswiner, 2019-10-25:
Yes, turn it off. or set it to 500 Hz.

shomedas, 2019-10-25:
DLPF_CFG set to 0x07:  3600 Hz gyro bandwidth
A_DLPF_CFG set to 0x07:  460 Hz accel bandwidth

still the loop timing reads 63.5ms.

shomedas, 2019-10-25:
I reduced the print statements length from 50 characters to 10 characters and timing changed from 63.5 ms to 17.5ms. 

Is it possible that some interrrupt cycles are lost if we print text in the loop.

kriswiner, 2019-10-25:
UART is slow. Move the print statement out of the timing loop.

shomedas, 2019-10-25:
The printing has to be taken out of the void loop().

Anyway this means that if we print anything in the void loop, we will loose some interrupt cycles.