kriswiner/MPU9250 #400: Wrong accel data when reading data from multiple MPU9250 via SPI with Arduino UNO

PJiaHeng, 2019-11-15:
Hi Kriswiner,

I am trying to get accel and gyro data from multiple MPU9250 via SPI(mode 3) with Arduino UNO.
The cable linking MPU9250 and Arduino is DuPont line.
The way to change SPI slave is to set corresponding cspin to `OUTPUT` and `LOW`.
But occasionally there will be some wrong acceleration data appearing when MPU9250s are in the **at-rest**，shown as table 1. Addtionally, my accel scale is **2g**. In fact, the ax data should be a value of about 0.

Table 1

| ax | ay | az | gx | gy | gz |
| :-: | :-: | :-: | :-: | :-: | :-: |
| -2.0000 | -0.0024 | 0.9224 | -0.0610 | 0.3738 | 0.0763 |
| -2.0000 | -0.0015 | 0.9141 | -0.1068 | 0.1221 | -0.1373 |
| -2.0000 | -0.0098 | 0.9067 | 0.0763 | -0.1678 | 0.0076 |
| -2.0000 | -0.0073 | 0.9165 | -0.1068 | 0.3433 | -0.0763 |
| -2.0000 | -0.0059 | 0.9126| 0.0458 | 0.1526 | 0.0229 |

And when MPU9250s are motioned, the accel data will be changed into these data as table 2. In fact, the ax data should be a value of about 0.98.

Table 2

| ax | ay | az | gx | gy | gz |
| :-: | :-: | :-: | :-: | :-: | :-: |
| -1.3931 | 0.0112 | -0.1455 | -0.0839 | 0.3357 | -0.0610 |
| -1.3916 | 0.0166 | -0.1338 | -0.0839 | 0.2365 | 0.0458 |
| -1.3828 | 0.0054 | -0.1406 | -0.0610 | 0.2747 | -0.1450 |
| -1.3877 | 0.0132 | -0.1465 | -0.0610 | 0.2747 | -0.1907 |
| -1.3926 | 0.0151 | -0.1392 | -0.0534 | 0.0839 | -0.0229 |

The value of ACCEL_CONFIG2 is `0x03`.
The value of CONFIG is `0x03`.
The value of SMPLRT_DIV is `0x00`.

When I re-plugged the Arduino USB serial cable, the accel data returned to normal.

So, my question to you are:
1. Do you meet this situation?
2. Can you give me some advance about how to relese this wrong situation?
3. More, do you know what reason does lead to this problem?

Thanks a lot! Your Github page helped so many people getting the MPU9250 working correctly. I highly appreciate your work.

PJiaHeng

kriswiner, 2019-11-15:
Looks like the accel calibration is not working properly, so  I would
suggest trying the calibration with the sensor laying flat on a table. In
other words, if the sensor is not flat, depending on which calibration
routine you are using, you could be adding 1 incorrectly.

I would also start with one sensor and make sure you can get reasonable
results. The accel/gyro should read 0, 0, 1 g and 0, 0, 0 dps when the
sensor is laying flat and motionless.

You might need to pupgrade to an STM32 or other 32-bit MCU since an Arduino
UNO migh not be fast enough to manage more than one sensor at a time.

And lastly, Dupont wires are not a robust connection for fast SPI, I would
solder 24 guage copper wires instead.

PJiaHeng, 2019-11-16:
Thanks.
I will consider to change 24 gauge copper wires.

Then, let me say sorry to you. Maybe I didn't describe my situation clearly, let me make some supplements:

The sensors were always laying flat on a table when I calibrated sensors. And when I got these accel/gyro datas which are in above tables, the sensor were flat on a table(no motion).

I can get reasonable results when the number of connected MPU9250s are less than 7(I have connected 1/3/6 MPU9250s). But when I connect 7 or more MPU9250s, the wrong accel data occurs frequently. Once I re-plugged the Arduino USB serial cable, the accel data returned to normal.

kriswiner, 2019-11-16:
Maybe the UNO can't manage to read seven or more sensors before the next
data sample is ready?

"  Once I re-plugged the Arduino USB serial cable, the accel data returned
to normal."

You would need 7 nCS pins, does the UNO have this many?

Not sure why this would be. Again, maybe get a better (faster) MCU?

PJiaHeng, 2019-11-18:
Thank you very much.
The wrong data no longer appears when I try to power the arduino independently.
In addition, I would like to ask you, what kinds of wires can be used as a reliable connection for SPI except 24 gauge cooper wires?

kriswiner, 2019-11-18:
almost any insulated copper wire, solid core or braided. 24 - 30 gauge. I
like the silicone insulated wires sold by Adafruit.