kriswiner/MPU9250 #401: Integrating with Feather M0

TeslaHoax, 2019-11-18:
Hey Kriswiner,

I've had some communications issues trying to adapt your sketch, specifically the one SparkFun provided, to work with the Feather M0. I've been testing it out on the UNO and been happy with the calibration and readings. 

When I try to adapt it with the Feather M0, some how the Communications (likely the I2C) is interfering with the USB Coms, and upon successful upload, the USB Coms port drops out. It's important to note that it is not a problem with the controller, all of my other programs work fine and I2C_Scanner picks up the MPU9250 fine. I have been trying to find where the SDA and SCL ports are defined in the libraries but with no luck. Been stuck on this for awhile, any help would be greatly appreciated. 

Josh

kriswiner, 2019-11-18:
Sorry, I have never used the Feather. Maybe Adafruit can help?

TeslaHoax, 2019-11-18:
It seems as though they can't help since the feather is operating fine with other code and hardware.

kriswiner, 2019-11-18:
So the feather works fine by itself and with other sensors. The MPU9250
works fne with other MCUs? Sounds like a connection problem or some other
kind of interference specific to how you have chosen to connect the MPU9250
and Feather. I am afraid I can't offer any help here since there is nothing
wrong with the MPU9250 sketch per se.

TeslaHoax, 2019-11-18:
Yes that is the case, As soon as I take the main function out of the script it uploads fine. 
MPU9250 myIMU(MPU9250_ADDRESS, I2Cport, I2Cclock);

Must be the I2C coms.

Thanks anyways.

TeslaHoax, 2019-11-21:
Hey,

So I finally just scrapped the SparkFun Library and integrated your sketch into what I had. Works perfect. Weird.

Thanks!