kriswiner/MPU9250 #402: magCal issue with library

miniflyer-nd, 2019-11-18:
Hi kris, hi guys

For a project i am using the current mpu9250.h 
I need to skip the calibration routine on standard startup, and calibrate only on demand, then saving biases and scale to the eeprom.
For the accel and gyros this works fine, however mag returns nonsense readings:

> 
Mag Calibration: Wave device in a figure 8 until done!
  4 seconds to get ready followed by 15 seconds of sampling)
Mag Calibration done!
AK8963 mag biases (mG)
5559.74
5833.02
5623.62
AK8963 mag scale (mG)
nan
nan
nan

> 

I have found out that the mag calibration works fine ONLY when called in the standard startup routine. Even placing it in an if-clause during startup renders all results useless.

> 

int cal=1;

if (cal=1) {

  myIMU.magCalMPU9250(myIMU.magBias, myIMU.magScale);

> 

One thing i noticed: the calibrate order for the mpu turns orange just like a normal order in the arduino IDE. The magCal order however stays black in the IDE.....

Please help me out

kriswiner, 2019-11-18:
store calibration values in the array declaration and comment out the
calibration call.

miniflyer-nd, 2019-11-18:
This will not enable me to do an on demand calibration run.

I want to load all biases and scales from eeprom. Upon pressing a button, i want to start a seperate order to start a calibration routine and write them to the eeprom to be used as standard until a new calibration is done. This only works if i can call the calibration routine from another point in the code...
(I created a void CalToEEPROM routine to do this, and signal via LED what was expected....)

If i get useless results as soon as i move the code, this does not work....

kriswiner, 2019-11-18:
we do something like this, should be straightforward.

not sure what to tell you. you have all of the pieces, just need to put
them together so they work.

i can't program it for you.

miniflyer-nd, 2019-11-18:
Yes should be.....it also is for the gyros and accelerometers, because the code allows me to move it around or call it at any given time with usable results. It works wonderfully.

I just cannot figure out why i cannot call the magnetic calibration just like the others. Well i can...it works....it just delivers wrong output. I have checked the ccp file, the h file....no matter what i do, it will only run correctly directly in the setup code.

(Copying the magcal routine into the calibration routine in the library strangely corrupts the entire library)

I'm not asking you to program anything for me, i just need help calling the magCal function outside the setup (or tied into an if clause). 

I simply cannot figure out why precicely this void is not recognized as an order, and i've been on it a week nearly fulltime now.....i was hoping that you as the ultimate god of the code could explain or fix the abnormal behavior of this one function....

kriswiner, 2019-11-18:
Not all sketches are alike, I would start with this one:

https://github.com/kriswiner/MPU9250/blob/master/MPU9250_BME280_SPIFlash_Ladybug/MPU9250_BME280_SPIFlash_Ladybug.ino

kriswiner, 2019-11-18:
If you comment out the mg calibration call, do you still get weird results?
I don't....

miniflyer-nd, 2019-11-19:
If i don't calibrate the magnetometers the values are zero. The entire unit becomes drifty and jittery.

I can calibrate, write the values to eeprom on a seperate sketch, and reboot using eeprom values. This works great until the environment changes (unit is moved 200 miles, different electrical environment....). Then the output is back to drifty and jittery.

If i now recalibrate accel and gyros only, the new results remain drifty and jittery.....

miniflyer-nd, 2019-11-19:
I will try the linked sketch tomorrow. Past 1am here...

kriswiner, 2019-11-19:
The mag data are multiplied by a scalar as well as have offsets applied. if
you don;t load calibration data into the mag calibration arrays then the
mag values will always be zero (because anything times 0 scale factor is,
not surprisingly, 0). If the mag field is always 0, then the orientation
solution is garbage.

So agaon, load calibration data into the calibration arrays to avoid this.

miniflyer-nd, 2019-11-19:
Yesssss.....load calibration data. But i cannot load calibration values if i'm not able to obtain calibration data. I can load calibration data with fixed values, but these values are only useful for one single environment.

If i move 200 miles before the reboot, my old values are useless.
And i do not want to dismantle everything, plug in the laptop, recalibrate, write to eeprom, reassemble, and then do everything all over when i move again. 
I need to be able to call a calibration routine, providing valid information, called at any time (within an if clause in the setup will be enough).

miniflyer-nd, 2019-11-19:
Target:
Day 1; Position A: calibrate with figure 8s and automatically save. System works
Day 2; Position A: system works with saved values.
Day 3; Position B: system jitters. Recalibrate at push of button, saving new values. System works.
Day 4; Position B: system works....
Day 5; Position C: jittery drifty results. Push the button.....

kriswiner, 2019-11-19:
I understand what you want to do, and it is trivial to do so, I am not sure
why you can;t make it work...

miniflyer-nd, 2019-11-19:
It would be trivial if i could move the magCal routine to within an if clause. 
I have a fully functional code. Just moving the 1 single order to within a simple if clause makes it misbehave. Taking it from the if clause fixes everything. Calling it in a seperate void gives out horsesh.....oes, calling it in setup makes MPU great again.....

The magCal is not recognized as a standard order (thats why it does not turn orange in arduino), and that makes this trivial peanut problem unsolveable to me....the problem lies within the library somewhere....

miniflyer-nd, 2019-11-19:
Kris, i really appreciate your time and effort, but i have to come back to you again on this issue. I have spent most of today trying to figure it out, and i'm feeling pretty stupid because i'm not one step further. 
I spent much of today copying library parts back and forth, and also modifying your Ladybug codes so they will run on my setup, but i find _the same issue_ there. The MagCal routine runs ONLY in an unconditioned setup. 

First of all i want to show you the code portion of my programming in the setup.
Green arrow marks the spot where Accel and Gyro are calibrated. Notice the order is orange, as are all the other orders called below.
Orange arrow marks the commented out calibrate() void routine located at the end of the code, which calls the calibrate and MagCal routines, and then writes the numbers to designated EEPROM adresses. 
Red arrow marks the MagCal routine. It runs fine like this, but notice it is not orange as are all other valid library orders! 
Calling this order in the commented out calibrate() void, or placing it in an if clause, makes it run but instead of the expected numbers i get during a standard setup routine, they vary a lot.
For the biases i am expecting [-63.34; -251.56; -255.15] at my desk with all lamps on, i am getting [5559.74; 5833.02; 5623.62]. Scales are all fairly close to 1, the results they read are "nan".

[image]

What am i running?
Its an Arduino Nano (mounted either to a breadboard or a cnc stepper shield, same results), an MPU9250 /6500 without pressure sensor, connected via I2C, power supply via USB and through the 3.3V output of the Nano.
The Libraries are now again unchanged libraries from SparkFun (SparkFun_MPU-9250_Breakout_Arduino_Library) which is forked from your site here.
Code is the MPU9250Basic_AHRS_I2C example included in the code, but freed of all display code particles. I will be glad to provide the full code (which includes a lot of stepper codes to move according MPU), but the behavior can be seen from an empty sketch including just the wire and MPU library, a basic intialisation and a call of the orders...

I would appreciate any hint to help me solve this, because i still cannot understand why some orders work in the void calibrate(), and others deliver mashed potatoe....

kriswiner, 2019-11-19:
This sounds like a question for Sparkfun.

miniflyer-nd, 2019-11-19:
I contacted them. They pointed me to their github account. And opening an issue got me right here.

The header of the code states its written by you.....it also closely matches, the libraries are almost interchangeable apart from a few designations....

If its so trivial and easy, then why can noone give a hint?

kriswiner, 2019-11-19:
You are using Sparkfun's modification of my oldest and least well-developed
code. They didn't ask me, they just guessed, and chose poorly. And now they
have their own MPU9250 code written completely from scratch, so you could
use that. I can't fix Sparkfun's code, no matter that they copied a lot of
it from me.

What I would recommend to you is to write your own mag calibration
function. Start small and make sure it is orange or whatever. Then add
functionality until it does what you want. All this function is doing is
averaging a bunch of data, calculating max and min and then calculating
scale and offset from these. Like this.
<https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration>
Surely you can program this...

miniflyer-nd, 2019-11-20:
Ahhhhhh, i thought it is an official collaboration since they are pointing all their support to here.

I fully understand youre not going to improve on a stolen old code.....

I would like to talk to you about my project which i am currently working on. Can you give me an email where i can reach you?

kriswiner, 2019-11-20:
Please contact me at tleracorp@gmail.com.