kriswiner/MPU9250 #404: MPU9250 Chinese Clones No ACK on Magnetometer

JakeBoothy, 2019-12-12:
I bought 4 chinese [MPU9250's on Amazon]((https://www.amazon.ca/gp/product/B07TV4L6HT/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)) and I can't get values out of the magnetometer on any of them. At one point I had one of them fully working but shortly thereafter it stopped. Since then I've never seen any of the magnetometer's work at all. Depending on the library I use, either only accel/gyro will work or neither will work. Is this simply just a hardware issue? I just didn't expect them to be quite this bad so I'm having trouble convincing myself that this is purely a hardware problem. What would make these boards so prone to failure? Why would it work for only a few seconds and then stop working? Or could the problem still be with the software?

kriswiner, 2019-12-12:
Probably fake MPU9250 or bad pcb design or both.

JakeBoothy, 2019-12-12:
Ok, will purchase something real next time!

closed 2019-12-12 by JakeBoothy

JakeBoothy, 2019-12-12:
Just to put a pin in the hat of these weird Chinese boards... I got to work today and one of them was fully working again! I've had electronics spontaneously break on me but rarely do they spontaneously start working again. These boards are an enigma...