kriswiner/MPU9250 #405: Dual_Ladybug bug in Self-Test function?

sebvaillancourt, 2019-12-19:
Hi Kris,
I spotted some odd lines in your Self-Test function of the Dual_Ladybug code.

Dual_MPU9250/MPU9250.cpp:
"
616     uint8_t FS = 0;
[...]
619     writeByte(MPUnum, CONFIG, 0x02);   // Set gyro sample rate to 1 kHz and DLPF to 92 Hz
620     writeByte(MPUnum, GYRO_CONFIG, 1 << FS); // Set full scale range for the gyro to 250 dps
621     writeByte(MPUnum, ACCEL_CONFIG2, 0x02); // Set accelerometer rate to 1 kHz and bandwidth to 92 Hz
622     writeByte(MPUnum, ACCEL_CONFIG, 1<<FS); // Set full scale range for the accelerometer to 2 g
"

I compared it to MPU9250BasicAHRS.ino:
"
924     uint8_t FS = 0;
[...]
927     writeByte(MPU9250_ADDRESS, CONFIG, 0x02); // Set gyro sample rate to 1 kHz and DLPF to 92 Hz
928     writeByte(MPU9250_ADDRESS, GYRO_CONFIG, FS<<3); // Set full scale range for the gyro to 250 dps
929     writeByte(MPU9250_ADDRESS, ACCEL_CONFIG2, 0x02); // Set accelerometer rate to 1 kHz and bandwidth to 92 Hz
930     writeByte(MPU9250_ADDRESS, ACCEL_CONFIG, FS<<3); // Set full scale range for the accelerometer to 2 g
"

It didn't really make sense to me to shift 1 by 0 and to send that 1 to the GYRO_CONFIG. If I'm not mistaking, that would bypass the dlpf that you just set to 92Hz.

Cheers. Sébastien Vaillancourt

kriswiner, 2019-12-19:
Yes, 620 and 622 do not look right...

sebvaillancourt, 2019-12-19:
Also in MPU9250BasicAHRS.ino:
"
987   factoryTrim[0] = (float)(2620/1<<FS)*(pow( 1.01 , ((float)selfTest[0] - 1.0) )); // FT[Xa] factory trim calculation
...
"
The expression "2620/1<<FS" should probably be written "2620/(1<<FS) to evaluate the bit-shifting before the division. Although, FS is equal to 0 in your code, so there's not gonna be any difference. Though when working with other FS values, it will produce a different result:

2620/(1<<1) = 1310
2620/1<<1 = 5240

This brings me to think that, to carry out the self-test calculations, specific values have to be used anyway (250 dps, 1000Hz ODR and and DLPF 92Hz), if I'm not mistaking. Since it makes no sense to change FS anyway, why not hard-code them into the self-test function instead of using a variable?

"2620.0*(pow(1.01 , ((float)selfTest[4] - 1.0);"
and
"writeByte(MPUnum, ACCEL_CONFIG, 0x00)" instead.

Sébastien

kriswiner, 2019-12-19:
I wouldn't relay on this sketch: MPU9250BasicAHRS.ino

This was my first attempt and it likely has lots of errors like this. The
documentation for the self test is in the repository, better to create your
own, or at least use this to check.

sebvaillancourt, 2019-12-19:
Sure! No problem. I'm actually writing my own code. Though it's based on yours and while studying it, I noticed these eventual bugs and just wanted to notify you in case you want to fix them.

Sébastien

kriswiner, 2019-12-19:
Appreciate that. I have written so many MPU9250 sketches over the years of
varying sophistication and accuracy that I don;t want to go back and update
all of the errors, etc. Besides, the MPU9250 is out of production so I have
been focusing on the LSM6DSM and LIS2MDL which are better sensors anyway.

sebvaillancourt, 2019-12-19:
Very good to know!

closed 2019-12-19 by sebvaillancourt