kriswiner/MPU9250 #406: Failure to load DMP FW via SPI

lasithf, 2020-01-30:
Hi Kris, 

Hope you could shed some light here. 

I am trying to use the MPU 9250 DMP driver via SPI (due to hardware limitations I cannot use I2C interface!). 

However I can talk to the 9250 registers (e.g. WhoAmI) but when the call comes to dmp_load_motion_driver_firmware ( )  function, this fails with error -2 ( fails at the time of comparison of 16 byte data what's written vs what's read, in the mpu_load_firmware ( ) function).

I have translated I2C calls of i2c_write ( ) & i2c_read ( ) functions to respective SPI calls. 

I wonder what could be the problem, my understanding is read/write DMP memory is something different to regular registry read/write of the MPU9250. I may be wrong.

Appreciate if you could help.

kriswiner, 2020-01-30:
I am not the expert on the DMP. The last time I used it, many years ago,
one had to write the entire binary file, not two bytes at a time...

lasithf, 2020-01-30:
thanks for your replay @kriswiner , do you mean that you write the entier DMP FW, 1 byte at a time?

Do we need to enable (or so) DMP somehow before we attempt to write to it?

kriswiner, 2020-01-30:
These are questions better answered by Invensense(TDK) or Sparkfun. I never
use the DMP. When I did, long ago, yes, the binary file had to be rotted
one byte at a time. DMP should not be active when writing.

lasithf, 2020-01-31:
I found that it was the custom SPI write function had a bug. It's fixed and now I can see the raw accel, gyro data coming out from the FIFO. Thanks for the support.