kriswiner/MPU9250 #407: Save calibration values to EEPROM

quichedood, 2020-02-09:
Hi',

I'm actually using your last example "MPU9250_MS5637_AHRS_t3.ino"
It seems to work pretty well (no doubt about that) :)

I would like to use an EEPROM to store calibration data :
gyroBias & accelBias
magBias & magScale

magBias & magScale are already handled, looks good !
gyroBias & accelBias are not handled in the same way because you say in comments there is something not working with acceleration biases hardware registers.

Here are my two questions :
1) Could I handle gyro biases like accelerometer biases ?
Not storing them in hardware registers just doing a substraction in the main loop :
```
// Now we'll calculate the accleration value into actual g's
ax = (float)MPU9250Data[0] * aRes - accelBias[0]; // get actual g value, this depends on scale being set
ay = (float)MPU9250Data[1] * aRes - accelBias[1];
az = (float)MPU9250Data[2] * aRes - accelBias[2];

// Calculate the gyro value into actual degrees per second
gx = (float)MPU9250Data[4] * gRes - gyroBias[0]; // get actual gyro value, this depends on scale being set
gy = (float)MPU9250Data[5] * gRes - gyroBias[1];
gz = (float)MPU9250Data[6] * gRes - gyroBias[2];
```

2) If 1 is possible, I have to store and use the output **scaled** gyro/accel biases (not the raw gyro/accel biases, isn't it ? (the "dest1" and "dest2" values in "accelgyrocalMPU9250" function)

My personnal project is here : [https://github.com/quichedood/DAWA-6.x](url)

Thanks !

kriswiner, 2020-02-09:
>    1. If 1 is possible, I have to store and use the output *scaled*

" Could I handle gyro biases like accelerometer biases ?"  yes

I have to store and use the output *scaled* gyro/accel biases (not the raw
gyro/accel biases, isn't it ?"

You can store either, but it you store raw biases you will have to scale
them before applying. Most MCU's have embedded EEPROM.

quichedood, 2020-02-18:
Thanks a lot for your very quick reply !
I'll update my project Github if someone is interested :)

closed 2020-02-18 by quichedood