kriswiner/MPU9250 #409: MPU9250 Arduino Uno AK8693 0xFF

MMIC01, 2020-03-08:
Hi Kris, 
I am using the MPU9250BasicAHRS and connected the following:

VCC - 3.3V
GND - GND
SCL -10k RESISTOR - A5
SDA -10k RESISTOR - A4

I am getting the following error:
[image]

Would greatly appreciate your help!
[New Text Document.txt](https://github.com/kriswiner/MPU9250/files/4303196/New.Text.Document.txt)

MMIC01, 2020-03-08:
I followed the other treads available but still getting this errror

kriswiner, 2020-03-08:
When you perform an I2C scan, which devices show on the I2C bus? It is
likely you have the wrong MPU9250 address.

kriswiner, 2020-03-08:
Ah, I see...

You can talk with the MPU6500 but not the AK8963C. Then you need to put the
MPU9250 I2C into bypass mode.

If you are using one of my sketches that already does this, then you
probably are using a cheap Chinese board that doesn;t have an embedded.
AK8963C

kriswiner, 2020-03-08:
Or the board is poorly designed and missing the connection to 3V3 on pin 1.

MMIC01, 2020-03-08:
> When you perform an I2C scan, which devices show on the I2C bus? It is likely you have the wrong MPU9250 address.
> […](#)

Firstly thank you for your reply, I really appreciate! Basically a device is found at 0x68.... I think it is correct right?
[image]

MMIC01, 2020-03-08:
By any chance do you know if any scanners for the magnetometer only exist ?

MMIC01, 2020-03-08:
The MPU i'm using, I bought it locally it is the ACCELERATION, GYRO & MAGNETIC SENSOR MODULE MPU9250 - HAITRONIC Product Code: HR0402 Module Model  GY-9250

kriswiner, 2020-03-08:
Yes. 0x68.

Did you enable bypass mode in the MPU9250 initialization?

Could be  defective sensor or poor circuit design.

kriswiner, 2020-03-08:
Usually GY-9250 will work.

So likely a software problem. Check bypass mode enable.

MMIC01, 2020-03-09:
I believe that the bypass mode is enabled through these lines;

   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x12);  // INT is 50 microsecondpulse and any read to clear
   writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x01);  // Enable data ready (bit0) interrupt
   delay(100);

kriswiner, 2020-03-09:
You should be able to see the AK8963C on the I2C bus then. Did you try
simply reading the WHO_AM_I register of the AK8963C?

Could be that you have a faulty board. Try another.

MMIC01, 2020-03-09:
Sorry for asking... I am new to this and have minimal experience but this is part a university project I was given as a student. How do you read the WHO_AM_I register for the AK8963 please?
Thanks again!

If I will not manage to resolve the problem then I'll buy another board

MMIC01, 2020-03-09:
On another note, for this line 
writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x12);
there is a commented line
//writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x22);

Would it be possible to explain the difference please ?

kriswiner, 2020-03-09:
Consult the register map.

kriswiner, 2020-03-09:
Look at the register map.

MMIC01, 2020-03-09:
> Consult the register map.
> …

WHO_AM_I register is returning 0xFF

kriswiner, 2020-03-09:
Try a known good board then...

MMIC01, 2020-03-09:
Do you recommend any other brands please?

MMIC01, 2020-03-09:
I found SparkFun 9DoF Sensor Stick  SEN-13944

kriswiner, 2020-03-09:
https://www.notwired.co/products/detail/nwmotmpu9250-notwired-co/605716/

MMIC01, 2020-03-09:
Thanks a lot :)

closed 2020-03-09 by MMIC01

MMIC01, 2020-04-25:
Problem was with the model. replaced it with a GY91, worked fine.