kriswiner/MPU9250 #411: is MPU9250 works well with arduino uno ?

ahmedgamaleng, 2020-03-31:
hi kriswiner ,
is MPU9250 works well with Arduino Uno ?

kriswiner, 2020-03-31:
yes, but MPU9250 is 3V3 only and the fusion rate will be very low

ahmedgamaleng, 2020-03-31:
the data stays the same even if I move the sensor
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = -470 my = -120 mz = -125 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -13.80, 0.00, 0.00
rate = 630.19 Hz

kriswiner, 2020-03-31:
You are not receiving any data.

Maybe you don;t have it connected properly.

ahmedgamaleng, 2020-03-31:
i was scanning my i2c
Scanning...
I2C device found at address 0x68  !
done

and WHO_AM_I_MPU9250 is working fine ,, 

MPU9250 I AM 71 I should be 71

and my connection is 
 VDD ---------------------- 5V (there is a voltage regulator (5 to 3.3 v) built-in  
 SDA ----------------------- A4
 SCL ----------------------- A5
 GND ---------------------- GND

kriswiner, 2020-03-31:
Then maybe don;t have the interrupt woking?

ahmedgamaleng, 2020-03-31:
sorry, I do not understand what you mean. can you give me more details please

kriswiner, 2020-03-31:
Send me a link to the exact sketch you are using.

ahmedgamaleng, 2020-03-31:
/* MPU9250 Basic Example Code
 by: Kris Winer
 date: April 1, 2014
 license: Beerware - Use this code however you'd like. If you 
 find it useful you can buy me a beer some time.
 
 Demonstrate basic MPU-9250 functionality including parameterizing the register addresses, initializing the sensor, 
 getting properly scaled accelerometer, gyroscope, and magnetometer data out. Added display functions to 
 allow display to on breadboard monitor. Addition of 9 DoF sensor fusion using open source Madgwick and 
 Mahony filter algorithms. Sketch runs on the 3.3 V 8 MHz Pro Mini and the Teensy 3.1.
 
 SDA and SCL should have external pull-up resistors (to 3.3V).
 10k resistors are on the EMSENSR-9250 breakout board.
 
 Hardware setup:
 MPU9250 Breakout --------- Arduino
 VDD ---------------------- 3.3V
 VDDI --------------------- 3.3V
 SDA ----------------------- A4
 SCL ----------------------- A5
 GND ---------------------- GND
 
 Note: The MPU9250 is an I2C sensor and uses the Arduino Wire library. 
 Because the sensor is not 5V tolerant, we are using a 3.3 V 8 MHz Pro Mini or a 3.3 V Teensy 3.1.
 We have disabled the internal pull-ups used by the Wire library in the Wire.h/twi.c utility file.
 We are also using the 400 kHz fast I2C mode by setting the TWI_FREQ  to 400000L /twi.h utility file.
 */
#include <SPI.h>
#include <Wire.h>   
#include <Adafruit_GFX.h>

// Using NOKIA 5110 monochrome 84 x 48 pixel display
// pin 9 - Serial clock out (SCLK)
// pin 8 - Serial data out (DIN)
// pin 7 - Data/Command select (D/C)
// pin 5 - LCD chip select (CS)
// pin 6 - LCD reset (RST)
//Adafruit_PCD8544 display = Adafruit_PCD8544(9, 8, 7, 5, 6);

// See also MPU-9250 Register Map and Descriptions, Revision 4.0, RM-MPU-9250A-00, Rev. 1.4, 9/9/2013 for registers not listed in 
// above document; the MPU9250 and MPU9150 are virtually identical but the latter has a different register map
//
//Magnetometer Registers
#define AK8963_ADDRESS   0x0C
#define AK8963_WHO_AM_I  0x00 // should return 0x48
#define AK8963_INFO      0x01
#define AK8963_ST1       0x02  // data ready status bit 0
#define AK8963_XOUT_L	 0x03  // data
#define AK8963_XOUT_H	 0x04
#define AK8963_YOUT_L	 0x05
#define AK8963_YOUT_H	 0x06
#define AK8963_ZOUT_L	 0x07
#define AK8963_ZOUT_H	 0x08
#define AK8963_ST2       0x09  // Data overflow bit 3 and data read error status bit 2
#define AK8963_CNTL      0x0A  // Power down (0000), single-measurement (0001), self-test (1000) and Fuse ROM (1111) modes on bits 3:0
#define AK8963_ASTC      0x0C  // Self test control
#define AK8963_I2CDIS    0x0F  // I2C disable
#define AK8963_ASAX      0x10  // Fuse ROM x-axis sensitivity adjustment value
#define AK8963_ASAY      0x11  // Fuse ROM y-axis sensitivity adjustment value
#define AK8963_ASAZ      0x12  // Fuse ROM z-axis sensitivity adjustment value

#define SELF_TEST_X_GYRO 0x00                  
#define SELF_TEST_Y_GYRO 0x01                                                                          
#define SELF_TEST_Z_GYRO 0x02

/*#define X_FINE_GAIN      0x03 // [7:0] fine gain
#define Y_FINE_GAIN      0x04
#define Z_FINE_GAIN      0x05
#define XA_OFFSET_H      0x06 // User-defined trim values for accelerometer
#define XA_OFFSET_L_TC   0x07
[... 1064 lines cut ...]
            norm = 1.0f/norm;
            s1 *= norm;
            s2 *= norm;
            s3 *= norm;
            s4 *= norm;

            // Compute rate of change of quaternion
            qDot1 = 0.5f * (-q2 * gx - q3 * gy - q4 * gz) - beta * s1;
            qDot2 = 0.5f * (q1 * gx + q3 * gz - q4 * gy) - beta * s2;
            qDot3 = 0.5f * (q1 * gy - q2 * gz + q4 * gx) - beta * s3;
            qDot4 = 0.5f * (q1 * gz + q2 * gy - q3 * gx) - beta * s4;

            // Integrate to yield quaternion
            q1 += qDot1 * deltat;
            q2 += qDot2 * deltat;
            q3 += qDot3 * deltat;
            q4 += qDot4 * deltat;
            norm = sqrtf(q1 * q1 + q2 * q2 + q3 * q3 + q4 * q4);    // normalise quaternion
            norm = 1.0f/norm;
            q[0] = q1 * norm;
            q[1] = q2 * norm;
            q[2] = q3 * norm;
            q[3] = q4 * norm;

        }

kriswiner, 2020-03-31:
I asked for a link.

What are ares and gres values?

ahmedgamaleng, 2020-03-31:
I was made some modifications to the code because it doesn't work well with Arduino oun . the modifications are : 
- add MadgwickQuaternionUpdate and MahonyQuaternionUpdate function from quaternionFilters file  to the file that linked below because it was not declared on the scope 
- removing all lcd commands because there is no memory in arduino .

https://github.com/kriswiner/MPU9250/blob/master/MPU9250BasicAHRS.ino

What are ares and gres values?
Gscale = GFS_250DPS
Ascale = AFS_2G
so, 
aRes = 2.0/32768.0;
gRes = 250.0/32768.0;

kriswiner, 2020-03-31:
Are these the values printed out on the serial monitor for ares and gres?

ahmedgamaleng, 2020-03-31:
I think the values didn't print at all .so this is the value printed out 👍   

MPU9250 I AM 71 I should be 71
MPU9250 is online...
x-axis self test: acceleration trim within : -100.0% of factory value
y-axis self test: acceleration trim within : -100.0% of factory value
z-axis self test: acceleration trim within : -100.0% of factory value
x-axis self test: gyration trim within : -100.0% of factory value
y-axis self test: gyration trim within : -100.0% of factory value
z-axis self test: gyration trim within : -100.0% of factory value
MPU9250 initialized for active data mode....
AK8963 I AM FF I should be 48
AK8963 initialized for active data mode....
X-Axis sensitivity adjustment value 0.50
Y-Axis sensitivity adjustment value 0.50
Z-Axis sensitivity adjustment value 0.50
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = -470 my = -120 mz = -125 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -13.80, 0.00, 0.00
rate = 0.16 Hz
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = -470 my = -120 mz = -125 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -13.80, 0.00, 0.00
rate = 647.51 Hz

kriswiner, 2020-03-31:
Looks like the MPU9250 is not out of sleep mode, which is the mode on
initial power on.

The sensor is providing no data.

ahmedgamaleng, 2020-03-31:
maybe the problem here 
(AK8963 I AM FF I should be 48 )
what do you think ?

kriswiner, 2020-03-31:
Your MPU9250 is not "on", you need to start the PLL and put it into normal
run mode.

luizamanfrini, 2020-08-24:
Hey Kris, I think I am having the same problem here. How do I put it into normal run mode?

kriswiner, 2020-08-24:
Consult the data sheet and register map or look at the github sketches

// wake up device
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x00); // Clear sleep mode bit
(6), enable all sensors
  delay(100); // Wait for all registers to reset

 // get stable time source
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x01);  // Auto select clock
source to be PLL gyroscope reference if ready else
  delay(200);

Mustafakyuz, 2021-06-13:
Hi Kris, firstly thanks for codes and informations. I have same problem. I'm using arduino uno and mpu9250. And I added MadgwickQuaternionUpdate  function from quaternionFilters file (MPU9250BasicAHRS.ino). But I have a error that 'q' was not declared in this scope. I dont know what I changing in code. Please help me.
this is code which I work:

[basic_ahrstest1.zip](https://github.com/kriswiner/MPU9250/files/6644759/basic_ahrstest1.zip)

when  I try compare this code, ı have error which :

C:\Users\makyu\Desktop\basic_ahrstest1\basic_ahrstest1.ino: In function 'void MadgwickQuaternionUpdate(float, float, float, float, float, float, float, float, float)':
basic_ahrstest1:181:24: error: 'q' was not declared in this scope
             float q1 = q[0], q2 = q[1], q3 = q[2], q4 = q[3];   // short name local variable for readability
                        ^
C:\Users\makyu\Desktop\basic_ahrstest1\basic_ahrstest1.ino:181:24: note: suggested alternative: 'q1'
             float q1 = q[0], q2 = q[1], q3 = q[2], q4 = q[3];   // short name local variable for readability
                        ^
                        q1
basic_ahrstest1:195:33: error: 'q2' was not declared in this scope
             float _2q2 = 2.0f * q2;
                                 ^~
C:\Users\makyu\Desktop\basic_ahrstest1\basic_ahrstest1.ino:195:33: note: suggested alternative: 's2'
             float _2q2 = 2.0f * q2;
                                 ^~
                                 s2
basic_ahrstest1:196:33: error: 'q3' was not declared in this scope
             float _2q3 = 2.0f * q3;
                                 ^~
C:\Users\makyu\Desktop\basic_ahrstest1\basic_ahrstest1.ino:196:33: note: suggested alternative: 's3'
             float _2q3 = 2.0f * q3;
                                 ^~
                                 s3
basic_ahrstest1:197:33: error: 'q4' was not declared in this scope
             float _2q4 = 2.0f * q4;
                                 ^~
C:\Users\makyu\Desktop\basic_ahrstest1\basic_ahrstest1.ino:197:33: note: suggested alternative: 's4'
             float _2q4 = 2.0f * q4;
                                 ^~
                                 s4
basic_ahrstest1:252:61: error: 'beta' was not declared in this scope
             qDot1 = 0.5f * (-q2 * gx - q3 * gy - q4 * gz) - beta * s1;
                                                             ^~~~
C:\Users\makyu\Desktop\basic_ahrstest1\basic_ahrstest1.ino:252:61: note: suggested alternative: 'gets'
             qDot1 = 0.5f * (-q2 * gx - q3 * gy - q4 * gz) - beta * s1;
                                                             ^~~~
                                                             gets
basic_ahrstest1:258:27: error: 'deltat' was not declared in this scope
             q1 += qDot1 * deltat;
                           ^~~~~~
C:\Users\makyu\Desktop\basic_ahrstest1\basic_ahrstest1.ino:258:27: note: suggested alternative: 'delay'
             q1 += qDot1 * deltat;
                           ^~~~~~
                           delay
C:\Users\makyu\Desktop\basic_ahrstest1\basic_ahrstest1.ino: In function 'void loop()':
basic_ahrstest1:500:3: error: 'MahonyQuaternionUpdate' was not declared in this scope
   MahonyQuaternionUpdate(ax, ay, az, gx*PI/180.0f, gy*PI/180.0f, gz*PI/180.0f, my, mx, mz);
   ^~~~~~~~~~~~~~~~~~~~~~
C:\Users\makyu\Desktop\basic_ahrstest1\basic_ahrstest1.ino:500:3: note: suggested alternative: 'MadgwickQuaternionUpdate'
   MahonyQuaternionUpdate(ax, ay, az, gx*PI/180.0f, gy*PI/180.0f, gz*PI/180.0f, my, mx, mz);
   ^~~~~~~~~~~~~~~~~~~~~~
   MadgwickQuaternionUpdate
exit status 1
'q' was not declared in this scope

kriswiner, 2021-06-13:
Add the QuaternionFilter.ino to your sketch tab; put the two files in the
same sketch folder.

MoradSayed, 2023-03-14:
@Mustafakyuz 
Can you pass the working code, if you still have it 
Thanks in advance.