kriswiner/MPU9250 #412: Accelerometer bias

Mounir146, 2020-04-06:
Hey Kris

I am convinced that such a work has required from you a lot of efforts and time, so i want to thank you for sharing such a great work with us.

I still have plenty of questions, let's start by :

1- Did you figure out why the acceleration biases are not working for the MPU-9250 ?
2- Does the DMP inside the MPU9250 accomplish the 9Dof sensor fusion ?
3- What are the advantages of using a Dual MPU9250 ?

Thank you

kriswiner, 2020-04-06:
1. Yes, it has to do with the fact the LSbit in the accel offset register
is also a temperature compensation bit, and I couldn;t figure out how to
manage this properly. Others claim they do, but in the end it is more
convenient to simply manage this in the main loop anyway.

2. No, 6 DoF only. The newer ICM20948 does do this correctly using several
DMPs.

3. None.

Mounir146, 2020-04-07:
Thank you for the quick response !

Can you please tell me where did you exactly invoke the 6Dof DMP in your code ? and if you did, what's its role exactly ?

kriswiner, 2020-04-07:
I do not use the DMP ever.

Mounir146, 2020-04-09:
Thanks again for the quick response

Can you please tell me what is exactly the principle of the device self test ? And what is it used for ?

kriswiner, 2020-04-09:
Take look in the documents in the repository for these answers.

drankinatty, 2021-08-20:
Just a note on accelerometer X, Y, Z offset values (e.g. registers  `XA_OFFS_H` - `ZA_OFFS_L`). I could not make them work either. The Register Map documentation isn't that helpful on the issue. Preserving bit-0 of each low-byte register, saving the original content and adding the new offset, results in behavior that doesn't add a bias. Bit-0 of the low-byte will either be `0` or `1` -- which seems trivial to preserve and restore, but even writing `0` back to the offset registers, does not restore the original X, Y, Z acceleration values -- strange. The registers are additive in some way, because you cannot simply write the same values you obtain from the registers back either. Given this chip is out of production -- I'm not sure how much more effort is warranted here.

kriswiner, 2021-08-20:
There are some who have claimed to have fixed this and/or gotten it to work
properly. I wouldn;t waste any time on this since it is just as easy to
subtract the bias values in the main loop.

drankinatty, 2021-08-21:
Chuckling... I'm with you there. 3 floating-point additions is simple solution to guessing how Invensense actually intended to the accelerometer offsets to the calculated and stored. (but due to my failed attempts to set them, I did become much more familiar with `SIGNAL_PATH_RESET` and `SIG_COND_RST`...