kriswiner/MPU9250 #413: MPU9250 clone ?

iFrostizz, 2020-04-12:
Hi.

I bought this MPU-9250 module: https://www.ebay.fr/itm/MPU-9250-9-axes-Gyroscope-Accelerateur-magnetometre-MPU9250-1300Z/293106686659?_trkparms=ispr%3D1&hash=item443e851ac3:g:8~UAAOSwkLBc77qH&enc=AQAEAAACQBPxNw%2BVj6nta7CKEs3N0qXGRBquWmwR0g8541HJZFA%2BTerOzTshkq1bInEEMT9xhtbc5AUr%2FhlYg9S7jriVujLpTA1K%2BmVRripX6fNJT%2F45%2Bo9rLK%2By%2BFQ1IuNNwULO29wy0WMBqaKm0OePjqElXfEGQx%2B2vUmYP%2Fk%2BBsYVyq%2F7%2BoWxpnmtN6Ln0xt%2BBiS1CBTDy6eD11xTFOHHVslMR%2BjT3itIqIffvujPZpwBdHnLLxVypB1lj%2F8ijgkDVZFGXn%2BtJIOD2hkV9Aifql7MKks72m48pMBnpGYa7IHPEJG%2FZqP%2BEu2JebHZ9AJyT%2FShpw0axQAsZC4NrgqS9VDTsQAkJShyyrZ%2FOC2%2FeD4lB2GxZ%2F5JGB7rkZCXG5zzkW605tpXIOvS%2BE9Ye4Q3j92y2k%2FeC4r6NBkc8%2BOxy3Gwo4Vr11B4QMS9Z4XAO1R%2FherZk2%2B6pzFDzHhbb7hXsokRG9HSkIC5PhCtDnNkO%2FAR0A8fyc5Ww5taZx6fktqcg7VrdLZtRjcgSSoZ8ucgllrggArKJ7J6QslOGRryz244WHCzykhn8Q16iNeD1226fg8imkqnVpL3sqmsv2HP6kOsyP5FtHah0peOgCyMT152b2egC%2FyhNndYFMq2nJJvQDQ4TOTDuu3YRxBYi6ufbHZrVwal5RIuq1C2wZ9srwsIPjeH87tmYbxhRif%2BgFFf3mXzNyOh9z%2Bzlmp8TFL3mhwMyT56718tJiwZlzxJ1C31efO5VmkYrA85LueigX1ikt%2Bstw%3D%3D&checksum=293106686659de06800a7a304ba58a379c8b9499bc8e

Doesn't seems the cheapest, So that's why I've chosen it.

Here is my wiring:

MPU9250 ----> Arduino nano
VCC          ----> 5V (There is an on-board regulator)
GND         ----> GND
SDA          ----> A4
SCL           ----> A5

In the I2C scanner I get 0x68 adress, and using the MPU9250basicAHRS, I get this at start: 
MPU9250 I AM FF I should be 71
Could not connect to MPU9250: 0xFF

I'm confused and can't find any answer, even if it's not the first problem of this type here ...

Thanks a lot

kriswiner, 2020-04-12:
#define ADO 1
#if ADO
#define MPU9250_ADDRESS 0x69 // Device address when ADO = 1
#else
#define MPU9250_ADDRESS 0x68 // Device address when ADO = 0
#define AK8963_ADDRESS 0x0C // Address of magnetometer

#endif

Maybe try setting ADO to 0?

iFrostizz, 2020-04-12:
Thanks a lot ❤️ will try this asap and I'll tell you

iFrostizz, 2020-04-12:
It works ! Thanks a lot :)
Which beer do you prefere? :D

kriswiner, 2020-04-12:
Lagunitas Imperial Stout, if you can find it....

iFrostizz, 2020-04-12:
> Lagunitas Imperial Stout, if you can find it....
> […](#)

Seems very hard to find in France ... 
Also, I noticed the same thing than 8 monthes, when I learned to play with registers, on my board the roll and pitch registers are switched

closed 2022-02-26 by iFrostizz