kriswiner/MPU9250 #414: MPU9250BasicAHRS

labibbenali, 2020-04-14:
hey sir
i am using arduino pro mini , when i used MPU9250BasicAHRS it did not ask for waving sensor to calibrated , it directly spew data , and i my mag value is give always same  even i move sensor mx=-470 , my=-120 ,mz=125 . and when i use a sketch wich ask me to wave sensor for 15 sec , i get garbage value for magbias and get nan for magscale and after that give zero for three axis.
The ak8963 is 0xFF so i canged (d != 0x48) to (d != 0xFF)
need help plz
tanks

kriswiner, 2020-04-14:
MPU9250BasicAHRSis basic so no calibration, look at one of the later
sketches.

mx=-470 , my=-120 ,mz=125is mag bias for my particular sensor, put in your
values.

ak8963 is 0xFF  no idea, broken MPU9250?

labibbenali, 2020-04-14:
thanks for your response kris
when i run i get that value
it did not give me a raw data
so when i use other sketch witch have a magCalMPU9250 function to calibrated it show me that ak8963 is 0xFF.
 even i add this

#define ADO 0
#if ADO 
#define MPU9250_ADDRESS 0x69 // Device address when ADO = 1
#else
#define MPU9250_ADDRESS 0x68 // Device address when ADO = 0
#define AK8963_ADDRESS 0x0C // Address of magnetometer
#endif
and i tried with both ADO 0 and 1 
it's gievme akways 0xFF
so when i changed (d != 0x48) to (d != 0xFF) it give me me garbage value to magbias and nan for magscale  and mx,my and mz return always zero
there a longtime that i blocked in this issue

kriswiner, 2020-04-14:
Try just putting this in your sketch

#define MPU9250_ADDRESS 0x68
#define AK8963_ADDRESS 0x0C // Address of magnetometer

labibbenali, 2020-04-14:
MPU9250 I AM 0x71 I should be 0x71
MPU9250 is online...
x-axis self test: acceleration trim within : -0.1% of factory value
y-axis self test: acceleration trim within : 0.1% of factory value
z-axis self test: acceleration trim within : 1.9% of factory value
x-axis self test: gyration trim within : -3.2% of factory value
y-axis self test: gyration trim within : 0.0% of factory value
z-axis self test: gyration trim within : 0.8% of factory value
MPU9250 initialized for active data mode....
AK8963 I AM 0xFF I should be 0x48
Communication failed, abort!

that's what it give me :(

kriswiner, 2020-04-14:
When you scan for I2C devices, do you see the AK8963 on the I2C bus?

labibbenali, 2020-04-14:
you can give me a sketch for I2C i have one bu i am not sure about it
just to be 100% sure plz

kriswiner, 2020-04-14:
No

labibbenali, 2020-04-14:
i scan it and give me this:
Scanning...
I2C device found at address 0x68  !
done
and just repeat that

by the way this is the code for scanning maybe it's only for mpu9250 ID :
#include <Wire.h>
 
 
void setup()
{
  Wire.begin();
 
  Serial.begin(9600);
  Serial.println("\nI2C Scanner");
}
 
 
void loop()
{
  byte error, address;
  int nDevices;
 
  Serial.println("Scanning...");
 
  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
 
    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");
 
      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknow error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
 
  delay(5000);           // wait 5 seconds for next scan
}

kriswiner, 2020-04-14:
So if you do this after you initialize the MPU9250 and you don;t see the
AK8963 at 0x0C then your board is defective.

labibbenali, 2020-04-14:
i will put it after initMPU9250() and i will see what it will give me and i will tell you

labibbenali, 2020-04-14:
it show me only 0x68

MPU9250 I AM 0x71 I should be 0x71
MPU9250 is online...
x-axis self test: acceleration trim within : 0.3% of factory value
y-axis self test: acceleration trim within : -0.1% of factory value
z-axis self test: acceleration trim within : 2.3% of factory value
x-axis self test: gyration trim within : -3.6% of factory value
y-axis self test: gyration trim within : -0.3% of factory value
z-axis self test: gyration trim within : 0.5% of factory value
Scanning...
I2C device found at address 0x68  !
done

MPU9250 initialized for active data mode....
AK8963 I AM 0xFF I should be 0x48
Communication failed, abort!
soo my board is defective!!!!
so my only chance to buy another !!!

labibbenali, 2020-04-14:
i tried this code from other lib just to test and be sure and it gives me à value and when i move sensor value change
what's that mean ?
i really need help kris i'am starting to be crazy !

/*
Basic_I2C.ino
Brian R Taylor
brian.taylor@bolderflight.com

Copyright (c) 2017 Bolder Flight Systems

Permission is hereby granted, free of charge, to any person obtaining a copy of this software 
and associated documentation files (the "Software"), to deal in the Software without restriction, 
including without limitation the rights to use, copy, modify, merge, publish, distribute, 
sublicense, and/or sell copies of the Software, and to permit persons to whom the Software is 
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in all copies or 
substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING 
BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND 
NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, 
DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, 
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
*/

#include "MPU9250.h"

// an MPU9250 object with the MPU-9250 sensor on I2C bus 0 with address 0x68
MPU9250 IMU(Wire,0x68);
int status;

void setup() {
  // serial to display data
  Serial.begin(115200);
  while(!Serial) {}

  // start communication with IMU 
  status = IMU.begin();
  if (status < 0) {
    Serial.println("IMU initialization unsuccessful");
    Serial.println("Check IMU wiring or try cycling power");
    Serial.print("Status: ");
    Serial.println(status);
    while(1) {}
  }
}

void loop() {
  // read the sensor
  IMU.readSensor();
  // display the data
  
  Serial.print(IMU.getAccelX_mss(),6);
  Serial.print("\t");
  Serial.print(IMU.getAccelY_mss(),6);
  Serial.print("\t");
  Serial.print(IMU.getAccelZ_mss(),6);
  Serial.print("\t");
  Serial.print(IMU.getGyroX_rads(),6);
  Serial.print("\t");
  Serial.print(IMU.getGyroY_rads(),6);
  Serial.print("\t");
  Serial.print(IMU.getGyroZ_rads(),6);
  Serial.print("\t");
  
  Serial.print(IMU.getMagX_uT(),6);
  Serial.print("\t");
  Serial.print(IMU.getMagY_uT(),6);
  Serial.print("\t");
  Serial.print(IMU.getMagZ_uT(),6);
  Serial.println("\t");
 Serial.println(IMU.getTemperature_C(),6);
  delay(100);
}

kriswiner, 2020-04-14:
If you have this in your init function:

  // Configure Interrupts and Bypass Enable
  // Set interrupt pin active high, push-pull, hold interrupt pin level
HIGH until interrupt cleared,
  // clear on read of INT_STATUS, and enable I2C_BYPASS_EN so additional
chips
  // can join the I2C bus and all can be controlled by the Arduino as master
//   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x22);
   writeByte(MPU9250_ADDRESS,* INT_PIN_CFG, 0x12*);  // INT is 50
microsecond pulse and any read to clear

and see no AK8963 on the I2C bus after then you do not have a MPU9250 or
the board is defective or not properly designed.

kriswiner, 2020-04-14:
OK, if you don't undertand the sensor or the codes you are using then pick
the code that seems to work for you.

labibbenali, 2020-04-14:
that's what i have in my init function
writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x22);    
   writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x01);  // Enable data ready (bit 0) interrupt
   delay(100);
}

kriswiner, 2020-04-14:
If brian's code works for you, then  your sensor is OK and you just need to
figure out what is wrong with the basic sketch or use brian's sketch.

labibbenali, 2020-04-14:
i will check between the two lib and i will see
maybe he have a mistake in his lib
i appreciate and i really like your work that's why i am surprised , it suppose to work witth your sketch  if my board is ok , my bord is mpu9250 sparkfun
i really appreciate work
thak you very much for ur time