kriswiner/MPU9250 #415: MPU9250 Self test failed

sdrouet, 2020-04-23:
Hello Kris,

First of all thanks a lot for your great work.
I am currently using the sketch PU9250BasicAHRS and I am struggling with the Self test which is far away from what it should be

Here are the values:

x-axis self test: acceleration trim within : 1.7% of factory value
y-axis self test: acceleration trim within : -100.0% of factory value
z-axis self test: acceleration trim within : 2.0% of factory value
x-axis self test: gyration trim within : 20.5% of factory value
y-axis self test: gyration trim within : -1.7% of factory value
z-axis self test: gyration trim within : 0.6% of factory value
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
mag x min/max:
30
-1
mag y min/max:
263
232
mag z min/max:
179
147
Mag Calibration done!
AK8963 mag biases (mG)
25.01, 442.68, 281.63
AK8963 mag scale (mG)
1.02, 1.02, 0.96

Output of my MPU9250:
[
[image]
]

How can I get the calibration done and corrected in order to get true pitch,roll and yaw values?

Thanks in advance for your answer.

kriswiner, 2020-04-23:
Do you have this in your sketch?

magbias[0] = +470.; // User environmental x-axis correction in milliGauss,
should be automatically calculated
magbias[1] = +120.; // User environmental x-axis correction in milliGauss
magbias[2] = +125.; // User environmental x-axis correction in milliGauss

then remove this or use you particular settings.

How do you know the calibration is right/wrong?

sdrouet, 2020-04-23:
**For you first question:**
I had it but I removed it.
I added the function void magcalMPU9250(float * dest1, float * dest2) in order to get the magbias corrected with my value.
Unfortunately it didn't change a thing.

As you can see on the output I have:
AK8963 mag biases (mG)
25.01, 442.68, 281.63

**To answer your second question:**
From what I understand on the MPU9250SelfTest function it is written on comment: +/- 14 or less deviation is a pass. So I presume that mine is a failed one as I have more than that.

Furthermore when I move my MPU9250 it gives me really crap values....

kriswiner, 2020-04-23:
The self test is implemented incorrectly. If you look at the self test
document you will see that only the 8 LSbits are to be used. This is almost
never thre cause of poor results.

Poor calibration is responsible for 99% of the poor results people see.

On this topic, the calibration values you output do not seem to be correct

On Mz you see:

mag z min/max:
179
147

this is unlikely if you are moving the board in all possible directions and
not just spinning the board on the table, etc.

sdrouet, 2020-04-23:
So how can I implement it correctly?
Can I post my code and you pin point my mistake?

kriswiner, 2020-04-23:
Read the self test document and fix it yourself. Or look at some of the
later sketches, but really, self test is not that important. Your sensor is
fine.

I would pay more attention to your calibration, something not right.

sdrouet, 2020-04-23:
So on the calibrateMPU9250 function?
I changed nothing on that function so I don't understand why it works for others when it doesn't for me?

kriswiner, 2020-04-23:
Pilot error?

Check your results.

Collect a bunch of data on the serial monitor and plot Mx vs. My, My vs.
Mz, and Mx vs. Mz. Are the points circular and centered on the origin after
calibration?

https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration

sdrouet, 2020-04-23:
Thanks a lot I will take a look and investiguate

sdrouet, 2020-04-24:
Hello Kris,

I took a look on your link and indeed my MPU was uncalibrated. After some tests I manually changed the Mx, My and Mz value.
[image]

I also corrected my declination according to my location.

Unfortunately nothing really changed so far. I am using the MahonyQuaternionUpdate function btw.
Here what I got:

Flat surface:
[image]

Yaw with a 360 degree turn:
[image]

Pitch with a 360 degree turn:
[image]

Roll with a 360 degree turn:
[image]

if you have any clue it would be really appreciated.
Thanks again

sdrouet, 2020-04-24:
just to add a comment my Y-acceleration is always stuck at 2000 mg and I guess it is due to the y-axis self test: acceleration trim which is within : -100.0% of factory value as stated in my first comment. 

Is there any possibilities to change that value? If I am able to change it, will it solve the problem?

X-acceleration: 159.24 mg Y-acceleration: 1999.94 mg Z-acceleration: 1000.67 mg 
X-gyro rate: -0.092 degrees/sec Y-gyro rate: -0.069 degrees/sec Z-gyro rate: 0.031 degrees/sec
X-mag field: -256.12 mG Y-mag field: -35.88 mG Z-mag field: 393.73 mG
Temperature is 25.9 degrees C
X-acceleration: 158.45 mg Y-acceleration: 1999.94 mg Z-acceleration: 1002.08 mg 
X-gyro rate: -0.130 degrees/sec Y-gyro rate: -0.023 degrees/sec Z-gyro rate: 0.000 degrees/sec
X-mag field: -256.12 mG Y-mag field: -35.88 mG Z-mag field: 390.27 mG
Temperature is 26.0 degrees C
X-acceleration: 160.58 mg Y-acceleration: 1999.94 mg Z-acceleration: 1001.46 mg 
X-gyro rate: -0.069 degrees/sec Y-gyro rate: -0.038 degrees/sec Z-gyro rate: -0.015 degrees/sec
X-mag field: -248.98 mG Y-mag field: -50.21 mG Z-mag field: 383.36 mG
Temperature is 25.9 degrees C
X-acceleration: 157.53 mg Y-acceleration: 1999.94 mg Z-acceleration: 998.41 mg 
X-gyro rate: -0.069 degrees/sec Y-gyro rate: 0.023 degrees/sec Z-gyro rate: 0.023 degrees/sec
X-mag field: -266.84 mG Y-mag field: -46.63 mG Z-mag field: 390.27 mG
Temperature is 25.9 degrees C

kriswiner, 2020-04-24:
There is something very wrong with your accel values. I would take a look
at the raw bytes without any calibration or scaling applied and see if they
make any sense.

I wouldn't relay on the self test results, but accel values at rest of 2000
mgs is just wrong.

sdrouet, 2020-04-24:
Y-acceleration - raw value: 32767

So I don't understand what is wrong and why i have this value

I am using your function in order to get the raw value
void readAccelData(int16_t * destination)
{
  uint8_t rawData[6];  // x/y/z accel register data stored here
  readBytes(MPU9250_ADDRESS, ACCEL_XOUT_H, 6, &rawData[0]);  // Read the six raw data registers into data array
  destination[0] = ((int16_t)rawData[0] << 8) | rawData[1] ;  // Turn the MSB and LSB into a signed 16-bit value
  destination[1] = ((int16_t)rawData[2] << 8) | rawData[3] ;  
  destination[2] = ((int16_t)rawData[4] << 8) | rawData[5] ; 
}

kriswiner, 2020-04-24:
Looks like your MPU is busted. But what are the individual bytes,
rawData[2] and rawaDta[3]?

sdrouet, 2020-04-24:
rawData[3]: 255
rawData[2]: 127

kriswiner, 2020-04-24:
These should be ~0, so I would say your board is hosed. Try another one.

sdrouet, 2020-04-24:
fortunately i bought 2 of them. I come back to you with my result.

Thanks again for your time

sdrouet, 2020-04-24:
Second try with a second MPU9250....
And still bad results... I really don't know what I am doing wrong
**I got it perfectly calibrated.**
[image]

**Self test:**
x-axis self test: acceleration trim within : 0.6% of factory value
y-axis self test: acceleration trim within : -0.4% of factory value
z-axis self test: acceleration trim within : 4.2% of factory value
x-axis self test: gyration trim within : 69.3% of factory value
y-axis self test: gyration trim within : 20.9% of factory value
z-axis self test: gyration trim within : -100.1% of factory value

**On flat surface:**
X-acceleration: -760.86 mg 	Y-acceleration: -2.14 mg 	Z-acceleration: 994.69 mg 
X-gyro rate: -0.084 degrees/sec 	Y-gyro rate: 0.046 degrees/sec 	Z-gyro rate: 0.084 degrees/sec
X-mag field: 122.73 mG 	Y-mag field: 70.74 mG 	Z-mag field: 401.45 mG

X-acceleration raw value: -0.76

Dont know why the X-accel is so big when it should be close to 0.

Does it mean that my other MPU is broken?

sdrouet, 2020-04-24:
sorry the X accel raw value is X-acceleration raw value: -12479

kriswiner, 2020-04-24:
Maybe, why don;t you try to calibrate you accel and gyro...

sdrouet, 2020-04-24:
I finally got it works and it was definitely the X-accel which was wrong.

I don't understand why the calibrateMPU9250 didn't do its duty correctly. I had to correct it somehow manually
 accel_bias_reg[0] -= (accel_bias[0]/8) **-1400**; // Subtract calculated averaged accelerometer bias scaled to 2048 LSB/g (16 g full scale)
  accel_bias_reg[1] -= (accel_bias[1]/8);
  accel_bias_reg[2] -= (accel_bias[2]/8);

Do you have any idea why this function doesn't manage to make it correctly?

Thanks a lot again for your time

kriswiner, 2020-04-24:
Probably some error or other in the sketch you are using...or the way your
MCU interprets int16_t, etc