kriswiner/MPU9250 #417: Gyro Y and Z axis not reading

sfeek, 2020-05-26:
I am using a RPi and MPU-9250 set for address 0x69. I can read and calibrate all of the sensors except the Y and Z gyro which appear "stuck" no matter how many different ways the board is tilted. The raw values without calibration are -48.xxx for the Y and -2.xxx for the Z. The mag X,Y,Z and accel X,Y,Z and gyro X show normal values with movement. If I perform the calibration, the Y and Z gyro values stay near zero all the time. If I turn on the gyro Y axis calibration bit, I then can get some readings from it but nothing will make the Z axis work.

kriswiner, 2020-05-26:
Try another MPU9250 board....

sfeek, 2020-05-26:
That was what I was afraid of... thanks for your time.

kriswiner, 2020-05-26:
This is debugging 101.

If another board does the same thing it is your software. If not then,
well, you bought a cheap board that is now a drink coaster, etc...