kriswiner/MPU9250 #419: madgwick filter

29676, 2020-06-01:
Hi
I have a question about how to filter mpu9250 data. I read the raw data of the sensor, but I had trouble getting Quaternion. The output q is unstable at all, and the data of the roll, pitch, and yaw I obtained are constantly increasing. Most likely they are wrong. How can I solve this problem? If you can help, thanks

kriswiner, 2020-06-01:
This <https://github.com/kriswiner/MPU9250/tree/master/AK8963_as_slave>
sketch has methods that work well for this.

29676, 2020-06-02:
Thanks for the response. I wrote the program with atmega32 and in codvision , and to do this I repeated the madgwick filter 100 times using the  loop, then I printed q. Outputs are constantly increasing or decreasing and not stable . Can uncalibrated data cause quaternion unstability? I have listed quaternion data below .
This data is in a position where the sensor is fixed and has no movement.

thanks for your help.

q0=0.952220, q1=-0.100866, q2=-0.100129, q3=-0.264017
q0=0.946733, q1=-0.103507, q2=-0.101369, q3=-0.281663
q0=0.941104, q1=-0.105357, q2=-0.102238, q3=-0.298993
q0=0.935290, q1=-0.106757, q2=-0.102158, q3=-0.316268
q0=0.929400, q1=-0.107310, q2=-0.102020, q3=-0.333046
q0=0.923336, q1=-0.107924, q2=-0.101661, q3=-0.349420
q0=0.917256, q1=-0.107638, q2=-0.100741, q3=-0.365423
q0=0.911033, q1=-0.107042, q2=-0.099944, q3=-0.381050
q0=0.904832, q1=-0.106224, q2=-0.098741, q3=-0.396071
q0=0.898601, q1=-0.105470, q2=-0.097243, q3=-0.410565
q0=0.892194, q1=-0.104208, q2=-0.095513, q3=-0.425011
q0=0.885881, q1=-0.102844, q2=-0.093752, q3=-0.438720
q0=0.879550, q1=-0.101485, q2=-0.091820, q3=-0.451984
q0=0.873104, q1=-0.099661, q2=-0.090195, q3=-0.465026
q0=0.866777, q1=-0.097886, q2=-0.088298, q3=-0.477438
q0=0.860479, q1=-0.096014, q2=-0.086519, q3=-0.489385
q0=0.854239, q1=-0.094317, q2=-0.085004, q3=-0.500781
q0=0.848075, q1=-0.092283, q2=-0.083003, q3=-0.511849
q0=0.842057, q1=-0.090524, q2=-0.081202, q3=-0.522283
q0=0.836008, q1=-0.088794, q2=-0.079476, q3=-0.532463

kriswiner, 2020-06-02:
Yes, you have to calibrate your sensors other wise you will get garbage
output.

29676, 2020-06-02:
Should I enable fifo to calibrate the sensor or can it be calibrated without fifo?
Thank you

kriswiner, 2020-06-02:
either way

29676, 2020-06-02:
I have a few questions
I get compass data in two ways, but I don't know if they're right, how can I be sure?
And for the madgwick filter, how many times should the loop be repeated to make the correct outputs? 
If possible, help me calibrate the sensor.
Does the madgwick filter give stable outputs or Kalman? which one is better?
 How to get GyroMeasError values, and
 GyroMeasDrift?
 Thank you for your help

kriswiner, 2020-06-02:
" help me calibrate the sensor."

See sketches in the repository.

Madgick filter should be iterated 10 - 20 times per new data set, again,
see sketches, such as here
<https://github.com/kriswiner/MPU9250/tree/master/AK8963_as_slave>.

29676, 2020-06-03:
Hi ,
Thanks a lot for the response,
I have a problem with the calibration function
Do you get the values ​​of magBias and magscale from the magcalMPU9250 function?
values ​​of magCount (in the following function that is inside the main program)  are raw  data or the outputs of the magcalMPU9250 function?

   MPU9250.readMagData(magCount); 
  
      mx = (float)magCount[0]*mRes*magCalibration[0] - magBias[0];  
      my = (float)magCount[1]*mRes*magCalibration[1] - magBias[1];  
      mz = (float)magCount[2]*mRes*magCalibration[2] - magBias[2];  
      mx *= magScale[0];
      my *= magScale[1];
      mz *= magScale[2]; 
   }   
And another question, where is the output of the magcalMPU9250 function used in the main program ?

kriswiner, 2020-06-03:
Which main program are you using?

29676, 2020-06-03:
In this program, that madgwick filter is called and the roll, pitch and yaw are also calculated  
AK8963Slave0_MPU9250_Ladybug.ino

kriswiner, 2020-06-03:
line 159

29676, 2020-06-03:
So the values ​​of magbias and mag scale (in The following function) are obtained from the magcalMPU9250 function.
And mag count[0] ,mag count[1] , mag count[2]  are raw values magnet that get than readMagData function . it's true?

 MPU9250.readMagData(magCount);  
  
      mx = (float)magCount[0]*mRes*magCalibration[0] - magBias[0];  // get actual magnetometer value, this depends on scale being set
      my = (float)magCount[1]*mRes*magCalibration[1] - magBias[1];  
      mz = (float)magCount[2]*mRes*magCalibration[2] - magBias[2];  
      mx *= magScale[0];
      my *= magScale[1];
      mz *= magScale[2]; 
    }

29676, 2020-06-03:
Can you tell me how the values ​​GyroMeasDrift and GyroMeas Error are obtained? Should they be calculated by testing?

kriswiner, 2020-06-03:
These are irrelevant.

29676, 2020-06-03:
Sorry i don't understand.
Does this mean that the two are fixed in all program  that are used in the filter  madgwick or their value does not important?
thanks alot .

kriswiner, 2020-06-03:
Not important, verify this for yourself...

29676, 2020-06-03:
thanks.
Is it important to read the temperature register in sensor and  also  Gyro chip temperature? 

What are the benefits of temperature values?

29676, 2020-06-04:
Hi kris,
Thank you for your help
I have a question
I don't want to use fifo. If I don't use this feature, will there be a problem? I saw that it was used in the calibration function. Is it possible to calibrate the sensor without it?

kriswiner, 2020-06-04:
Don't need to use it. Just convenient and useful.

29676, 2020-06-05:
Hi kris,
 I saw in your program that you read Magnet data in two ways , which method 
 is better?
and , Why  the value ​​of  this  register read first , then written ?
uint8_t c = readByte(MPU9250_ADDRESS, ACCEL_CONFIG2);  
  c = c & ~0x0F; 
  c = c | 0x01; 
  writeByte(MPU9250_ADDRESS, ACCEL_CONFIG2, c); 

I will  replace this part of the code *  (it,s the below)  ،use loop (for (i = 0; i < 20; i += 1) ) for madgwick filter ,
 Is this wrong  or i can use the loop Instead of the below code?

* Now = micros();
    deltat = ((Now - lastUpdate)/1000000.0f); 
    lastUpdate = Now;

    sum += deltat; // sum for averaging filter update rate
    sumCount++;

and 
the function( accelWakeOnMotion( )  )
is important?
If I don't write it, what's the problem?

 thanks  a lot for your help

kriswiner, 2020-06-05:
This sketch uses the internal I2C bus to Tx mag data to the MPU6500 so that
all host reads are to MPU6500 registers. Tjis is best.

Bits to be written should be cleared first in such a way as not to change
the bits that are not to be written.

29676, 2020-06-05:
Ok,thanks,
but, why read data mag by slave is best?
and  i don,t use the function (accelWakeOnMotion( ) ) is it nead?

kriswiner, 2020-06-05:
fastest and no.

29676, 2020-06-11:
Hi kris ;
I get the raw values ​​of the sensor correctly" 
I wrote the calibration function
 And I made the settings like your program,
Now that I'm running the Madgwick filter, it gives the value q0= 1 and other quaternion to zero, which is the same initial values ​​I've defined for them.

I don't know what to do to solve the problem
Thank you for your help

kriswiner, 2020-06-11:
Plenty of examples in the repository to consult.

29676, 2020-06-11:
In some programs I saw that the input of the filter is in the form
 MadgwickAHRSupdate(gx, gy, gz, ax, ay, az, mx, my, mz);
and in some it is like this:
 MadgwickQuaternionUpdate(-ax, ay, az, gx*pi/180,-gy*pi/180,-gz*pi/180, -my,mx, mz);
How do I know which one is correct?

kriswiner, 2020-06-11:
Read Madgwicks paper.

29676, 2020-06-11:
ok ,thanks;
 and if i not write self test function and not call it in  main program has problem?

kriswiner, 2020-06-11:
No

29676, 2020-06-20:
Hello kris ;
Thank you for your answers and help

I have a few  questions:

1) I wrote the program with mega 32 and i2c with 100k frequency. Can low speed cause an error in the output of the madgwick filter?

2) What is the need to use RTC in the program? If not, is there a problem?

3) Is it better to write a program using spi or i2c?

4) What number should I use if I didn't use fifo in this part of the program?
instead( fifo_count)
packet_count = fifo_count/12;// How many sets of full gyro and accelerometer data for averaging

  for (ii = 0; ii < packet_count; ii++) 
   
thanks alot again

kriswiner, 2020-06-20:
1) I would use 400 kHz, but this alone will not cause an error. If the
fusion rate is much lower than the data sample rate, then you will get poor
results.

2) The RTC is specific to the STM32L4 and not necessary for the Madgwick
filter to work.

3) Either will do well but SPI is faster.  Unless you are running the
sample rates at 8000 Hz I2C is simpler.

4) If you don;t use the FIFO you will have to rewrite this calibration
function, it is designed for FIFO use.

29676, 2020-06-20:
Thank you
So if I use i2c with 100k, the output of the madgwick function is not wrong, only the speed is low, right?
And when I activate fifo, do I have to activate the general intrrupt for micro?

kriswiner, 2020-06-20:
The accuracy will depend on the speed.

So you want sensor sample rate of 200 - 400 Hz abd sensor fusion rate 10x
faster. Can you MCU do this? AVRs cannot...

29676, 2020-06-20:
What should I do to get a good output?
What should be the relationship between sampling rate and micro speed?
Can I get a good output with spi whit speed 1 meg? How much should I set the spi speed to? And in this case, how much should I set the sampling rate to?
Please help me more .thanks

kriswiner, 2020-06-20:
What MCU are you using?

Might want to get one of these
<https://www.tindie.com/products/tleracorp/ladybug-stm32l432-development-board/>

29676, 2020-06-20:
At mega32

29676, 2020-06-20:
Can't get a good output with this micro?

kriswiner, 2020-06-20:
8 bit, 8 MHz CPU, are you kidding? This will never work for sensor
fusion....

29676, 2020-06-23:
Hi kris 
What is the best amount of spi speed to be set with this 8-bit micro(mega32) to good work sensor fusion...?
Thank you so much for your help.

kriswiner, 2020-06-23:
8 bit 16 MHz...you are wasting your time with this...

29676, 2020-06-23:
Do you mean to use other micro?
What do you think I should do?
I don't have much experience, please help me
thanks

kriswiner, 2020-06-23:
https://www.tindie.com/products/tleracorp/ladybug-stm32l432-development-board/

Or buy a Teensy 3.2

29676, 2020-06-23:
Ok thanks Mr Kris;
But what is the reason that this micro  is not suitable for this job? 
Because of speed?
Or accuracy?
Or filter?
Or not stable output filter?
Roll and pitch and yaw obtained by this method
Not valid?
Thanks alot again for your answers and  your help.

kriswiner, 2020-06-23:
The fusion rate needs to be 10 - 20x the gyro sample rate, the gyro sample
rate is typically 200- 400 Hz for the MPU9250, so the MPU needs to be able
to iterate the Madgwick solution 10 - 20 times for every new gyro sample.
An 8 bit, 8 MHz AVR MCU cannot do this, best
<https://github.com/kriswiner/MPU6050/wiki/Affordable-9-DoF-Sensor-Fusion>that
it can do is ~100 Hz fusion rate.

29676, 2020-07-20:
Hello Kris;
Thank you very much for your help ;
I want to use stm32 for mpu9250  ;
Is it needed another tools ، besides Mic  for this job?
Should I use  special board/Piece  for calibration ?

kriswiner, 2020-07-20:
Why do you need a Mic?

29676, 2020-07-20:
I mean, other than Mcu stm32, do I need another board or device  for get output and calibration sensor?