kriswiner/MPU9250 #421: QuaternionFilter Unstable q values

beaumr2, 2020-06-12:
Hi Kris

First of all, thank you for this.

So I'm using your code and having trouble with the QuaternionFilter. The accelerometer, gyroscope, and magnetometer values are stable but my q values vary wildly (and seem to want to alternate between positive and negative) which is messing up my pitch, roll, and yaw.

Do you have any idea what might be causing this? I'm using Arduino Uno and your code. I can paste the code if needed but here's the output: 

x-axis self test: acceleration trim within : 0.7% of factory value
y-axis self test: acceleration trim within : -0.5% of factory value
z-axis self test: acceleration trim within : 2.3% of factory value
x-axis self test: gyration trim within : 34.6% of factory value
y-axis self test: gyration trim within : 14.5% of factory value
z-axis self test: gyration trim within : 0.8% of factory value
accel biases (mg)
-318.97
370.91
-494.32
gyro biases (dps)
-4.59
-9.87
-0.75
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
AK8963 mag biases (mG)
71.04
122.43
-36.90
AK8963 mag scale (mG)
1.01
1.03
0.96
Calibration values: 
X-Axis sensitivity adjustment value 1.16
Y-Axis sensitivity adjustment value 1.17
Z-Axis sensitivity adjustment value 1.13

ax = 0.73 ay = 0.37 az = 995.42 mg
gx = -0.03 gy = -0.23 gz = -0.17 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Gyro temperature is 34.0 degrees C
Yaw, Pitch, Roll: 13.80, 0.00, 0.00
Grav_x, Grav_y, Grav_z: 0.00, 0.00, 1000.00 mg
Lin_ax, Lin_ay, Lin_az: 0.73, 0.37, -4.58 mg
rate = 1.04 Hz

ax = 0.55 ay = -1.34 az = 995.30 mg
gx = 0.05 gy = 0.05 gz = -0.12 deg/s
mx = 27111 my = 164 mz = 35 mG
q0 = 0.64 qx = -0.00 qy = 0.00 qz = -0.77
Gyro temperature is 34.1 degrees C
Yaw, Pitch, Roll: 273.04, 0.00, -0.13
Grav_x, Grav_y, Grav_z: 2.33, 0.08, 1000.00 mg
Lin_ax, Lin_ay, Lin_az: -1.78, -1.43, -4.70 mg
rate = 4.87 Hz

ax = 376.28 ay = -1855.29 az = 494.32 mg
gx = 0.05 gy = 0.05 gz = -0.12 deg/s
mx = 27111 my = 164 mz = 35 mG
q0 = 0.51 qx = -0.33 qy = 0.60 qz = -0.52
Gyro temperature is 21.1 degrees C
Yaw, Pitch, Roll: 268.09, 16.15, -86.28
Grav_x, Grav_y, Grav_z: 958.51, 278.15, 62.35 mg
Lin_ax, Lin_ay, Lin_az: -582.23, -2133.44, 431.97 mg
rate = 4.87 Hz

ax = 376.28 ay = -1855.29 az = 494.32 mg
gx = 0.05 gy = 0.05 gz = -0.12 deg/s
mx = 27111 my = 164 mz = 35 mG
q0 = 0.41 qx = -0.17 qy = -0.18 qz = -0.88
Gyro temperature is 21.1 degrees C
Yaw, Pitch, Roll: 241.13, -26.40, 10.91
Grav_x, Grav_y, Grav_z: -169.48, -444.57, 879.57 mg
Lin_ax, Lin_ay, Lin_az: 545.76, -1410.72, -385.24 mg
rate = 4.87 Hz

beaumr2, 2020-06-12:
I should mention this is with the MPU9250 on a flat table and not moving, of course

kriswiner, 2020-06-12:
You are not getting any mag data, same corrupted output every cycle...

beaumr2, 2020-06-12:
I get mag data after the first few iterations I thought: mx = 27111 my = 164 mz = 35 mG

If not then what would be preventing me from getting mag data?

kriswiner, 2020-06-12:
our accel data is corrupted too:

ax = 376.28 ay = -1855.29 az = 494.32 mg

I would say your I2C API or conversion to int16_t is messed up. What is
your MCU?

beaumr2, 2020-06-13:
Arduino Uno

Not sure why it's corrupted? Here's the code I'm using. The quaternionfilter file is copied exactly from your website:

/*Demonstrate basic MPU-9250 functionality including parameterizing the register addresses, initializing the sensor, 
 getting properly scaled accelerometer, gyroscope, and magnetometer data out. 
 Addition of 9 DoF sensor fusion using open source Madgwick and Mahony filter algorithms. 
 Sketch runs on the 3.3 V Dragonfly STM32L476 Breakout Board.

 The BME280 is a simple but high resolution pressure/humidity/temperature sensor, which can be used in its high resolution
 mode but with power consumption of 20 microAmp, or in a lower resolution mode with power consumption of
 only 1 microAmp. The choice will depend on the application.
 
 SDA and SCL have 4K7 pull-up resistors (to 3.3V).
*/
 
#include "Wire.h"   
#include "MPU9250.h"
#include "RTC.h"

#define SerialDebug true   // set to true to get Serial output for debugging

// MPU9250 Configuration
// Specify sensor full scale
/* Choices are:
 *  Gscale: GFS_250 == 250 dps, GFS_500 DPS == 500 dps, GFS_1000 == 1000 dps, and GFS_2000DPS == 2000 degrees per second gyro full scale
 *  Ascale: AFS_2G == 2 g, AFS_4G == 4 g, AFS_8G == 8 g, and AFS_16G == 16 g accelerometer full scale
 *  Mscale: MFS_14BITS == 0.6 mG per LSB and MFS_16BITS == 0.15 mG per LSB
 *  Mmode: Mmode == M_8Hz for 8 Hz data rate or Mmode = M_100Hz for 100 Hz data rate
 *  (1 + sampleRate) is a simple divisor of the fundamental 1000 kHz rate of the gyro and accel, so 
 *  sampleRate = 0x00 means 1 kHz sample rate for both accel and gyro, 0x04 means 200 Hz, etc.
 */
uint8_t Gscale = GFS_250DPS, Ascale = AFS_2G, Mscale = MFS_16BITS, Mmode = M_100Hz, sampleRate = 0x04;         
float aRes, gRes, mRes;      // scale resolutions per LSB for the sensors
float motion = 0; // check on linear acceleration to determine motion
// global constants for 9 DoF fusion and AHRS (Attitude and Heading Reference System)
float pi = 3.141592653589793238462643383279502884f;
float GyroMeasError = pi * (40.0f / 180.0f);   // gyroscope measurement error in rads/s (start at 40 deg/s)
float GyroMeasDrift = pi * (0.0f  / 180.0f);   // gyroscope measurement drift in rad/s/s (start at 0.0 deg/s/s)
float beta = sqrtf(3.0f / 4.0f) * GyroMeasError;   // compute beta
float zeta = sqrtf(3.0f / 4.0f) * GyroMeasDrift;   // compute zeta, the other free parameter in the Madgwick scheme usually set to a small or zero value
bool wakeup;

// Pin definitions
int  intPin = 9;  //  MPU9250 interrupt

bool intFlag = true;
bool newMagData = false;

int16_t MPU9250Data[7]; // used to read all 14 bytes at once from the MPU9250 accel/gyro
int16_t magCount[3];    // Stores the 16-bit signed magnetometer sensor output
float   magCalibration[3] = {0, 0, 0};  // Factory mag calibration and mag bias
float   temperature;    // Stores the MPU9250 internal chip temperature in degrees Celsius
float   SelfTest[6];    // holds results of gyro and accelerometer self test

// These can be measured once and entered here or can be calculated each time the device is powered on
float   gyroBias[3] = {0.96, -0.21, 0.12}, accelBias[3] = {0.00299, -0.00916, 0.00952};
float   magBias[3] = {71.04, 122.43, -36.90}, magScale[3]  = {1.01, 1.03, 0.96}; // Bias corrections for gyro and accelerometer

uint32_t delt_t = 0;                      // used to control display output rate
uint32_t count = 0, sumCount = 0;         // used to control display output rate
float pitch, yaw, roll;                   // absolute orientation
float a12, a22, a31, a32, a33;            // rotation matrix coefficients for Euler angles and gravity components
float deltat = 0.0f, sum = 0.0f;          // integration interval for both filter schemes
uint32_t lastUpdate = 0, firstUpdate = 0; // used to calculate integration interval
uint32_t Now = 0;                         // used to calculate integration interval

float ax, ay, az, gx, gy, gz, mx, my, mz; // variables to hold latest sensor data values 
float lin_ax, lin_ay, lin_az;             // linear acceleration (acceleration with gravity component subtracted)
float q[4] = {1.0f, 0.0f, 0.0f, 0.0f};    // vector to hold quaternion
[... 185 lines cut ...]
    Serial.print(", ");
    Serial.print(a33*1000.0f, 2);  Serial.println(" mg");
    Serial.print("Lin_ax, Lin_ay, Lin_az: ");
    Serial.print(lin_ax*1000.0f, 2);
    Serial.print(", ");
    Serial.print(lin_ay*1000.0f, 2);
    Serial.print(", ");
    Serial.print(lin_az*1000.0f, 2);  Serial.println(" mg");
    
    Serial.print("rate = "); Serial.print((float)sumCount/sum, 2); Serial.println(" Hz");
    sumCount = 0;
    sum = 0;    
    delay(2000);
    }

}

//===================================================================================================================
//====== Set of useful functions
//===================================================================================================================

void myinthandler()
{
  intFlag = true;
}

kriswiner, 2020-06-13:
First thing to try is connect out bolded lines here:

*if( MPU9250.checkNewMagData() == true) { // wait for magnetometer
data ready bit to be set*
  MPU9250.readMagData(magCount);  // Read the x/y/z adc values

// Calculate the magnetometer values in milliGauss
// Include factory calibration per data sheet and user environmental corrections
  mx = (float)magCount[0]*mRes*magCalibration[0] - magBias[0];  // get
actual magnetometer value, this depends on scale being set
  my = (float)magCount[1]*mRes*magCalibration[1] - magBias[1];
  mz = (float)magCount[2]*mRes*magCalibration[2] - magBias[2];
  mx *= magScale[0];
  my *= magScale[1];
  mz *= magScale[2];

*}*

and why is this line commented out?

//intFlag = false;     // reset newData flag

shouldn't be...

beaumr2, 2020-06-13:
When I leave intflag = false in there it never loops through and keeps spitting out the same data over and over

kriswiner, 2020-06-13:
Then you are never getting an interrupt. Is pin 9 an interrupt pin on your
MCU?

beaumr2, 2020-06-13:
Honestly I don't understand the concept of an interrupt. I googled it and it said pin 2 was my interrupt. So I can change that in the code. Does pin 2 have to be wired to something on the MPU9250?

beaumr2, 2020-06-13:
Ah I connected it to the int pin on the MPU. Hold on

beaumr2, 2020-06-13:
Thank you for being patient with my stupidity. Let's try to start over. I pulled your exact code and just changed my intPin, date, and time. I believe there is still a problem as it never gets into this if statement:

    if(alarmFlag) { // update RTC output (serial display) whenever the RTC alarm condition is achieved  
       alarmFlag = false;

Everything up to that point seems alright. If I comment out that if statement it gives me a slew of data that is still not accurate. I'll repost the full code but like I said it should be nearly exactly yours

/* 07/6/2017 Copyright Tlera Corporation
 *  
 *  Created by Kris Winer
 * 
 Demonstrate basic MPU-9250 functionality including parameterizing the register addresses, initializing the sensor, 
 getting properly scaled accelerometer, gyroscope, and magnetometer data out. 
 Addition of 9 DoF sensor fusion using open source Madgwick and Mahony filter algorithms. 
 Sketch runs on the 3.3 V Dragonfly STM32L476 Breakout Board.

 The BME280 is a simple but high resolution pressure/humidity/temperature sensor, which can be used in its high resolution
 mode but with power consumption of 20 microAmp, or in a lower resolution mode with power consumption of
 only 1 microAmp. The choice will depend on the application.
 
 SDA and SCL have 4K7 pull-up resistors (to 3.3V).

 Library may be used freely and without limit with attribution.
*/
 
#include "Wire.h"   
#include "MPU9250.h"
#include "RTC.h"

#define SerialDebug true   // set to true to get Serial output for debugging

// RTC set time using STM32L4 natve RTC class
/* Change these values to set the current initial time */
const uint8_t seconds = 30;
const uint8_t minutes = 40;
const uint8_t hours = 17;

/* Change these values to set the current initial date */
const uint8_t day = 12;
const uint8_t month = 6;
const uint8_t year = 20;

uint8_t Seconds, Minutes, Hours, Day, Month, Year;

bool alarmFlag = false; // for RTC alarm interrupt

// MPU9250 Configuration
// Specify sensor full scale
/* Choices are:
 *  Gscale: GFS_250 == 250 dps, GFS_500 DPS == 500 dps, GFS_1000 == 1000 dps, and GFS_2000DPS == 2000 degrees per second gyro full scale
 *  Ascale: AFS_2G == 2 g, AFS_4G == 4 g, AFS_8G == 8 g, and AFS_16G == 16 g accelerometer full scale
 *  Mscale: MFS_14BITS == 0.6 mG per LSB and MFS_16BITS == 0.15 mG per LSB
 *  Mmode: Mmode == M_8Hz for 8 Hz data rate or Mmode = M_100Hz for 100 Hz data rate
 *  (1 + sampleRate) is a simple divisor of the fundamental 1000 kHz rate of the gyro and accel, so 
 *  sampleRate = 0x00 means 1 kHz sample rate for both accel and gyro, 0x04 means 200 Hz, etc.
 */
uint8_t Gscale = GFS_250DPS, Ascale = AFS_2G, Mscale = MFS_16BITS, Mmode = M_100Hz, sampleRate = 0x04;         
float aRes, gRes, mRes;      // scale resolutions per LSB for the sensors
float motion = 0; // check on linear acceleration to determine motion
// global constants for 9 DoF fusion and AHRS (Attitude and Heading Reference System)
float pi = 3.141592653589793238462643383279502884f;
float GyroMeasError = pi * (40.0f / 180.0f);   // gyroscope measurement error in rads/s (start at 40 deg/s)
float GyroMeasDrift = pi * (0.0f  / 180.0f);   // gyroscope measurement drift in rad/s/s (start at 0.0 deg/s/s)
float beta = sqrtf(3.0f / 4.0f) * GyroMeasError;   // compute beta
float zeta = sqrtf(3.0f / 4.0f) * GyroMeasDrift;   // compute zeta, the other free parameter in the Madgwick scheme usually set to a small or zero value
bool wakeup;

// Pin definitions
int  intPin = 2;  //  MPU9250 interrupt
int  myLed  = 13; // red led
[... 224 lines cut ...]
    Serial.print(", ");
    Serial.print(lin_az*1000.0f, 2);  Serial.println(" mg");
    
    Serial.print("rate = "); Serial.print((float)sumCount/sum, 2); Serial.println(" Hz");
    sumCount = 0;
    sum = 0;    
    }

} /* end of alarm handling */

}

//===================================================================================================================
//====== Set of useful functions
//===================================================================================================================

void myinthandler()
{
  intFlag = true;
}

void alarmMatch()
{
  alarmFlag = true;
}

kriswiner, 2020-06-13:
Does your Uno have an RTC?

beaumr2, 2020-06-13:
No it does not. I see the problem now. What MCU do you use? I'll buy that one unless there's an easy workaround for the RTC code

kriswiner, 2020-06-13:
You need something better than an Uno anyway if you have any hope of sensor
fusion rates fast enough for accuracy. Cheap, go with this
<https://www.tindie.com/products/tleracorp/ladybug-stm32l432-development-board/>.
But I would recommend this
<https://www.tindie.com/products/tleracorp/dragonfly-stm32l47696-development-board/>
.

Both open source so easy to customize, both programmable using the Arduino
IDE via USB.

beaumr2, 2020-06-13:
Thank you I just placed an order for both. I will update you when I get the boards. Thanks again!

beaumr2, 2020-06-18:
So I'm now using that Dragonfly STM32L47696 Board and using Pin 9 as the interrupt and having the same issue. The RTC flag isn't changing to true. I have input my current date and time

kriswiner, 2020-06-18:
" and having the same issue"

remind me what the issue is?

kriswiner, 2020-06-18:
Since you have a Dragonfly, try this:

https://github.com/kriswiner/Dragonfly/blob/master/MPU9250Optimized/MPU9250BasicAHRS3_Dragonfly.ino

beaumr2, 2020-06-18:
So I downloaded that and get a new error: 
exit status 1
no matching function for call to 'TwoWireEx::begin(int, int, bool)'

Weirdly it highlights this line of code:
Wire.begin(0, 400000, true); // set master mode, I2C frequncy at 400 kHz
  delay(1000);

I put the Wire folder in with your Dragonfly-Master. Is there a folder called TwoWireEx I need?

beaumr2, 2020-06-18:
I see the TwoWireEx in GrumpyOldPizza's Wire.cpp file. I downloaded the board from here:

https://github.com/GrumpyOldPizza/arduino-STM32L4

I'm thinking there's been an update that no longer likes his code for TwoWireEx::begin

kriswiner, 2020-06-18:
This looks like an old context for Wire, try this instead:

  Wire.begin(TWI_PINS_20_21); // set master mode
  Wire.setClock(400000);      // I2C frequency at 400 kHz

beaumr2, 2020-06-18:
Okay and that seems to work. Does this wiring sound good:

3V3->VCC
GND->GND
SDA->SDA
SCL->SCL
9->INT this is the one I'm really asking about. Based on your link here: https://github.com/kriswiner/Dragonfly or do I have to solder a wire to the back pad 41 that is INT? Then I might as well also solder to SDA and SCL

kriswiner, 2020-06-18:
I usually use pins 20 and 21 for I2C (SDA and SCL). You can use any other
pins for the interrupt. 8, 9, 10 are convenient but any pin will do. I
wouldn;t solder anything until you are sure everything is working as it
should.

Just make sure whichever pin you select as interrupt is specified in the
sketch.

[image: image.png]

beaumr2, 2020-06-18:
So many issues... didn't even get to test it. I had no serial port showing up even though it still showed with my Arduino Uno. Now when I plug in the Dragonfly STM32L476 the light on it doesn't even come one

kriswiner, 2020-06-18:
Is there a problem with your USB cable then? Do you have a multimeter? Can
you verify that there is 3V3 on the board?

If there is no power issue, close the Arduino IDE, plug in the USB cable,
press and hold boot, press and release reset, release boot. Then open the
Arduino IDE, select the proper board variant and flash a known good
program. This recovery procedure will fix problems usually due to trying to
flash with the wrong board variant selected.

What I don;t understand is I thought you said it was working, and now it is
totally fubared. What happened in between?

beaumr2, 2020-06-18:
It was working originally and then all of a sudden the port for it disappeared. I tried multiple USB cables and even plugged it into the wall and it doesn't light up. But I used a multimeter to the ground and 3.3V and measured 3.3V so I guess it's working?

Did all that and still no serial port. Like I said the serial port just disappeared. I was trying to mess around with FTDI drivers and CH340 based on what I saw online but to be honest this stuff is over my head and I don't really understand it. I'm operating on OS X Catalina. I also plugged in my arduino Uno and it discovered that port just fine

kriswiner, 2020-06-18:
Did you follow the reboot procedure I specified? Can you verify that the
Dragonfly is in STM32 boot loader mode before flashing?

beaumr2, 2020-06-18:
Just got it to appear again by moving around my FTDIusbSerial.kext file. Uploaded code fine even though there is no light and I got a new prompt to wave the board in a figure 8 to calibrate the magnetometer. Can't really do that with how my wires are currently set up and I have to leave for work. Tonight or tomorrow I'll solder some and update you. Thanks for everything! (And yes I tried your method, the blink file uploaded and worked even without the Serial port being there)

kriswiner, 2020-06-18:
FTDIusbSerial.kext file

Not sure what this is, a Mac thing?

beaumr2, 2020-06-19:
Yeah. And sorry I'm in the army and yesterday I got orders to move halfway across the country in <3 weeks. May be a little bit before I get back to this but I'll let you know when I do

beaumr2, 2020-06-30:
I got some free time and put it all together. Everything seems to be working! I do have some questions: what are my options as far as calibration goes? I'd like to not have to wave it in a figure eight to do the magnetic calibration every time I turn it on. I would like to permanently mount it in my vehicle. Maybe a way to save the calibration values?

kriswiner, 2020-06-30:
Save them in the sketch or on emulated eeprom.

beaumr2, 2020-07-22:
I think I finally got it all working. Now I have a new problem trying to run different code. I took some LCD display code from my Arduino Uno handbook. The LCD screen gets power but doesn't display anything. Does the liquid crystal library work with the dragonfly board? Or are the pins I'm choosing not digital outputs

Code:

#include <LiquidCrystal.h>
LiquidCrystal lcd(8, 7, 5, 4, 3, 2);

const int switchPin = 6;
int switchState = 0;
int prevSwitchState = 0;

void setup() {
lcd.begin(16, 2);
pinMode(switchPin, INPUT);

}

void loop() {
lcd.clear();
lcd.print("Pitch");
lcd.setCursor(0, 1);
lcd.print("Roll");
switchState = digitalRead(switchPin);
if (switchState != prevSwitchState) {
  if (switchState == LOW) {
    lcd.setCursor(0,0);
    lcd.print("Max values:");
    lcd.setCursor(0,1);
    lcd.print("Pitch and roll");
    delay(5000);
  }
}
prevSwitchState = switchState;
}

beaumr2, 2020-07-22:
Or is it possible the LCD screen wants 5V and the dragonfly only provides 3.3V

kriswiner, 2020-07-22:
I don't know the answer to your questions. I haven't used this display
before. I have used TFT displays with success using Adafruit's graphics
library

beaumr2, 2020-07-22:
Are pins 2-8 all digital output?

kriswiner, 2020-07-22:
yes

beaumr2, 2020-07-23:
And is there any way to get 5V from this board? I don't believe so based on the pinout I only see 3.3V

kriswiner, 2020-07-23:
Dragonfly is 3v3.

beaumr2, 2020-07-23:
Okay thanks! I'll get a 3V LCD

beaumr2, 2020-08-12:
Is this a normal amount of variance? This is with the Dragonfly and MPU9250 totally still, seems like it's accurate to within ~8 degrees:

Pitch & Roll: 5.81, -0.66
Pitch & Roll: 5.10, 6.83
Pitch & Roll: -2.27, 0.69
Pitch & Roll: 5.57, -3.60
Pitch & Roll: 0.37, -1.00
Pitch & Roll: -0.41, 0.63
Pitch & Roll: -4.34, 3.57
Pitch & Roll: -2.13, 2.08
Pitch & Roll: 1.52, 1.44
Pitch & Roll: 4.12, -0.08
Pitch & Roll: -0.42, -0.82
Pitch & Roll: -0.59, -5.62
Pitch & Roll: 1.99, 0.90
Pitch & Roll: -3.62, -1.53
Pitch & Roll: -0.76, 1.97
Pitch & Roll: 1.04, 2.14
Pitch & Roll: -1.76, 1.12
Pitch & Roll: 1.46, 1.07
Pitch & Roll: 2.63, 3.01
Pitch & Roll: 2.31, 1.15
Pitch & Roll: 3.60, -0.02
Pitch & Roll: -1.72, 6.03
Pitch & Roll: -3.40, 1.09
Pitch & Roll: 0.49, 0.72
Pitch & Roll: 3.77, -0.80
Pitch & Roll: 1.62, -0.09
Pitch & Roll: 1.73, 0.76
Pitch & Roll: 0.12, 1.48

kriswiner, 2020-08-12:
No. This is way bad.

How did you calibrate your accel and gyro? What sensor settings are you
using?

beaumr2, 2020-08-12:
So I notice after leaving it still for a while (~5minutes) it becomes steady:

Pitch & Roll: 2.52, 176.42
Pitch & Roll: 2.52, 176.45
Pitch & Roll: 2.50, 176.42
Pitch & Roll: 2.44, 176.34
Pitch & Roll: 2.41, 176.32
Pitch & Roll: 2.53, 176.43
Pitch & Roll: 2.39, 176.28
Pitch & Roll: 2.59, 176.53
Pitch & Roll: 2.39, 176.27
Pitch & Roll: 2.44, 176.31
Pitch & Roll: 2.55, 176.45
Pitch & Roll: 2.56, 176.47
Pitch & Roll: 2.52, 176.40
Pitch & Roll: 2.43, 176.32
Pitch & Roll: 2.47, 176.37
Pitch & Roll: 2.56, 176.47
Pitch & Roll: 2.50, 176.38
Pitch & Roll: 2.63, 176.56
Pitch & Roll: 2.42, 176.31
Pitch & Roll: 2.41, 176.31
Pitch & Roll: 2.41, 176.30
Pitch & Roll: 2.43, 176.32
Pitch & Roll: 2.53, 176.40
Pitch & Roll: 2.58, 176.44
Pitch & Roll: 2.56, 176.39
Pitch & Roll: 2.67, 176.51
Pitch & Roll: 2.67, 176.51
Pitch & Roll: 2.54, 176.37
Pitch & Roll: 2.41, 176.22

It asks me to calibrate by waving in a circle. I found that not waving it actually gave it better results. Not sure what you mean by sensor setting

kriswiner, 2020-08-12:
Sensor setting means full scale range, sample rate, bandwidth, etc.

Are you calibrating the gyro and accel? How?

beaumr2, 2020-08-13:
For sensor settings I'm using the following:

uint8_t Gscale = GFS_250DPS;
uint8_t Ascale = AFS_2G;
uint8_t Mscale = MFS_16BITS; // Choose either 14-bit or 16-bit magnetometer resolution
uint8_t Mmode = 0x06;        // 2 for 8 Hz, 6 for 100 Hz continuous magnetometer data read

To calibrate gyro and accel I'm using your calibrateMPU9250 function. I see this part of that function's code near the end is commented out. Should I uncomment it?

// Apparently this is not working for the acceleration biases in the MPU-9250
// Are we handling the temperature correction bit properly?
// Push accelerometer biases to hardware registers
//  writeByte(MPU9250_ADDRESS, XA_OFFSET_H, data[0]);
//  writeByte(MPU9250_ADDRESS, XA_OFFSET_L, data[1]);
//  writeByte(MPU9250_ADDRESS, YA_OFFSET_H, data[2]);
//  writeByte(MPU9250_ADDRESS, YA_OFFSET_L, data[3]);
//  writeByte(MPU9250_ADDRESS, ZA_OFFSET_H, data[4]);
//  writeByte(MPU9250_ADDRESS, ZA_OFFSET_L, data[5]);

kriswiner, 2020-08-13:
Seems fine.

How did you check the calibration?

What is your fusion rate?

beaumr2, 2020-08-13:
I did not check the calibration. What should those values be?

Also do not know what my fusion rate is or how to check that :/

kriswiner, 2020-08-13:
Sketch should output the fusion rate.

accel should be 0.0, 0.0, 1000 millig to within a few millig

gyro should be 0.0, 0.0, 0.0 dps to within a few millidps when the device
is flat and motionless...

beaumr2, 2020-08-13:
This is what I have:
Looks like the access biases are way off. That's with the device totally still, not even moving it when it asks for it to be waved in a figure 8. Don't see anything about a fusion rate. I'll post another output for when I wave it in a figure 8

MPU9250 is online...
x-axis self test: acceleration trim within : -0.2% of factory value
y-axis self test: acceleration trim within : -0.5% of factory value
z-axis self test: acceleration trim within : 1.5% of factory value
x-axis self test: gyration trim within : 33.5% of factory value
y-axis self test: gyration trim within : 13.8% of factory value
z-axis self test: gyration trim within : 0.4% of factory value
accel biases (mg)
-1726.07
120.54
-997.19
gyro biases (dps)
-4.50
-9.84
-0.52
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
Mag Calibration done!
AK8963 mag biases (mG)
183.27
1294.17
-220.05
AK8963 mag scale (mG)
1.08
1.08
0.88
Calibration values: 
X-Axis sensitivity adjustment value 1.16
Y-Axis sensitivity adjustment value 1.17
Z-Axis sensitivity adjustment value 1.13

beaumr2, 2020-08-13:
Ignore that last part I said, I realized it puts out calibration values for accelerometer and gyroscope before doing the figure 8 magnetic calibration

kriswiner, 2020-08-13:
I am asking what are the at-rest accel and gyro values after calibration.
They should be 0, 0, 1000 millig and 0, 0, 0 dps. Are they?

beaumr2, 2020-08-13:
Oh, yeah, they mostly are except the Accel z value is negative: 

ax = 0.98 ay = 1.89 az = -1004.76 mg
gx = -0.04 gy = 0.20 gz = 0.01 deg/s
mx = 6 my = 13 mz = 6 mG
q0 = 0.03 qx = -0.44 qy = 0.90 qz = -0.04
Pitch & Roll: 0.96, -174.92
Maximum Pitch & Roll
32.73,179.73
ax = 2.99 ay = -0.92 az = -999.82 mg
gx = -0.07 gy = 0.33 gz = 0.15 deg/s
mx = 4 my = -7 mz = -1 mG
q0 = -0.02 qx = -0.45 qy = 0.89 qz = -0.03
Pitch & Roll: -3.48, -178.32
Maximum Pitch & Roll
32.73,179.73
ax = 0.73 ay = -0.06 az = -1000.06 mg
gx = -0.01 gy = 0.34 gz = 0.08 deg/s
mx = 9 my = -1 mz = 0 mG
q0 = -0.01 qx = -0.50 qy = 0.87 qz = -0.01
Pitch & Roll: -1.45, -179.37
Maximum Pitch & Roll
32.73,179.73
ax = 3.17 ay = -0.55 az = -1001.16 mg
gx = -0.24 gy = 0.43 gz = 0.21 deg/s
mx = 11 my = -11 mz = 1 mG
q0 = 0.03 qx = -0.53 qy = 0.85 qz = 0.04

kriswiner, 2020-08-13:
Little more jitter than I would expect but good enough. Now try to
calibrate the mag. if you just need pitch and roll, then you should be
using a 6 DoF fusion algorithm that ignores the mag.

beaumr2, 2020-08-13:
When it's still after doing the figure 8 calibration the values are actually worse:

ax = -768.07 ay = 123.60 az = -695.25 mg
gx = 21.07 gy = 6.95 gz = 5.22 deg/s
mx = -223 my = 789 mz = -749 mG
q0 = -0.08 qx = -0.27 qy = 0.88 qz = 0.39
Pitch & Roll: 3.82, 133.26
Maximum Pitch & Roll
45.92,135.16
ax = -765.32 ay = 125.18 az = -695.07 mg
gx = 21.22 gy = 6.90 gz = 5.19 deg/s
mx = -207 my = 787 mz = -748 mG
q0 = -0.11 qx = -0.34 qy = 0.85 qz = 0.38
Pitch & Roll: 4.11, 133.08
Maximum Pitch & Roll
45.92,135.16
ax = -757.69 ay = 126.89 az = -698.12 mg
gx = 20.90 gy = 7.16 gz = 5.50 deg/s
mx = -214 my = 794 mz = -746 mG
q0 = -0.14 qx = -0.41 qy = 0.82 qz = 0.38
Pitch & Roll: 4.54, 132.92
Maximum Pitch & Roll
45.92,135.16
ax = -764.83 ay = 122.80 az = -698.97 mg
gx = 21.13 gy = 6.47 gz = 5.00 deg/s
mx = -248 my = 795 mz = -749 mG
q0 = -0.16 qx = -0.47 qy = 0.79 qz = 0.37
Pitch & Roll: 4.89, 132.89
Maximum Pitch & Roll
45.92,135.16
ax = -758.54 ay = 125.43 az = -700.99 mg
gx = 21.16 gy = 7.45 gz = 5.43 deg/s
mx = -214 my = 787 mz = -755 mG
q0 = -0.19 qx = -0.53 qy = 0.75 qz = 0.35
Pitch & Roll: 5.19, 133.03
Maximum Pitch & Roll
45.92,135.16
ax = -757.32 ay = 126.65 az = -703.12 mg
gx = 21.18 gy = 6.62 gz = 5.09 deg/s
mx = -236 my = 785 mz = -763 mG
q0 = -0.21 qx = -0.59 qy = 0.70 qz = 0.34
Pitch & Roll: 5.68, 132.98
Maximum Pitch & Roll
45.92,135.16
ax = -757.57 ay = 126.71 az = -701.35 mg
gx = 21.19 gy = 6.75 gz = 5.14 deg/s
mx = -229 my = 795 mz = -752 mG

kriswiner, 2020-08-13:
This makes no sense.

First you should let the sensor lie still to calibrate the accel and gyro,
then after this is done, the mag calibration requires you move the board in
all directions. If these are done in sequence, the gyro values should still
read near zero when the device is again at rest...

beaumr2, 2020-08-13:
That's what I'm doing. I let the sensor lie still on the table until it asks me to wave in a figure 8 for magnetic calibration. Then I wave it around until complete and then let it lie still again.

Maybe it's because it's not flat on the table? The soldering + wire prevents it from laying flat

beaumr2, 2020-08-13:
Okay so I think I may have realized the problem. The accel values change based on how it lays on the table (it's not exactly flat every time). So when it's calibrated initially it's calibrated for one specific orientation on the table and then if I put it down again it lays differently and gives different accel values. It's probably feeling the effects of gravity in different axes

beaumr2, 2020-08-13:
And you're saying the magnetic calibration is totally irrelevant to pitch and roll? Only affects yaw? So I can take that out completely?

kriswiner, 2020-08-13:
Yeah, but the problem you have is that after mag calibration, the gyro says
21 dps, which is just nuts for a motionless sensor. So something is
uber-wrong here...

beaumr2, 2020-08-13:
I did the procedure a few more times and am consistently getting gx, gy, and gz values close to zero when at rest. So that's good. Not sure what happened the other time

[30 more replies]

closed 2020-09-20 by beaumr2