kriswiner/MPU9250 #422: quaternionFilters.ino different from Madgwick's Internal Report

luizamanfrini, 2020-06-24:
Hi Kris, hope you are doing great!

I was looking into you quaternionFilters.ino file when I noticed that in your code h_z is not calculated. However, in Madgwick's Internal Report, h_z is calculated and taken into account when finding b_z. Can you explain me why you didn't consider it in your code?

Thank you for your help.

kriswiner, 2020-06-24:
I am sure it is in there, I probably took some short cuts to limit the
number of variables defined. If you find an error, please let me know...

luizamanfrini, 2020-06-26:
Hey Kris! It is definitely in there, you did take a short cut. Just realized it yesterday.
Another thing I was wondering is where did you use the zeta parameter in your code. It was defined but it was not used in quaternionFilters.ino for the gyro correction.

kriswiner, 2020-06-26:
I use neither zeta or beta. I think zeta is for the Mahony filter.

luizamanfrini, 2020-06-26:
Actually, Madgwick also uses zeta in his Internal Report to remove the gyro bias. However, since the gyro is calibrate in your main code, I don't know how important it is to consider zeta.

kriswiner, 2020-06-26:
I ignore it and get good results.

samu158820, 2020-11-20:
Hello Kris, hope you are fine.
I'm considering using your code to reproduce finger rotation on Unity...
I noticed your Madwick's filter does not use inverse square root, differently from he's code...
And also that the last updated MPU9250's code of yours is sending negative values to the quaternion.
Why use normal square root? And why send negative values to the quaternion filter? Can you explain me that, please?

kriswiner, 2020-11-20:
I use SQRTF on MCUs with an FPU. The Madgwick filter requires the sensor
data in NED order.

samu158820, 2020-11-21:
I see, thank you for your response. After some more researching, I managed to understand.
Your code seems to be the best around.
Best regards.