kriswiner/MPU9250 #423: Yaw angle stuck in the same value

luizamanfrini, 2020-06-29:
Hey Kris! I am trying to implement your code with an Arduino Uno. I know it is not the most appropriate board, however it is the only one I have with me right now. 

I am having good results with roll and pitch angles. However, my yaw angle is really weird. When I move the sensor, it shows some changes, however the value always returns to the same value, around 315 degrees, no matter which moves I make. Do you have any ideia of how to help me?

Thank you!

kriswiner, 2020-06-29:
Did you calibrate your sensors? How?

luizamanfrini, 2020-06-29:
Yes. Thats how a did for each sensor

- **magnetometer**: I used magneto12 . I collected the data by moving the sensor around in an 8 shape.

- **accelerometer**: I aligned each axis with the gravity direction and estimated a bias and scale, by taking the mean of the values in each direction.

- **gyroscope**: I moved the sensor to a known angle, and estimated it using the gyro iteratively. From that, I found the real w for each movement and estimated bias and scale.

kriswiner, 2020-06-29:
For the gyro, best approach is to collect data when the gyro is at rest and
motionless, then average and subtract the resulting offset biases from
subsequent data. Otherwise you are introducing  a large bias that will make
accurate heading estimation impossible.

Similarly, with accel, I would start with the same method. Accel flat and
motionless, collect data, calculate average offset from 0 and use to
correct the data.

Mag requires more care.

In each case it should be possible to check the calibration.

When the device is motionless, gyro should read 0.0, 0.0, 0.0 to within 1
millidegree per sec.

When the device axes are oriented with gravity the accel should read  (for
z-axis up) 0.0, 0.0, 1000.0 millig to within 10 millig or less. Try this on
several axes.

The mag should read the local magnetic field values for your location. For
me near San Francisco, this is 50, 240, and 450 milli G for x, y, and z
axes. If I place the device flat and motionless on the table, I see 450
milliG. If I flip the device 180 degrees so top is not bottom, etc I read
-450 mG within 10 mG. Then I know the device is calibrated.

Once calibrated, you have to feed the calibrated sensor data into the
Madgwick or Mahony filter in  NED order, so aN, aE, aD, etc

luizamanfrini, 2020-06-29:
Thank you so much! I am going to try this way and then I tell you what I get.

luizamanfrini, 2020-07-08:
Hey Kris! I've recently done the sensor calibration and I believe the results are good now. However, my yaw angle is still strange. When I do a 90 degree rotation, it shows only a 20 degree variation.

kriswiner, 2020-07-08:
Sounds like the mag is not properly calibrated. How did you check your
calibration?

luizamanfrini, 2020-07-08:
I plotted the mag's output before and after the calibration. After it, the results are way more spherical. 

here is my plot without calibration: https://drive.google.com/file/d/1RgPhC6sa5xBJtevTbVsD_8wm9lcxR9v4/view?usp=sharing
here is my plot with calibration: https://drive.google.com/file/d/1qVGKfrd4GEpqrgnYdAadbD1uiHoBxZmM/view?usp=sharing

kriswiner, 2020-07-08:
OK, how did you calibrate the gyro and accel?

luizamanfrini, 2020-07-08:
I did it the way you suggested up here. 
With gyro at rest, I took many measurements and calculeted an offset value for each axis. I did the same with the accel, considering that my Z axis should measure 9.87m/s^2

kriswiner, 2020-07-08:
Then I am not sure what the trouble is. What is your fusion rate?

luizamanfrini, 2020-07-08:
My sample frequency is 500Hz and I apply the Madgwick Algorithm 15 times per new data set

kriswiner, 2020-07-08:
There is something very wrong then.

If you put the sensor flat on a table what do the data look like?

Should be 0, 0, 0 for gyro, 0 0 1 g for accel and whatever your local mag
fields are for the mag, Mx, My, and Mz, but mag magnitude should be ~500
mGauss.

If you reverse the board so the top is now on the bottom, then what do the
data look like?

Should be 0 0 0 for gyro, 0 0 -1 for accel, and -Mz for mag.

Are you using NED for the data input into the Madgwick filter?

luizamanfrini, 2020-07-09:
My data is good for the accel and gyro, for both cases. However, there is something wrong with my mag. Its magnitude is way bigger than 500mg. I am going to check this out

luizamanfrini, 2020-07-09:
Just checked: I am getting a magnitude of 1780.0 mGauss

kriswiner, 2020-07-09:
Something very wrong then...

luizamanfrini, 2020-07-09:
Do you think this has something to do with calibration?

kriswiner, 2020-07-09:
Don't know. Check the conversion from bytes to int16, check the scale mag
data, etc