kriswiner/MPU9250 #425: Multiple 5 MPU9250 onto a Arduino

rlalpha, 2020-07-16:
Hello All,

Can I ask about the feasibility to connect 5 MPU9250 to one Arduino Mega/other Board?  Or, it is better to buy some alternative one?

Thank you for helping.

kriswiner, 2020-07-16:
Can be done fairly easily using I2C or SPI, see this
<https://github.com/kriswiner/MPU9250/tree/master/Dual_MPU9250>.

rlalpha, 2020-07-16:
Thank you for rapid response.
Can I ask about is that applicable to 5 MPU9250 cases?
like I can connect 5IMUs to 10 digital I/O port and read all 5 IMUs data

kriswiner, 2020-07-16:
Yes, you will have to toggle the ADO's so that you read from only one MPU
at a time. So at any one time, one MPU will have I2C address 0x68 and the
other will have 0x69, then toggle so that the next is 0x68 and the rest
0x69, etc. This is not very fast, but it can work, especially if you use
the FIFO properly..