kriswiner/MPU9250 #427: stabilisation YAW

Lacommunitedelinfo, 2020-07-23:
Hi, again thanks for your work.
I have a issue/question.
I'm interesseted on watching the YAW.
Its good (not a big drift) but when you move the sensor (for exemple 30°), the Yaw take a long time to stabilize to the good value (about 30-40 seconds). It is normal? I have a bug ?

kriswiner, 2020-07-23:
> thx for help

What is your fusion rate? I would guess that if you are working with an AVR
MCU it will be quite low. If you run the gyro sample rate at 200 - 400 Hz
you will need to iterate the fusion filter 10 - 20 times for each new gyro
data set. So you will need fusion rates of 2 - 10 kHz. Better use this
<https://www.tindie.com/products/tleracorp/ladybug-stm32l432-development-board/>or
something like it.

korbinianbauer, 2020-08-02:
Is your Magnetometer properly calibrated?
Even though I'm using an Arduino Mega, I get stable (+-1°) values after <0.5s again.

kriswiner, 2020-08-02:
Ok, sounds like everything is working then...

closed 2022-08-13 by hicode-and