kriswiner/MPU9250 #428: Unused zeta?

s4m0r4m4, 2020-08-01:
Hello! I was trying to figure out how zeta was used, and it looks like it is not currently used in the code: https://github.com/kriswiner/MPU9250/search?q=zeta&unscoped_q=zeta

The [Magdwick Paper](https://www.x-io.co.uk/res/doc/madgwick_internal_report.pdf) indicates on page 12 that zeta is a multiplier on a summation over s. I'm not sure if this corresponds to the equations circa line 69 of quaternionFilters: https://github.com/kriswiner/MPU9250/blob/2eecba9656c68c121a4cd31265716bec6c9d371b/quaternionFilters.ino#L69. Or perhaps you took the value of zeta to be zero and just removed the corresponding terms in the summation.

Anyway, thought I bring it up to see if anything was missing! Thanks for an awesome repo, I forked it to create a full-scope quad-copter project and am eternally thankful to your hard work here!

kriswiner, 2020-08-01:
I played with beta and zeta when I first started using this algorithm but
found almost no effect, so I have ignored it ever since. YMMV...

s4m0r4m4, 2020-08-04:
I see, thanks for clarifying and for the quick response!

closed 2020-08-04 by s4m0r4m4