kriswiner/MPU9250 #429: Euler Angles High Drift Rates

oozarslan15, 2020-08-26:
Hello Kris,

Thanks for the outsanding work. I am trying to obtain euler angles from quaternions with same sensor sampling rates. I am using 80Mhz esp8266 and updating madgwick ~10 times for 1 sensor reading. However, all euler angles are drifting at high rate. Sensor readings are taken at with 100ms interval. Those magnetometer plots are obtained with 8Hz magnetometer readings with 135 ms intervals. In order to calibrate, I took 200 samples with 8Hz sampling rate and 135 ms interval. 

[image]
[image]
[image]

Below graph is the magnetometer measurements.

[image]

That graph is the Roll - Pitch - Yaw. I am not even sure that problem is drift it seems like nonsense readings. Can you tell me what I am missing?
[image]

kriswiner, 2020-08-26:
What is the problem exactly?

Your sensor sample rates are rather low for capturing types of motion from
typical human activity. Why not use 200 - 400 Hz for gyro/accel and 100 Hz
for mag?

The sensor data need to be fed into the Mamdgwick or Mahoney filter in NED
order.

oozarslan15, 2020-08-26:
that euler angles and mag measurements are obtained when sensor is at rest. Problem is the same when I am doing calculations with 100 hz magnetometer. I am using NED order as your https://github.com/kriswiner/MPU9250/blob/master/MPU9250_MS5637_AHRS_t3.ino  script.

paulrossmanith, 2021-02-07:
Be sure to calibrate your IMU beforehand. I had a similar issue where the filter would drift indefinitly opposed to converging properly. 
Also try increasing the filters gain, beta=0.5 worked fine for my MPU9250 + Madgwick-Filter