kriswiner/MPU9250 #430: Teensy 3.2 and Mini

tpkendall, 2020-09-22:
Hello,

I'm pretty clueless about connecting all of this stuff. I'm getting the following errors:

MPU9250 I AM FF I should be 71 
Could not connect to MPU9250: 0xFF

I've seen your responses before and ran the I2C scanner and get "unknown error" for all addresses and "No I2C devices found."

I've attached a picture of my set up in case I am (very likely) doing something wrong.
[image]

kriswiner, 2020-09-22:
Not sure why you expect this to work at all.

Firstly, you will need to close the solder pads that enable the pullup
resistors on the board unless you already have some pullups on the I2C bus.

Then, simply laying the add-on board on top of the pin headers as you have
shown does not make for a good enough connection to the Teensy. Maybe
better to solder female pin headers on the MPU9250 add-on so you can place
it directly onto the Teensy as shown? Or use a breadboard and connect the
two using DuPont jumpers and J-hook test leads? But more secure connections
will surely be needed.

tpkendall, 2020-09-22:
As I indicated, I have absolutely no experience with this sort of activity, so thanks for your patience and response.

So when you say close the solder pads that enable the pullup resistors, does that mean I need to put solder over top of all three plates that are to the left of the GND port? I'm going with this because I don't know what "some pullups on the I2C bus" means (apologies).

I'll work on the more secure connections in the mean time.

kriswiner, 2020-09-22:
" put solder over top of all three plates that are to the left of the GND
port" Yes, so all three pads are in contact. This is a very old design;
subsequent design simply connect the pullups directly.

tpkendall, 2020-09-23:
I got it working, thanks to your response. Thanks for what you do!

I'm going to put an explanation and picture of what I ended up doing so people with a similar level experience to mine (zero) can have something to understand when they google it (since google brought me here a bunch). If you see anything wrong, please comment!
I followed Mr. Winer's instructions about connecting the three pads to the left of the GND port. For this configuration I also closed the two solder pads on the under side to access an alternate GND port. I then soldered the pin headers as seen in the picture. Initially I had the MPU9250 SCL/SDA connected to pins 18/19 on the Teensy which was producing the error:
MPU9250 I AM FF I should be 71 
Could not connect to MPU9250: 0xFF
I fixed this by instead connecting the MPU9250 SCL/SDA to pins 17/16 on the Teensy.
[image]

kriswiner, 2020-09-23:
Glad you got it working.

kriswiner, 2020-09-23:
I think there is another solder jumper on the back that lets you use the
3V3 pin at the board edge next to the GND. This was designed this way for
just this sort of breadboard use.