kriswiner/MPU9250 #431: MPU9250 Basic Example Code on BlackPill 411 Interrupt miss after an  indefinite period of time.

JoergSH, 2020-09-28:
Hello Kris,
nice that you are still active and try to help.
I'm just trying to get the MPU9250 Basic Example working. However, in the adapted version of Sparkfun because the functions are outsourced there. The development environment is VS Code with PlattformIO. The code runs on a Blackpill Stm32f411 with interrupt. The MPU delivers data by a rate of ca. 6000 Hz for an indefinite period of time, but then nothing comes. I've already tried that with different breakouts. So there is no longer any signal from the interrupt pin. Checked with oziloscope

kriswiner, 2020-09-28:
Not sure what to tell you. I use STM32L4 with the Arduino IDE and have no
trouble reading the MPU9250 with interrupt at 200 or 500 or 1000 Hz.
However, the interrupt is not designed to work at 6 kHz as far as I know.
Might, might not. The maximum accel sample rate is 8 kHz. I am not even
sure how to set it to 6 kHz. And I can not understand why such a high rate
would be necessary for most applications.

But in any case, I don't know anything about VS Code on PlatformIO, the
Black Pill, or Sparkfun's API so I really cannot help you here....

JoergSH, 2020-09-28:
Ok, I set up Arduino IDE for testing. Extract the display from the basic ahrs code and it runs stabel. 
q0 = 0.73 qx = -0.00 qy = 0.01 qz = 0.68<\r>
Yaw, Pitch, Roll: 72.35, 1.19, 0.65<\r>
rate = 1305.27 Hz
Roll is OK but Pitch and Yaw is not correct. 

What is t3 and MS5637?  MPU9250_MS5637_AHRS_t3.ino should be the best code to run You say in the redme.

kriswiner, 2020-09-28:
T3 is the Teensy 3.2 MS5637 is a baro.

1305 Hz is the sensor fusion rate. This can be changed by simply changing
the number fo iterations of the fusion filter per new gyro data sample.

You could try this one but I haven;t played with the MPU9250 in a while:

https://github.com/kriswiner/Dragonfly/tree/master/MPU9250Optimized

There is also this one:
https://github.com/kriswiner/MPU9250/tree/master/AK8963_as_slave