kriswiner/MPU9250 #433: Raspberry pi 4 multiple sensor with madgwick filter

zeeblaze, 2020-10-09:
Am trying to use the madgwick filter in this repo to get multiple (15) MPU9250 all placed on human body for motion sensing...is this a good filter?...and is the library in this repo compatible with raspberry pi?

zeeblaze, 2020-10-09:
Yes...am planning to use the TCA9548A i2c multiplexer

kriswiner, 2020-10-09:
Managing 15 MPU9250 is possible but I would worry about latency if you use
the method of toggling the I2C address as here
<https://github.com/kriswiner/MPU9250/tree/master/Dual_MPU9250>. The
alternative is to use an I2C or SPI multiplexer...

None of the sketches in this repo are designed for the RspB pi but could be
made to work with same with minor modifications.

zeeblaze, 2020-10-09:
The problem is is this library ok for cross platform build?
And am planning to use C++

zeeblaze, 2020-10-09:
Or if you can give a clue what will be added or removed to be compatible with Raspberry pi 4🙂

kriswiner, 2020-10-09:
No idea, maybe ask Simon Levy?

zeeblaze, 2020-10-09:
If this works on an Arduino...can I make it process the values locally and the Arduino locally and send it to raspberry pi via i2c with multiplexer?

kriswiner, 2020-10-09:
Maybe. You can certainly use something like a Ladybug or Teensy to collect
the data, and then how you get it to the pi is up to you. I2C/SPI or UART
all should work. But I don;t use the pi so really do not know for sure.

zeeblaze, 2020-10-09:
> Maybe. You can certainly use something like a Ladybug or Teensy to collect
> the data, and then how you get it to the pi is up to you. I2C/SPI or UART
> all should work. But I don;t use the pi so really do not know for sure.
> 
> 
> > If this works on an Arduino...can I make it process the values locally and
> > the Arduino locally and send it to raspberry pi via i2c with multiplexer?

The Arduino ladybug only supports spi...while the teensy supports i2c...how do I go about the Arduino support i2c?

kriswiner, 2020-10-09:
Ladybug
<https://www.tindie.com/products/tleracorp/ladybug-stm32l432-development-board/>
(or Dragonfly
<https://www.tindie.com/products/tleracorp/dragonfly-stm32l47696-development-board/>)
supports I2C, UART, SPI, and I2S.

zeeblaze, 2020-10-09:
I want a cheap one cause I need 15 of them😌...will stm328p work?

kriswiner, 2020-10-09:
What is it that you are trying to do?

You can read all 15 of the MPUs with a single Dragonfly board using I2C
(latency?) or SPI (better, lower latency). Why do you need 15?

zeeblaze, 2020-10-10:
> What is it that you are trying to do?
> 
> You can read all 15 of the MPUs with a single Dragonfly board using I2C
> (latency?) or SPI (better, lower latency). Why do you need 15?
> 
> 
> > I want a cheap one cause I need 15 of them😌...will stm328p work?

You mean I can process all 15 individual sensors from the ladybug and send it to the rpi?..cause the individual sensors will take the movement (rotation) of each body part they are connected to control bones of a 3d model.

kriswiner, 2020-10-10:
If you want to measure sensor data, then yes a single Dragonfly (Ladybug
has too few GPIOs) can read the data from 15 sensor boards and pass it via
I2C to an RPi. Lots of considerations including how to calibrate, select
each sensor, power usage for portability. This
<https://www.tindie.com/products/onehorse/chainable-10-dof-motion-sensor/#>
might be a useful solution.

If you can find a way to adequately calibrate each sensor, running the
sensor fusion on the Pi might get you to 4 degree heading accuracy. Pitch
and roll should be good to ~1 degree.

If you want absolute orientation at each point, I would use one of these
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-lsm6dsm-lis2md/>
or these
<https://www.tindie.com/products/onehorse/max32660-motion-co-processor/>
and simply send yaw, pitch and roll to the on-board SPI flash for storage
and upload to a pc, or send along to a Pi for further processing.

This avoids the need to do the fusion on the pi and will get you to ~2
degree heading accuracy with easier calibration in the case of the EM7180
and <1 degree heading accuracy with no calibration in the case of the
MAX32660 coprocessor solution..

Low cost is relative. If you want accuracy and ease of use, it will cost a
little more. But in any case, you just need the one Dragonfly.

zeeblaze, 2020-10-11:
Ok... thanks Mr kris