kriswiner/MPU9250 #434: Anomalous readings from MPU9250 with Feather MO

Birthon, 2020-10-20:
Hi Kris,
I'm trying to get MPU9255 to work with Feather MO using the sketch from:
https://media.digikey.com/pdf/Data%20Sheets/Sparkfun%20PDFs/MPU-9250_HookupGuide_Web.pdf
Everything seems to work but the results are completely abnormal, especially the Yall Pitch and Roll values ​​that fluctuate between positive and negative values ​​all the time. Having spent many hours working on it and doing a lot of research. However, I must confess that my lack of experience with 9-axis devices complicates things significantly. I tried two MPU9250 devices and both give me data with the same anomalies. There doesn't seem to be anything wrong with connection or addressing issues. I also do not believe that there is a failure of the code because the results of the serial monitor lead us to believe that everything is in order. From what I can see, many people have similar problems and the solution seems to converge on calibration. But I keep thinking that logically any calibration problems should give wrong results, but not fluctuating. I have no idea how to do a calibration and then use the results of the calibration inside the sketch, if the sketch code itself doesn't already do that. I really appreciate it if you can provide me with any help with this. Below is the result obtained from the serial monitor.

Thanks in advanced
[image]

kriswiner, 2020-10-20:
if these results are from the sensor sitting motionless flat on a table
then there is something seriously wrong with your gyro, or the code to
convert the gyro register data to scaled rotation data. With this
precision, the readings under these circumstances should all be 0.00
degrees per second.

You do have to calibrate the data, but calibration will not fix this large
gyro variation.

Are you using some kind of crappy Chinese sensors?

I would run the accel and gyro at least at 200 Hz with a low pass filter of
41 Hz. Are you?

How are you reading the data, via I2C?

Birthon, 2020-10-20:
First, thanks for your prompt reply. Trying to answer your questions;

- Yes, the data were obtained with the sensor sitting still and on the table.
- It is very likely that they are Chinese, I am not sure if they are of low quality, please look at the photo. What do you think?
- What is the procedure to run the accel and gyro at least 200 Hz at a low pass filter of 41 Hz. What exactly should I do?
- Yes I am reading data via I2C

Thanks
[image]

kriswiner, 2020-10-20:
These should work well enough. Look here
<https://github.com/kriswiner/MPU9250/tree/master/AK8963_as_slave> or here
<https://github.com/kriswiner/ESP32/tree/master/MPU9250_MS5637_AHRS_UDP>
for the how.

Birthon, 2020-10-21:
At the moment I only have Feather MO, Arduino Mega and Arduino Uno boards. None of them seems to match the sketchs you suggested, several errors appear in the compilation. So unless you have any other suggestion I need to wait some time before getting an ESP32. Anyway thanks for your help.

kriswiner, 2020-10-21:
I don't use these MCUs so have no specific sketch for them. You will have
to make mods to suit whichever one you choose. Better off with the ESP32
since the higher CPU speed translates into better heading accuracy.