kriswiner/MPU9250 #435: Stabilization of Yaw

brrnd99, 2020-10-21:
Hi, im new to MPUs and im currently working on improving the movement of robot so that it could move straightly and turns 90 degree when an obstruction is detected, in this case I will only nid to observe yaw. Im using this with arduino due, but was confused on the selection of code from ur library, cn i know which sketch is more suitable to be used and how what others library should be included too, already had i2c library tho.

kriswiner, 2020-10-21:
Could start from here
<https://github.com/kriswiner/MPU9250/tree/master/AK8963_as_slave> or maybe
easier here
<https://github.com/kriswiner/ESP32/tree/master/MPU9250_MS5637_AHRS_UDP>.

brrnd99, 2020-10-21:
already attempted the mpu9250basicahrs_12c code and it shows mpu failed to connect, tried the wake up code but seems like im lacking of some library.
[image]
[image]

kriswiner, 2020-10-21:
Yup, you are not connecting. What does an I2C scan show?

brrnd99, 2020-10-21:
i2c scanner show device found at 0x68

kriswiner, 2020-10-21:
Are you using 0x68 as your MPU9250 I2C address?

This doesn't look like one of my sketches. Your I2C "library" is probably
fubar...

brrnd99, 2020-10-21:
Im using this I2C library below, cn u attached ur version of i2c scanner here please...
#include <Wire.h>
 
 
void setup()
{
  Wire.begin();
 
  Serial.begin(9600);
  while (!Serial);             // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner");
}
 
 
void loop()
{
  byte error, address;
  int nDevices;
 
  Serial.println("Scanning...");
 
  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
 
    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");
 
      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
 
  delay(5000);           // wait 5 seconds for next scan
}

kriswiner, 2020-10-21:
Your scanner sees fine.

Why not use one of the sketches I linked to?

brrnd99, 2020-10-21:
> Could start from here <https://github.com/kriswiner/MPU9250/tree/master/AK8963_as_slave> or maybe easier here <https://github.com/kriswiner/ESP32/tree/master/MPU9250_MS5637_AHRS_UDP>.
> […](#)

These?

kriswiner, 2020-10-21:
For example.

Are you using 0x68 as the MPU9250 I2C address? How do you know?

brrnd99, 2020-10-21:
im not sure, working on it now( i2c address ).. is the adafruit library in ur sketch working fine with arduino due ?

kriswiner, 2020-10-21:
Never used the arduino due, don't know what this is.

What adafruit library?

If you are using adafruit's MPU9250 library, ask them why it is not
working...

brrnd99, 2020-10-21:
For the adafruit thing sorry i overlooked the others code, so does ur sketch ( attached jusnow) includes all the function like i2c scan, main function and library required?

kriswiner, 2020-10-21:
Yes, start with this
<https://github.com/kriswiner/ESP32/tree/master/MPU9250_MS5637> one, it
should require minor mods to get it to work...if you don;t have an MS5637
baro, you can comment this stuff out.

brrnd99, 2020-10-21:
im trying out with arduino uno, does the esp32 code works find with uno?

kriswiner, 2020-10-21:
You'll need to use this
<https://github.com/kriswiner/ESP32/blob/master/MPU9250_MS5637/MPU9250_MS5637_AHRS.ino>
sketch, and add this
<https://github.com/kriswiner/ESP32/blob/master/MPU9250_MS5637/quaternionFilters.ino>
in the tab for the fusion functions, comment out the MS5637stuff...

kriswiner, 2020-10-21:
You won't get any yaw accuracy with an AVR MCU, too slow
<https://github.com/kriswiner/MPU6050/wiki/Affordable-9-DoF-Sensor-Fusion>.

But if you make the proper mods, the sketch should at least run.

brrnd99, 2020-10-21:
but what if my whole robot already using arduino due, this means i cnt get yaw reading at all? what if i change to mpu6050 then? Any way to solve if arduino uno must be used..

kriswiner, 2020-10-21:
In this case, I would use this
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-lsm6dsm-lis2md/>,
or this
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-mpu9250/>,
or even this
<https://www.tindie.com/products/onehorse/max32660-motion-co-processor/>
for best accuracy.

brrnd99, 2020-10-21:
ok i will try it, cn i ask here further more in future?

kriswiner, 2020-10-21:
Ask me specific questions at tleracorp@gmail.com.