kriswiner/MPU9250 #436: Multiple MPU latency problem

hedayat95, 2020-10-25:
Hi Kris,
First of all thank you for this amazing code! I want to read 12 mpu9250 with teensy3.2 and I'm worried about latency and fution rate! I have some questions and I would really appreciate if you help me.
1. do you think I can read 12 sensors with this MCU and achieve 1 KHz  fution rate?
2. The thing that I can't actually understand  is how i2c multiplexer can solve the latency while the bus is serial( can I read all sensors at the same time?). can you explain it ?
3.I don't have SPI multiplexer and it isn't also possible to buy it, but I have i2c multiplexer my question is which method is faster , using i2c multiplexer or SPI without multiplexer?
4. I want to calibrate sensors individually and use calibration amount as const for every time. dose it make a problem?
5. I want to measure the finger joint angles when the hand is parallel with gravity and do grasping . if I use Tait-Brayan angles the Yaw is desired angle ( finger flextion) when roll is 90 degree. in your opinion for this task which Euler rotation sequence is better?
6. I use customize breakout board and used 10K pullup resistor for SDA and SCLK  in every breakout board. when I read 12 sensors this resistor  be parallel. dose it make problem?

kriswiner, 2020-10-25:
>    3. I want to calibrate sensors individually and use calibration amount
>    4. I want to measure the finger joint angles when the hand is parallel
>    5. I use customize breakout board and used 10K pullup resistor for SDA

1) you can read 12 MPU9250 but a fusion rate for all 12 at 1 kHz will be
quite difficult using the Teeny 3.2.
2)  You don't need an I2C multiplexer but using one would make the software
much easier and probably faster.
3) would probably work fine.
4)  yaw, pitch, roll is the same as Euler angles. Not sure I understand
your question.
5) depend on the length of the connecting wires. But 12 x 10 K in parallel
is going to be ~1K or so, you'll just have to try it.

hedayat95, 2020-10-25:
Thank you for the quick response ! 
Do you think I can get sufficient accuracy with this MCU for hand movement ? 
Which method do you recommend? SPI or i2c
When the order of rotation is changed subsequently the roll and etc.  will change  . I think depend on initial position which order is better . imagine the initial position of breakout board is perpendicular to the horizon plane. which order do you recommend?

hedayat95, 2020-10-25:
in each finger  the length of the connecting wires is 10 cm and total length of bus is 50cm.

kriswiner, 2020-10-25:
Probably better off using these
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-mpu9250/>
with an I2C multiplexer. Or this
<https://www.tindie.com/products/onehorse/max32660-motion-co-processor/>.

hedayat95, 2020-10-25:
another problem is when I soldered the mpu9250 on my board, the accel data became saturated, and I believe it is because of the bias shift due to high temperature during soldering process. I tried to write zero (0x00) to accel offset register, but this approach didn’t solve my problem. but when select 16_G  for ACCEL the problem was fixed

kriswiner, 2020-10-25:
Should be able to calibrate out this bias offset.

hedayat95, 2020-10-25:
I am not able to buy this chips due to Corona .(there aren't in my Country ).  again Do you think I can get sufficient accuracy with this MCU for hand movement ? or I waste my time

kriswiner, 2020-10-25:
Waste your time

hedayat95, 2020-10-25:
Thank you very much.

hedayat95, 2020-10-25:
may I read all sensors  and run the filter in labview that work with the CPU clock?

kriswiner, 2020-10-25:
Maybe...I would try with one or two first