kriswiner/MPU9250 #437: Insights on Madgwick Filter

nodeMike, 2020-10-27:
Hello!
First, I would like to express my gratitude for all the great work you have done! It made it much easier for me to start working with imu, for which I am grateful!

Im working on different hardware than mentioned in this repo, however my question is pretty generic and I hope this is a good place to ask it. My question is strictly related to fusion filter itself, rather than hardware/firmware. But first, I will say few words about my setup. 

My device is getting ACC/GYRO/MAG readings, then executes madgwick filter (original implementation from the paper). Quaternion is then sent via serial, and visualized on PC side. Everything works fine, quaternions are correct, there are no drift (when device is put motionless on the table). I've implemented bias corrections, and simple hard/soft iron calibration (based on your works). You're probably wondering why I'm bothering you since everything works, but there is a thing.

Im not really sure how to parametrize my program, to achieve best results. I were tinkering with it for days now, but still not sure, if my results are really best. 

My program goes like this in loop:

- [660us] Get raw IMU data from sensors
- [90us] Execute madgwick filter
- [270us] Transmit data

So entire cycle takes ~1ms so I'm able to get data, update filter and transmit it with 1kHz. Sensors ODR is 2kHz for ACC/GYRO and 100Hz for Mag. From what I've understood, filter algorythm should be executed on the same set of data multiple times, so far this is not met since ODR is 2kHz and loop goes with 1kHz. Also, there is no reason for me to transmit quaternions at this rate, and I can go down to 300Hz for my application.

After this longish introduction, I would like to ask, what would you be your approach to optimize results? Reducing ODR, and executing madgwick filter multiple times on the same set of data? Considering desired quaternion output rate at 300Hz I have ~3000us of time for sensor data receive and process, so its enough to, e.g.:

- get sensor data 3 times, and execute madgwick filter 3 times for every "fresh" data,
- get sensor data 2 times, and execute madgwick filter 9 times for every "fresh" data,
- get sensor data 1 time, and execute madgwick filter 25 times.

I would really appreciate any opinions on this topic :) Thank you for taking the time to read this post!

kriswiner, 2020-10-27:
The question is, what kind of heading accuracy are you getting with the
current set up and is it enough for your application?

nodeMike, 2020-10-28:
The main issue is, that I'm not sure how to measure it's accuracy. I dont have any special tools for this purpose and all I could do so far is try to rotate device with fixed angle and compare the results on screen, however this is not very consistent and precise method... Any ideas how should I approach this problem?

nodeMike, 2020-11-01:
Mr. Kris Winer, do you have any advices on the subject? Maybe you could suggest some sources for me to read on this topic ? :)

kriswiner, 2020-11-01:
We constructed
<https://hackaday.com/wp-content/uploads/2019/03/hackaday_journal-gregorytomasch_kriswiner-heading_accuracy_using_mems_sensors.pdf>
a 2-D goniometer from a used Zeiss rotation stage and use this for more
precise heading accuracy estimation. In the end you need to test the device
in its application anyway. So for a quadcopter, does it fly well, for a
boat, does it agree with the ship's compass, etc.

The ingredients for successful (accurate) AHRS estimations are in order of
importance:

1) pick accurate sensors (LSM6DSM, MMC5983A, for example)
2) calibrate them properly (see here
<https://github.com/gregtomasch/EM7180_SENtral_Calibration> for example)
...
99) sensor fusion algorithm.

nodeMike, 2020-11-02:
That's a good advice. Currently I'm using rather accurate sensors, so all what's left to do is test device in its application :)

closed 2020-11-02 by nodeMike