kriswiner/MPU9250 #439: UNO with 9250 'q was not declared in the scope' error

DroneUniversity, 2020-11-03:
Hi,

is anyone getting htis error when compiling the MPU 9250 basic script?

kriswiner, 2020-11-03:
Might try adding the Madgiwck filter sketch to the tab in the Arduino IDE.

DroneUniversity, 2020-11-04:
Hello SIr,

I did attempt that but the code does not compile and gets stuck at some point. Is this code available as a whole somewhere I could download please?

I also tried using the example code from SparkFun page but I cannot get the code to execute in its entity and more specifically the part that ca;calculates yaw, pitch, roll does not seem to execute at all. I am trying to evaluate yaw displacement for a project of mine. I have also added magnetometer settings for my region (4.6).

Thank you very much for your immediate support on this.

kriswiner, 2020-11-04:
Only sketches available for the MPU9250 are in this repository.

DroneUniversity, 2020-11-04:
Any ideas wht the code shown in the Spark Fun page (https://learn.sparkfun.com/tutorials/mpu-9250-hookup-guide/all), does not execute the yaw calculation section? I am using a 9250

kriswiner, 2020-11-04:
Maybe ask Sparkfun?

DroneUniversity, 2020-11-04:
It is your code I meant listed on their page :)

kriswiner, 2020-11-04:
Not sure what MCU you are using but have you tried any of the sketches here
<https://github.com/kriswiner/MPU9250>? You might have to modify for your
particular board but this is straightforward.

Maybe this <https://github.com/kriswiner/ESP32/tree/master/MPU9250_MS5637>
one for the ESP32?

Maybe this
<https://github.com/kriswiner/nRF52832DevBoard/blob/master/MPU9250_BMP280_BasicAHRS_nRF52.ino>
one for the nRF52?

Maybe this
<https://github.com/kriswiner/Dragonfly/tree/master/MPU9250Optimized>
simpler one for the STM32?

I doubt you will find the perfect sketch just right for your setup on my
repo, but you can find working sketches that can be easily modified to suit.