kriswiner/MPU9250 #440: Code running on UNO and Spark Fun 9250 not executing the part for calculating yaw, pitch, roll values

DroneUniversity, 2020-11-04:
Hello everyone ,

I have attempted to execute the code as shown in the SPark Fun page (https://learn.sparkfun.com/tutorials/mpu-9250-hookup-guide/all) and I notice that the part of code for calculating yaw, pitch, roll is not executed actually.

Has anyone come across this issue before?

Thank you

DroneUniversity, 2020-11-05:
Any ideas anyone?

luke-ff, 2020-11-17:
as mentioned on the SPark Fun page, you'll have to change some settings:
`#define AHRS true         // Set to false for basic data read`

DroneUniversity, 2020-11-23:
Hi there,

Thank you very much for your reply. I have the AHRS turned off and tun the Sparkfun code but never get any reading regarding Yaw, Pitch, Roll:

Actually not event the Serialprint("Yaw, Pitch, Roll: ") text is not shown in the Serial Monitor output.

Any ideas?

Thank you again

Spyros