kriswiner/MPU9250 #441: MPU-9250 on ESP8266

luke-ff, 2020-11-17:
I've troubles getting valid readings from MPU-9250. I'm using MPU9250BasicAHRS.ino as base. 
To get the code running on an ESP8266 I had to remove display-related code (as I have no display ;-) and the (not used?) variable `intPin` :

```
// Set up the interrupt pin, its set as active high, push-pull
// pinMode(intPin, INPUT);
// digitalWrite(intPin, LOW);`
```
The output for self test and acceleration look strange.
```
MPU9250 I AM 71 I should be 71
MPU9250 is online...
x-axis self test: acceleration trim within : -109.6% of factory value
y-axis self test: acceleration trim within : 81.1% of factory value
z-axis self test: acceleration trim within : -100.0% of factory value
x-axis self test: gyration trim within : 77.6% of factory value
y-axis self test: gyration trim within : 23.1% of factory value
z-axis self test: gyration trim within : -100.0% of factory value
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
X-Axis sensitivity adjustment value 1.18
Y-Axis sensitivity adjustment value 1.19
Z-Axis sensitivity adjustment value 1.14
ax = -2000.00 ay = -1862.79 az = -2000.00 mg
gx = -0.04 gy = 0.21 gz = 0.05 deg/s
mx = -819 my = 358 mz = -315 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -13.80, 0.00, 0.00
rate = 0.21 Hz
ax = -2000.00 ay = -1992.92 az = -2000.00 mg
gx = -0.10 gy = 0.02 gz = 0.11 deg/s
mx = -810 my = 346 mz = -324 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -13.80, 0.00, 0.00
rate = 783.71 Hz
```

any hints?

Thanks in advance...

kriswiner, 2020-11-17:
Yes, the loop is waiting for an interrupt flag before it will update any
data...

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

luke-ff, 2020-11-17:
> Yes, the loop is waiting for an interrupt flag before it will update any
data...

But this code is lacking any interrupt handling: https://github.com/kriswiner/MPU9250/blob/master/MPU9250BasicAHRS.ino  

I also can't find any other usage of _intPin_ other than pinMode.

Also self test is inside setup, so I asume intPin has nothing to do with self test ...

kriswiner, 2020-11-17:
Line 388

luke-ff, 2020-11-17:
Sorry, but I don't get it... what has this line 
`if (readByte(MPU9250_ADDRESS, INT_STATUS) & 0x01)`
to do with _intPin_ ?

kriswiner, 2020-11-17:
comment out this line and you should start getting data

luke-ff, 2020-11-17:
Please have a second look at the first posting. I get data, but self test looks invalid. And accelerations show 2g...

kriswiner, 2020-11-17:
Your accel looks broken. Please try the sketche I linked to.

luke-ff, 2020-11-17:
This sketch doesn't compile for the ESP8266. There is also a lot of stuff I don't need. Method MPU9250SelfTest is identical to the code I use atm. I'll try to get a new sensor....

kriswiner, 2020-11-17:
The self test code here is not correct. The actual self test uses only the
last byte of data IIRC, check the documentation.

https://github.com/kriswiner/MPU9250/blob/master/Documents/AN-MPU-9250A-03%20MPU-9250%20Accel%20Gyro%20and%20Compass%20Self-Test%20Implementation%20v1%200_062813.pdf

I would be more concerned about the too-high accel values, looks broken.
You can looks at the accel register values to be sure it is not a
conversion error...but yes, a second (known good) MPU9250 will tell.