kriswiner/MPU9250 #442: Can't read data from MPU9250 to PIC18F4550

PaoloRondot, 2020-11-26:
Hi!

I'm trying to get data from the MPU9250 to the PIC18F4550 using I2C. 

For now I've been able to communicate with my MPU as slave and my PIC as master. I've set the following MPU registers:

```
void MPU6050_Init()			
{
    
	MSdelay(150);		/* Power up time >100ms */
	I2C_Start_Wait(0xD0);	/* Start with device write address */
	I2C_Write(SMPLRT_DIV);	/* Write to sample rate register */
	I2C_Write(0x07);	/* 1KHz sample rate */
	I2C_Stop();
    
	I2C_Start_Wait(0xD0);
	I2C_Write(PWR_MGMT_1);	/* Write to power management register */
	I2C_Write(0x01);	/* X axis gyroscope reference frequency */
	I2C_Stop();
    
        I2C_Start_Wait(0xD0);
	I2C_Write(USER_CTRL);	/* Write to power management register */
	I2C_Write(0x40);	/* X axis gyroscope reference frequency */
	I2C_Stop();
    
	I2C_Start_Wait(0xD0);
	I2C_Write(CONFIG);	/* Write to Configuration register */
	I2C_Write(0x3);	/* Fs = 8KHz */
	I2C_Stop();

	I2C_Start_Wait(0xD0);
	I2C_Write(GYRO_CONFIG);	/* Write to Gyro configuration register */
	I2C_Write(0x18);	/* Full scale range +/- 2000 degree/C */
	I2C_Stop();

	I2C_Start_Wait(0xD0);
	I2C_Write(INT_ENABLE);	/* Write to interrupt enable register */
	I2C_Write(0x01);
	I2C_Stop();
}
```

But unfortunately, when it comes to reading data, I receive the same for accelerometer and gyroscope. Here is my code for reading data:

```
void MPU_Start_Loc()
{
	I2C_Start_Wait(0xD0);								/* I2C start with device write address */
	I2C_Write(ACCEL_XOUT_H);							/* Write start location address from where to read */ 
        I2C_Repeated_Start(0xD1);
        MSdelay(100);
}

void main()
{
        initMyPIC18F();
        glcd_Init(GLCD_ON);
	char buffer[20];
	int Ax,Ay,Az,T,Gx,Gy,Gz;
	float Xa,Ya,Za,t,Xg,Yg,Zg;
        I2C_Init();											/* Initialize I2C */	
	MPU6050_Init();										/* Initialize Gyro */
        MSdelay(100);
	while(1)
	{
		MPU_Start_Loc();								/* Read Gyro values continuously and send to terminal over USART */
		Ax = (((int)I2C_Read(0)<<8) | (int)I2C_Read(0));
		Ay = (((int)I2C_Read(0)<<8) | (int)I2C_Read(0));
		Az = (((int)I2C_Read(0)<<8) | (int)I2C_Read(0));
		T  = (((int)I2C_Read(0)<<8) | (int)I2C_Read(0));
		Gx = (((int)I2C_Read(0)<<8) | (int)I2C_Read(0));
		Gy = (((int)I2C_Read(0)<<8) | (int)I2C_Read(0));
		Gz = (((int)I2C_Read(0)<<8) | (int)I2C_Read(1));
		I2C_Stop();
		
		Xa = (float)Ax/16384.0;							/* Divide raw value by sensitivity scale factor to get real values */
		Ya = (float)Ay/16384.0;
		Za = (float)Az/16384.0;
		Xg = (float)Gx/131.0;
		Yg = (float)Gy/131.0;
		Zg = (float)Gz/131.0;
		t = ((float)T/340.00)+36.53;					/* Convert temperature in °/c using formula */

		sprintf(buffer," Ax = %.2f g",Xa);                  /* Take values in buffer to send all parameters over USART */
                glcd_SetCursor(0,0);
                glcd_WriteString(buffer, f8X8, 1);
        

		sprintf(buffer," Ay = %.2f g",Ya);
                glcd_SetCursor(0,1);
                glcd_WriteString(buffer, f8X8, 1);
	 	
		sprintf(buffer," Az = %.2f g",Za);
                glcd_SetCursor(0,2);
                glcd_WriteString(buffer, f8X8, 1);

		sprintf(buffer," T = %.2f",t);          
                glcd_SetCursor(0,3);
                glcd_WriteString(buffer, f8X8, 1);

		sprintf(buffer," Gx = %.2f",Xg);
                glcd_SetCursor(0,4);
                glcd_WriteString(buffer, f8X8, 1);

		sprintf(buffer," Gy = %.2f",Yg);
                glcd_SetCursor(0,5);
                glcd_WriteString(buffer, f8X8, 1);
		
		sprintf(buffer," Gz = %.2f",Zg);
                glcd_SetCursor(0,6);
                glcd_WriteString(buffer, f8X8, 1);
        
                MSdelay(2000);
	}
}
```

This code is based on a code I found on the internet. As you may see, I'm printing my data on a GLCD and I'm always receiving same data for each x, y, z values on both accelerometer and gyro.

I checked multiple times the registers and their address are good as you may see here:

```
/*
 * MPU6050_res_define.h
 *
 * Created: 04/21/2016 22:47:10
 *  Author: Suraj
 */ 

#ifndef MPU6050_RES_DEFINE_H_
#define MPU6050_RES_DEFINE_H_

#define XG_OFFS_TC 0x00
#define YG_OFFS_TC 0x01
#define ZG_OFFS_TC 0x02
#define X_FINE_GAIN 0x03
#define Y_FINE_GAIN 0x04
#define Z_FINE_GAIN 0x05
#define XA_OFFS_H 0x06 
#define XA_OFFS_L_TC 0x07
#define YA_OFFS_H 0x08 
#define YA_OFFS_L_TC 0x09
#define ZA_OFFS_H 0x0A 
#define ZA_OFFS_L_TC 0x0B
#define XG_OFFS_USRH 0x13
#define XG_OFFS_USRL 0x14
#define YG_OFFS_USRH 0x15
#define YG_OFFS_USRL 0x16
#define ZG_OFFS_USRH 0x17
#define ZG_OFFS_USRL 0x18
#define SMPLRT_DIV 0x19
#define CONFIG 0x1A
#define GYRO_CONFIG 0x1B
#define ACCEL_CONFIG 0x1C
#define FF_THR 0x1D
#define FF_DUR 0x1E
#define MOT_THR 0x1F
#define MOT_DUR 0x20
#define ZRMOT_THR 0x21
#define ZRMOT_DUR 0x22
#define FIFO_EN 0x23
#define I2C_MST_CTRL 0x24
#define I2C_SLV0_ADDR 0x25
#define I2C_SLV0_REG 0x26
#define I2C_SLV0_CTRL 0x27
#define I2C_SLV1_ADDR 0x28
#define I2C_SLV1_REG 0x29
#define I2C_SLV1_CTRL 0x2A
#define I2C_SLV2_ADDR 0x2B
#define I2C_SLV2_REG 0x2C
#define I2C_SLV2_CTRL 0x2D
#define I2C_SLV3_ADDR 0x2E
#define I2C_SLV3_REG 0x2F
#define I2C_SLV3_CTRL 0x30
#define I2C_SLV4_ADDR 0x31
#define I2C_SLV4_REG 0x32
#define I2C_SLV4_DO 0x33
#define I2C_SLV4_CTRL 0x34
#define I2C_SLV4_DI 0x35
#define I2C_MST_STATUS 0x36
#define INT_PIN_CFG 0x37
#define INT_ENABLE 0x38
#define DMP_INT_STATUS 0x39
#define INT_STATUS 0x3A
#define ACCEL_XOUT_H 0x3B
#define ACCEL_XOUT_L 0x3C
#define ACCEL_YOUT_H 0x3D
#define ACCEL_YOUT_L 0x3E
#define ACCEL_ZOUT_H 0x3F
#define ACCEL_ZOUT_L 0x40
#define TEMP_OUT_H 0x41
#define TEMP_OUT_L 0x42
#define GYRO_XOUT_H 0x43
#define GYRO_XOUT_L 0x44
#define GYRO_YOUT_H 0x45
#define GYRO_YOUT_L 0x46
#define GYRO_ZOUT_H 0x47
#define GYRO_ZOUT_L 0x48
#define EXT_SENS_DATA_00 0x49
#define EXT_SENS_DATA_01 0x4A
#define EXT_SENS_DATA_02 0x4B
#define EXT_SENS_DATA_03 0x4C
#define EXT_SENS_DATA_04 0x4D
#define EXT_SENS_DATA_05 0x4E
#define EXT_SENS_DATA_06 0x4F
#define EXT_SENS_DATA_07 0x50
#define EXT_SENS_DATA_08 0x51
#define EXT_SENS_DATA_09 0x52
#define EXT_SENS_DATA_10 0x53
#define EXT_SENS_DATA_11 0x54
#define EXT_SENS_DATA_12 0x55
#define EXT_SENS_DATA_13 0x56
#define EXT_SENS_DATA_14 0x57
#define EXT_SENS_DATA_15 0x58
#define EXT_SENS_DATA_16 0x59
#define EXT_SENS_DATA_17 0x5A
#define EXT_SENS_DATA_18 0x5B
#define EXT_SENS_DATA_19 0x5C
#define EXT_SENS_DATA_20 0x5D
#define EXT_SENS_DATA_21 0x5E
#define EXT_SENS_DATA_22 0x5F
#define EXT_SENS_DATA_23 0x60
#define MOT_DETECT_STATUS 0x61
#define I2C_SLV0_DO 0x63
#define I2C_SLV1_DO 0x64
#define I2C_SLV2_DO 0x65
#define I2C_SLV3_DO 0x66
#define I2C_MST_DELAY_CTRL 0x67
#define SIGNAL_PATH_RESET 0x68
#define MOT_DETECT_CTRL 0x69
#define USER_CTRL 0x6A
#define PWR_MGMT_1 0x6B
#define PWR_MGMT_2 0x6C
#define BANK_SEL 0x6D
#define MEM_START_ADDR 0x6E
#define MEM_R_W 0x6F
#define DMP_CFG_1 0x70
#define DMP_CFG_2 0x71
#define FIFO_COUNTH 0x72
#define FIFO_COUNTL 0x73
#define FIFO_R_W 0x74
#define WHO_AM_I 0x75

#endif /* MPU6050_RES_DEFINE_H_ */
```

I can't figure out what's causing that... I hope I'll get some help!

Thanks!

kriswiner, 2020-11-26:
I don't understand what you are doing here:

Ax = (((int)I2C_Read(0)<<8) | (int)I2C_Read(0));

		Ay = (((int)I2C_Read(0)<<8) | (int)I2C_Read(0));

		Az = (((int)I2C_Read(0)<<8) | (int)I2C_Read(0));

		T  = (((int)I2C_Read(0)<<8) | (int)I2C_Read(0));

		Gx = (((int)I2C_Read(0)<<8) | (int)I2C_Read(0));

		Gy = (((int)I2C_Read(0)<<8) | (int)I2C_Read(0));

		Gz = (((int)I2C_Read(0)<<8) | (int)I2C_Read(1));

		I2C_Stop();

Why I2C_Read(0) everywhere but for Gz?

Why not just read all 14 bytes in one I2C read operation and then construct
the values?

PaoloRondot, 2020-11-26:
So basically, I'm reading and shifting the register ACCEL_XOUT_H when  I do ```((int)I2C_Read(0)<<8)``` and when I use the | operator, it enables to get the register ACCEL_XOUT_L on the same integer variable.

Then the parameter in the I2C_Read function is the acknowledge response from the PIC, as you may see here:
```
char I2C_Read(char flag)            /* Read data from slave devices with 
                                     * 0=Ack & 1=Nack */
{
    char buffer;
    RCEN = 1;                       /* Enable receive */
    while(!SSPSTATbits.BF);         /* Wait for buffer full flag which set 
                                     * after complete byte receive */
    buffer = SSPBUF;                /* Copy SSPBUF to buffer */
    if(flag==0)
        I2C_Ack();                  /* Send acknowledgment */
    else
        I2C_Nack();                 /* Send negative acknowledgment */
    I2C_Ready();
    return(buffer);
}
```

I tried to put 1 everywhere and I have the same issue.

I didn't try to do it this way, but either way I have to read all the data byte by byte with my function I2C_Read, no?

kriswiner, 2020-11-26:
I have never used a pic so no idea here. On the STM32 we read all the bytes
in one burst read. Perhaps lower down in the system layer this is done one
at a time anyway. But you problem seems to be with the Pic I2C API, not the
MPU9250. Can you read the WHO-AM-I register?

PaoloRondot, 2020-11-26:
I get the char 'p' so 0x70 when I read WHO-AM-I register

PaoloRondot, 2020-11-26:
Does the initialisation of the MPU registers look okay though ?

kriswiner, 2020-11-26:
no idea, I am not going to try to debug your code.

kriswiner, 2020-11-26:
who am i should be 0x71

PaoloRondot, 2020-11-26:
So what does it means, having a wrong value for WHO-AM-I ?

kriswiner, 2020-11-26:
no idea

PaoloRondot, 2020-11-27:
Alright I have some new info that might be interesting for my problem. 

I'm able to read the register in which I write during the initialisation. I still don't have the right value for WHO-AM-I, though.

But I'm not able to read more than one register... Once I've read, for example CONFIG, I cannot read GYRO_CONFIG afterwards. Here is the code that I used:
```
void MPU6050_Init()										/* Gyro initialization function */
{
    
	MSdelay(150);		/* Power up time >100ms */
	I2C_Start_Wait(0xD0);	/* Start with device write address */
	I2C_Write(SMPLRT_DIV);	/* Write to sample rate register */
	I2C_Write(0x07);	/* 1KHz sample rate */
	I2C_Stop();
    
	I2C_Start_Wait(0xD0);
	I2C_Write(PWR_MGMT_1);	/* Write to power management register */
	I2C_Write(0x01);	/* X axis gyroscope reference frequency */
	I2C_Stop();
    
        I2C_Start_Wait(0xD0);
	I2C_Write(USER_CTRL);	/* Write to power management register */
	I2C_Write(0x40);	/* X axis gyroscope reference frequency */
	I2C_Stop();
    
	I2C_Start_Wait(0xD0);
	I2C_Write(CONFIG);	/* Write to Configuration register */
	I2C_Write(0x3);	/* Fs = 8KHz */
	I2C_Stop();

	I2C_Start_Wait(0xD0);
	I2C_Write(GYRO_CONFIG);	/* Write to Gyro configuration register */
	I2C_Write(0x18);	/* Full scale range +/- 2000 degree/C */
	I2C_Stop();

	I2C_Start_Wait(0xD0);
	I2C_Write(INT_ENABLE);	/* Write to interrupt enable register */
	I2C_Write(0x01);
	I2C_Stop();
}

void MPU_Start_Loc(unsigned char add)
{
	I2C_Start_Wait(0xD0);								/* I2C start with device write address */
	I2C_Write(add);							/* Write start location address from where to read */ 
        I2C_Repeated_Start(0xD1);
        MSdelay(100);
}

void main()
{
    initMyPIC18F();
    glcd_Init(GLCD_ON);
    I2C_Init();											/* Initialize I2C */	
    MPU6050_Init();										/* Initialize Gyro */

    MPU_Start_Loc(CONFIG);
    char a = I2C_Read(1);
    I2C_Stop();

    MPU_Start_Loc(GYRO_CONFIG);
    char b = I2C_Read(1);
    I2C_Stop();

    glcd_SetCursor(0,0);
    glcd_WriteChar8X8(a+70, 1);
    glcd_SetCursor(0,1);
    glcd_WriteChar8X8(b+50, 1);
    MSdelay(10000);
    while(1){}
}
```

So basically, in my main, I'm initializing some stuff, then I call the function ```MPU_Start_Loc()``` which enables to tell to the MPU where we're reading. I put the value in a char that I'm then displaying on my GLCD.

I expect to have the letter I and J displayed. The offset added to the char in ```glcd_WriteChar8X8()``` is just here to have a real letter displayed, as 0x3 and 0x18 are not char that can be displayed. What I have is always the first value. What I read first is displayed, what comes after is not.

Any clue?

kriswiner, 2020-11-27:
can you successfully read any i2c sensor data? This sounds like a pic
problem, not an mpu9250 problem. Maybe ask at a pic forum?

PaoloRondot, 2020-11-27:
For now, I didn't read any data coming from the sensor... Yes I'm gonna do that.

Thanks for your help though!!

PaoloRondot, 2020-11-27:
I'm gonna give a last info that I have found out. 

So when I read the data in repeated start mode, when I acknowledge the received data, it seems that the MPU is not going to the next register.. It keeps giving me the value of the same register.

PaoloRondot, 2020-11-27:
And I just found out that the MPU doesn't acknowledge when I try to read again a different register. It acknowledges the 0xD0 but then it doesn't acknowledge the following register which is the one where I'm trying to read.

PaoloRondot, 2020-11-28:
Found the answer, it was just the debug config that I had to change in my PIC code. 
Thanks!

closed 2020-11-28 by PaoloRondot