kriswiner/MPU9250 #444: Something weird about Yaw value

YouMinJung, 2020-12-03:
Hi. Thank you for your code.

Current, I'm tested with **MPU9250BasicAHRS.ino**.
I'm using the **MPU9250 and Arduino nano**. 

This is the first result when I execute the program.
///////////////////////////////////////////////////////////////////////////////////////////////
MPU9250: I AM 71. I should be 71
MPU9250 is online...
x-axis self test: acceleration trim within : 0.0% of factory value
y-axis self test: acceleration trim within : 0.0% of factory value
z-axis self test: acceleration trim within : -1.0% of factory value
x-axis self test: gyration trim within : 0.2% of factory value
y-axis self test: gyration trim within : 0.6% of factory value
z-axis self test: gyration trim within : 0.0% of factory value
MPU9250 bias
	x	y	z
	1999	1999	-178	mg
	-1.3	2.2	2.6	o/s
MPU9250 initialized for active data mode....
AK8963: I AM 48. I should be 48
AK8963 initialized for active data mode....
X-Axis sensitivity adjustment value 1.19
Y-Axis sensitivity adjustment value 1.19
Z-Axis sensitivity adjustment value 1.15
///////////////////////////////////////////////////////////////////////////////////////////////

After above result, I could get the below results.
///////////////////////////////////////////////////////////////////////////////////////////////
ax = 1999.94 ay = 1999.94 az = 889.59 mg
gx = -0.07 gy = -0.04 gz = 0.05 deg/s
mx = -202 my = -474 mz = 542 mG
q0 = 0.20 qx = 0.51 qy = 0.30 qz = 0.78
Yaw, Pitch, Roll: 108.94, -42.45, 65.86
rate = 429.79 Hz
	x	y	z
	1999	1999	889	mg
	0	0	0	o/s
	-202	-474	542	mG
	108	-42	65	ypr
rt: 429.79 Hz
ax = 1999.94 ay = 1999.94 az = 895.02 mg
gx = -0.04 gy = 0.00 gz = -0.13 deg/s
mx = -214 my = -480 mz = 524 mG
q0 = 0.20 qx = 0.51 qy = 0.30 qz = 0.78
Yaw, Pitch, Roll: 109.14, -42.43, 65.84
rate = 429.98 Hz
	x	y	z
	1999	1999	895	mg
	0	0	0	o/s
	-214	-480	524	mG
	109	-42	65	ypr
rt: 429.98 Hz
ax = 1999.94 ay = 1999.94 az = 897.77 mg
gx = 0.02 gy = -0.01 gz = 0.20 deg/s
mx = -194 my = -469 mz = 531 mG
q0 = 0.21 qx = 0.51 qy = 0.30 qz = 0.78
Yaw, Pitch, Roll: 108.52, -42.38, 65.89
rate = 428.38 Hz
	x	y	z
	1999	1999	897	mg
	0	0	0	o/s
	-194	-469	531	mG
	108	-42	65	ypr
rt: 428.38 Hz
///////////////////////////////////////////////////////////////////////////////////////////////

In these circumstance, I have several questions.
1. when I move the sensor, accelerometer value never changed. 
    So I used the different sensor, then this sensor is changing accelerometer value well. 
    These two sensor have different factory value. Why these happen? How to set the sensor correctly?

2. I want to get the correct yaw value. However, yaw value is not that correct. 
    If I rotate the sensor clock wise, yaw value changed like this. "90 -> 83 -> 81 -> 83 -> 85 -> 87 -> ... ".
    Also, yaw value is affected by the pitch and roll value.
    How can I get the correct yaw value?

kriswiner, 2020-12-03:
The first result shows your accel is broken, throw that sensor away,
nothing can be done.

You need to properly calibrate the sensors and use NED convention for
feeding the sensor data to the madgwick filter. This is done properly
elsewhere, like here
<https://github.com/kriswiner/Dragonfly/tree/master/MPU9250Optimized>, for
example. The basic sketch is not appropriate for accurate heading
estimation without modification.

YouMinJung, 2020-12-04:
Thank you for your support!

By the way, I have a very little knowledge about the IMU sensor. 
So... How to calibrate the sensor properly? Could you give me some example?

kriswiner, 2020-12-04:
I linked to an example that has provision for calibration of all three
sensors. Did you look at it?

YouMinJung, 2020-12-04:
I saw the code you linked. (= MPU9250Optimized.ino)
In this code I could do calibration the accel, gyro and magnet.
After calibration, I move the IMU sensor but values did not changed... (-> result is like above comments.)
What is the problem...??

kriswiner, 2020-12-04:
If your sensor values do not change then the sensor is broken,

If the sensor is OK, then you can check your calibration. What are the
values of the accel and gyro when the sensor is laying flat and motionless
on the table? SHould be 0.0, 0.0, 1.0 g and 0.0, 0.0 , 0.0 dps.

Then check magnetometer data. magnitude should be ~0.5 Gauss. With board
face pointed up Mz ~ 0.4 G, with board face pointed down Mz ~ - 0.4 G.

Is this what you see?

YouMinJung, 2020-12-04:
This is the sensor biases.
////////////////////////////////////////
accel biases (mg)
191.59
187.68
-17.40
gyro biases (dps)
-1.21
0.18
-2.26
K8963 mag biases (mG)
219.72
216.15
385.72
AK8963 mag scale (mG)
1.17
0.93
0.93
////////////////////////////////////////

After calibration, sensor value always out 0 like these. 
////////////////////////////////////////////////
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = 0 my = 0 mz = 0 mG
////////////////////////////////////////////////

(When I execute the MPU9250Optimized.ino)
When the sensor is laying flat and motionless on the table, or when the sensor waved... all of sensor value always 0.
When board face pointed down or up, also sensor value all 0.

Then.. Is this sensor broken..??

kriswiner, 2020-12-04:
If you are able to read sensor data during calibration then the problem is
probably something simpler like not connecting the interrupt pin.

YouMinJung, 2020-12-07:
I set the interrupt pin... (I changed the variable to "int intPin = 2", and then connected to Digital Out 2 in Arduino Nano.)
 but still all of the values are 0.
Is there another suspected cause..??

kriswiner, 2020-12-07:
Examine the raw register values. Sounds like you are multiplying the data
by zero somewhere...

YouMinJung, 2020-12-08:
Well. Thank you for your help. Maybe I think I fixed this problem.
However, The sensor values are still incorrect.

This is the values when the sensor is calibrated and laid flat and motionless on the table.
//////////////////////////////////////////////////////////
ax = -243.10 ay = -107.24 az = 801.33 mg
gx = -0.12 gy = 0.33 gz = -0.10 deg/s
mx = -110 my = 157 mz = 265 mG
q0 = 0.97 qx = -0.09 qy = 0.13 qz = 0.16
Yaw, Pitch, Roll: 4.20, 16.80, -7.59
rate = 405.29 Hz

ax = -246.09 ay = -109.50 az = 802.61 mg
gx = 0.00 gy = 0.07 gz = -0.03 deg/s
mx = -109 my = 165 mz = 267 mG
q0 = 0.97 qx = -0.09 qy = 0.13 qz = 0.17
Yaw, Pitch, Roll: 4.47, 16.78, -7.55
rate = 406.28 Hz
//////////////////////////////////////////////////////////

I think that the value of the accelerator is something wrong...
Do you think this sensor is broken?

kriswiner, 2020-12-08:
Not broken, just not calibrated properly.

YouMinJung, 2020-12-08:
When I calibrate the magnetor, I move the sensor like figure eight.
Then how to action, when I calibrate the acceler and gyro?

kriswiner, 2020-12-08:
Also, after you have fixed this, you will find your fusion rate at ~500 Hz
is a bit too low, should be >2KHz....maybe switch to a Teensy?

kriswiner, 2020-12-08:
Place the sensor flat and motionless when calibrating the accel and gyro.

YouMinJung, 2020-12-08:
Thank you. I get the much better result.
/////////////////////////////////////////////////
ax = -3.48 ay = 27.77 az = 977.05 mg
gx = 0.03 gy = 0.14 gz = 0.01 deg/s
mx = 95 my = -129 mz = 184 mG
q0 = -0.32 qx = -0.01 qy = 0.01 qz = 0.95
Yaw, Pitch, Roll: -156.52, 0.21, 1.66
rate = 407.43 Hz
/////////////////////////////////////////////////

Maybe I think that if I do better calibration, I can get the better result.
By the way, you are mention the Teensy.
Is it because the processor spec?

kriswiner, 2020-12-08:
Best would be this
<https://www.tindie.com/products/tleracorp/ladybug-stm32l432-development-board/>
or this
<https://www.tindie.com/products/tleracorp/dragonfly-stm32l47696-development-board/>.
But teensy 3.2 is also fast enough to give you fusion rates > 1 kHz.

Your calibration should show accel 0 0 1000 mg, gyro 0, 0 0 dps, and mag
consistent with your local magnetic field. Usually, the magnitude of the
magnetic field should be 450 - 500 mGauss.

Your values show the calibration could stand improvement. This is critical
to getting accurate absolute orientation estimation, as is a fast enough (5
- 10 x of the sample rate ) fusion rate.

YouMinJung, 2020-12-08:
Oh I see. I will consider to change the board.
Thank you for your help again!

kriswiner, 2020-12-08:
You can repay my help buying one of my boards. I recommend the Ladybug if
you are on a budget. Dragonfly for more gpios and features. Both are
programmable using the Arduino IDE via USB.