kriswiner/MPU9250 #445: Reliable MPU9250

luizamanfrini, 2020-12-03:
Hey Kris, how are you?

Where can I get reliable MPU-9250. I've been struggling with mine because and I think they are fake.

Thank you!

kriswiner, 2020-12-03:
As you might know, the MPU9250 are out of production since January 2020. I
have some real ones, and I have some fake ones. You can take your chances
and order from Aliexpress, but this is hit and miss...

luizamanfrini, 2020-12-03:
Thank you Kris! I've been having some trouble connecting to the magnetometer, that's why I think my sensor might be fake. Its who am I is 0xFF instead of 0x48. Also, I am having some trouble to use is as SPI.

kriswiner, 2020-12-03:
Yes, there are some Chinese variants that only have the accel/gyro...

If you just need a few I can sell them to you...

luizamanfrini, 2020-12-03:
Would you ship to Brazil?

kriswiner, 2020-12-03:
Yes, but is $14.95 first class airmail good/reliable enough? Maybe cheaper
for you to buy from aliexpress....

GitMoDu, 2020-12-07:
Any thoughts on a successor chip?

kriswiner, 2020-12-07:
Yes, TDK makes the ICM20948
<https://invensense.tdk.com/products/motion-tracking/9-axis/icm-20948/>
which is literally the successor to the MPU9250. You can get 2 degree
heading accuracy using its embedded DMP, so overall a pretty good choice.
However, I don;t like the dual power rail (1.8 and 3.3 V required) and I
don;t like loading a binary blob to get fusion performance. So we don;t use
this.

We use instead ST Micro's LSM6DSM accel/gyro combo sensor and LIS2MDL
magnetometer. Lately we have started using the MMC5983A
<https://www.digikey.com/en/products/detail/memsic-inc/MMC5983MA/10452795?s=N4IgTCBcDaILZwMYFYCcAOAzCAugXyA>
since it offers much higher resolution and much lower noise.

More info here
<https://hackaday.io/project/160283-max32660-motion-co-processor>.

GitMoDu, 2020-12-15:
I'll look into those new ICs. I agree with having magic blobs is counter to the embedded mentality of absolute control.

Thank you very much, your efforts to make these complicated devices usable have been wonderful!