kriswiner/MPU9250 #448: Help MPU9250 find North

flidsky, 2020-12-24:
Hello Kris

I'm making project that include a compass with tilt compensation and very much in need your advice. I bought a few MPU9250 and 9255 from different manufactures and connected it to ESP32 running on 240Mhz. Gyro and accel are awesome, but magnetometer never get north and its raw data is noisy. I use your code for this chip where I removed all stuff for air pressure sensor. I spent a week reducing noise, avoiding EMI, waving a figure eight and playing with settings and I never got north for more than a turn around. All of my modules have the same behavior: wierd drifting or waving between two values but often if I let them lie down for a while it shows north correctly.  I checked all what I could but still not sure about correct timings of filtering or integration. Also now i'm not sure that using 9250 is a proper solution.
And what else I observe: every load fixed compass slowly drift to north in real but random in degrees. 

As the last attempt can I ask you help me set it up?

flidsky, 2020-12-24:
https://youtu.be/FcbioW3APCw

kriswiner, 2020-12-24:
How did you calibrate your sensors? How did you verify proper calibration?

flidsky, 2020-12-24:
Thank you for answering. 
I did calibration in many different ways trying to guess the best, slow rotation in three axes, full revolution around roll pitch, yaw, and the same a bit faster, slower and more faster, more amplitude, less, just randomly, etc.
I check calibration by comparing values with three compasses: digital, analog and one more in my phone.
To lessen interference I did it
outside also.
It can't track precisely a single rotation, my guess is that the compass works only using accel and gyro and they can't track yaw. Cumulative error was more than 50° after one minute use.

flidsky, 2020-12-24:
Waiting you answer we research around calibration and made our own algorithm but is stll drifting as here
https://youtu.be/K-P8V6ZddM0
[image]

kriswiner, 2020-12-24:
Mag calibration doesn't look too good. You also have to calibrate the accel
and gyro.

When calibrated and with the board at read flat on a table for example the
accel should read 0.0, 0.0. 1000 milligs to within ~10 milligs on all three
axes, the gyro should read 0, 0, 0 dps to within 0.01 dps. The mag should
read whatever your local magnetic H and V values are, which you can look up
via USGS site or equivalent. Also, if the board is flat on a table and
reads Mz = X, it should read Mz = -X when the board is flipped upside down.

The sensor order fed into the Madgwick function also matters, and should
conform to NED convention. Does it?

kriswiner, 2020-12-24:
Send me a link to the sketch you are using (don't send me code!).