kriswiner/MPU9250 #449: I2C with Arduino Won't find MPU9250

abnoviello23, 2020-12-28:
Hi,
I am trying to use the MPU9250, but no matter what I do, the I2C Scanner I have does not find any I2c addresses. When I call endTransmission(), it returns two, which I believe indicates that it is receiving a NACK and that I need pull-up resistors of some sort. Is this correct?

If so, what does the wiring look like for the pull-up resistors? - no matter where I put them, nothing seems to work.

Thanks for everything!

kriswiner, 2020-12-28:
4K7 or 10K resistors must be attached between the SDA and 3V3 and between
the SCL and the 3V3 lines.

abnoviello23, 2020-12-28:
I have 10K resistors between the SDA and 3V3 and the SCL and 3V3 lines - it still doesn't work.

kriswiner, 2020-12-28:
Could be a lot of things. Do you have 3V3 on the board? Send me an image of
your setup.

abnoviello23, 2020-12-28:
Here is a picture - I will send you a clearer schematic, draw out in a few minutes.
[image]

abnoviello23, 2020-12-28:
Thanks for all the help in advance! - I will have the drawing in less than 5 minutes.

abnoviello23, 2020-12-28:
Schematic/Drawing of circuit, since the other picture may be hard to follow.
[image]

kriswiner, 2020-12-28:
There are probably already pullups on the board. I think you have to pull
NCS to HIGH.

Is there 3V3 on the board?

kriswiner, 2020-12-28:
In other words. Connect power and ground only. The measure the voltage
between SDA and GND and SCL and GND. Are these 3V3?

abnoviello23, 2020-12-28:
There is 3V3 on the arduino - should I just connect NCS to 3V3? Also, I don't have a multimeter to measure the voltage, but there shouldn't be anything other than 3V3. Also, since the drawing wasn't very good, the 10K resistors from SDA and SCL are only connected to 3V3, not ground. Is that correct?

kriswiner, 2020-12-28:
You will need a multimeter to measure voltages, otherwise there is no way
to diagnose your problem.

Does the board allow 5 V power input?

Please send me a picture of the sensor board without the wires. Is it the
GY-MPU6500?

Simply connect power/GND/SDA and SCL and try your scan. You might have to
connect NCS to 3V3.

abnoviello23, 2020-12-28:
The board itself allows for power input between 2.4V and 3.6V, but has a 3.5 V regulator, too, so 5V input is okay. Info on piece:https://components101.com/processors/mpu9250-9-axis-digital-motion-processor

I will try simply connecting power, gnd, sda, and scl, as well as ncs, without any resistors now. 

Here is the picture of the sensor:
[image]

abnoviello23, 2020-12-28:
No luck with GND connected to GND, VCC connected to 3V3, SDA to SDA, SCL to SCL, and NCS to 3V3 with no pull-up resistors.

kriswiner, 2020-12-28:
Could also be a bad board, no idea without a logic analyzer...

kriswiner, 2020-12-28:
What I would measure is the voltage on SDA and SCL in this configuration,
this will tell you about the power to the board and the state of the
pullups, etc.

abnoviello23, 2020-12-28:
Without a multimeter, would you recommend just trying a variety of resistors? Are the resistors even totally necessary for this sensor - all of the videos I have watched are of people simply connecting 5V on Arduino to VCC on sensor, SDA, SCL, and GND, and everything working just fine (no resistors or anything). What would be your best guess as to why that isn't working for me?

kriswiner, 2020-12-28:
There are already pullups on the board, no external pullups are needed.

abnoviello23, 2020-12-28:
The I2C Scanner found the sensor (same type, different individual one) on another computer and another Arduino, but still fails to  find the sensor in the other code, where I actually try to get readings. Any ideas on what might be causing that issue?

kriswiner, 2020-12-28:
Nope. Could it be the code? If you use the same code, same Arduino and
different sensor and get a different result, then the sensor is broken.

abnoviello23, 2020-12-28:
Scanner code is the same and the code to get the readings is sample code from the library. This code returns that the status of the sensor  is -1 sometimes, but other times it is actually giving readings. The only problem is that the readings are very strange - not correct at all, and the magnetometer's readings go to 0 after about 5 times. Here is the code we are using:
`#include "MPU9250.h"

MPU9250 IMU(Wire,0x68);
int status;

void setup() {
  // serial to display data
  Serial.begin(9600);
  while(!Serial) {}

delay(1000);
  // start communication with IMU 
  status = IMU.begin();
  if (status < 0) {
    Serial.println("IMU initialization unsuccessful");
    Serial.println("Check IMU wiring or try cycling power");
    Serial.print("Status: ");
    Serial.println(status);
    while(1) {}
  }
}

void loop() {
  // read the sensor
  IMU.readSensor();
  // display the data
  Serial.print("Accelerometer (X, Y, Z):      ");
  Serial.print(IMU.getAccelX_mss());
   Serial.print("   ");
  Serial.print(IMU.getAccelY_mss());
   Serial.print("   ");
  Serial.println(IMU.getAccelZ_mss());
  
  Serial.print("Gyro (X, Y, Z):      ");
  Serial.print(IMU.getGyroX_rads());
   Serial.print("   ");
  Serial.print(IMU.getGyroY_rads());
   Serial.print("   ");
  Serial.println(IMU.getGyroZ_rads());

 Serial.print("Magnetometer (X, Y, Z):      ");
  Serial.print(IMU.getMagX_uT());
 Serial.print("   ");
 Serial.print(IMU.getMagY_uT());
 Serial.print("   ");
 Serial.print(IMU.getMagZ_uT());
//D = arctan(yGaussData/xGaussData)∗(180/π)
int dir = atan2(IMU.getMagY_uT(), IMU.getMagX_uT()) * ((180)/ (PI));
Serial.print("     Direction Heading:   ");
Serial.println(dir);
  Serial.println();
  Serial.println();
  Serial.println();
  delay(1000);
}`

lbdroid, 2021-02-28:
I see a couple of problems with this setup.

1) the mpu9250 board MUST be powered by 5v. If you want to power it by 3.3v, you need to bypass the voyage regulator (the chip by the 5v/GND pins). The unpopulated 805 footprint beside the regulator can be bridged to do this. Because there will be a voltage drop across the regulator, to get 3.3v out of it, you need to feed it more than 3.3v. I'd say at least 4.5v. If you try powering it with 3.3v without bypassing the regulator, the i2c pins wont be pulled up high enough to trigger the logic high on the 5v microcontroller.

2) You are connecting to the analog pins of the MCU breakout board -- the labeling on the headers says that SDA/SCL are at the opposite corner of the board by the reset button.

3) For that particular 9250 board, the ONLY connections you need to make are 5v, GND, SDA, SCL. Optionally, you may also connect INT, which will save you from having to poll the chip for new data. All other pins may be left NOT CONNECTED.

zerovijay, 2023-10-27:
I have issue on my mpu9250/6500 I can't
Detect the mpu9250 via i2c I try somany library all the library showing error
Including micropython
So I cry to write a code acces who am I from address 117 the result except 0×71 but I got same value i provided 117 i falow the datasheet  the device address 0x68 A0 logic low
But one thing the temperature and gyro and Accel work fine i Take look the ic it's named MP92 and sumthing and 6007 The breakout board looks same on mpu9250 I saw on online and YouTube videos also mentioned on the board mpu9250/6500 please help me any one face this issue.

kriswiner, 2023-10-27:
Not sure what the problem is here, you say you see 0x71 as WHO_AM_I and the
sensors work fine. So what's the problem?

zerovijay, 2023-10-28:
Arduino library showing errors not recognise the mpu9250

zerovijay, 2023-10-28:
I try to  understand library the who am I want to return 0x71 but I got 0x75

zerovijay, 2023-10-28:
That's why the library showing error i think

kriswiner, 2023-10-28:
75 means this is not an MPU9250, but some variant.

zerovijay, 2023-10-28:
On the it's mentioned mp92

zerovijay, 2023-10-28:
On the chip it's mentioned mp92
And sumthing I can't see clearly but looks like same bord [image]

[image]

zerovijay, 2023-10-28:
I try to read who Ami and device address the address on the chip is 0x68 A0 logic low if i pull-up the A0 I got 0x69 temperature,gyro and Accel register working fine but looks like the values are incorrect i reset the chip via power register1 and 8th bit position after reset all registers no longer return anything
But I disconnect and reconnect the module it's back to normal any solution for this

zerovijay, 2023-10-28:
I am actually studying embedded software development last 6 months push my self in to write software from scratch but I am stuck on this

kriswiner, 2023-10-28:
> And sumthing I can't see clearly but looks like same bord [image:
> IMG_20231029_032529.jpg]

Yup, Invensense made a lot of variants, MPU9250, MPU9255, MPU92XX, etc
your board has some variant of the original MPU9250 with WHO_AM_I of 0.75.
Otherwise should be similar with either fewer registers or additional
registers.

https://github.com/kriswiner/MPU9250/issues/486

https://github.com/kriswiner/MPU9250/issues/47

search on "Invensense MPU92 WHO_AM_I 0x75" for other examples...

kriswiner, 2023-10-28:
"I try to read who Ami and device address the address on the chip is 0x68
A0 logic low if i pull-up the A0 I got 0x69 temperature,gyro and Accel
register working fine but looks like the values are incorrect i reset the
chip via power register1 and 8th bit position after reset all registers no
longer return anything"

Which values are incorrect? And how do you know?

zerovijay, 2023-10-28:
I try to read temperature and gyro with out calibration just for checking if register is working [image]

I am working on the code it's not fully finished but I implement sum futures to read 16bit sigen integer from registers and return tuple of xyz i try to move around the sensor the values not I want i know it's not calibrated but the values so match random that's why I say the reading wrong.

zerovijay, 2023-10-28:
I am in India. this thing so match expensive

kriswiner, 2023-10-28:
You have to convert the (unsigned) bytes into signed 16-bit integers first,
then convert to float.

Can you just read the bytes properly?

zerovijay, 2023-10-28:
i am sorry its typo error signed integer

kriswiner, 2023-10-28:
int16_t output = (int16_t) ( (int16_t) byte1 << 8) | byte0 );

zerovijay, 2023-10-28:
hear is my method reading sensor raw data from registers and unpack the data in to 16 bit signed integer 

```python
import ustruct
from machine import I2C
from micropython import const

class MPU9250:
    # Register Map for Gyroscope and Accelerometer (constant)
    # =============================================================================================
    __SELF_TEST_X_ACCEL: int = const(0x0D)
    __SELF_TEST_Y_ACCEL: int = const(0x0E)
    __SELF_TEST_Z_ACCEL: int = const(0x0F)

    __XG_OFFSET_H: int = const(0x13)
    __XG_OFFSET_L: int = const(0x14)
    __YG_OFFSET_H: int = const(0x15)
    __YG_OFFSET_L: int = const(0x16)
    __ZG_OFFSET_H: int = const(0x17)
    __ZG_OFFSET_L: int = const(0x18)

    __SMPL_RT_DIV: int = const(0x19)
    __CONFIG: int = const(0x1A)
    __GYRO_CONFIG: int = const(0x1B)
    __ACCEL_CONFIG1: int = const(0x1C)
    __ACCEL_CONFIG2: int = const(0x1D)
    __LP_ACCEL_ODR: int = const(0x1E)
    __WOM_THR: int = const(0x1F)
    __FIFO_EN: int = const(0x23)

    __I2C_MST_CTRL: int = const(0x24)
    __I2C_SLV0_ADDR: int = const(0x25)
    __I2C_SLV0_REG: int = const(0x26)
    __I2C_SLV0_CTRL: int = const(0x27)
    __I2C_SLV1_ADDR: int = const(0x28)
    __I2C_SLV1_REG: int = const(0x29)
    __I2C_SLV1_CTRL: int = const(0x2A)
    __I2C_SLV2_ADDR: int = const(0x2B)
    __I2C_SLV2_REG: int = const(0x2C)
    __I2C_SLV2_CTRL: int = const(0x2D)
    __I2C_SLV3_ADDR: int = const(0x2E)
    __I2C_SLV3_REG: int = const(0x2F)
    __I2C_SLV3_CTRL: int = const(0x30)
    __I2C_SLV4_ADDR: int = const(0x31)
    __I2C_SLV4_REG: int = const(0x32)
    __I2C_SLV4_DO: int = const(0x33)
    __I2C_SLV4_CTRL: int = const(0x34)
    __I2C_SLV4_DI: int = const(0x35)
    __I2C_MST_STATUS: int = const(0x36)
    __INT_PIN_CFG: int = const(0x37)
    __INT_ENABLE: int = const(0x38)
    __INT_STATUS: int = const(0x3A)

    __ACCEL_X_OUT_H: int = const(0x3B)
    __ACCEL_X_OUT_L: int = const(0x3C)
    __ACCEL_Y_OUT_H: int = const(0x3D)
    __ACCEL_Y_OUT_L: int = const(0x3E)
    __ACCEL_Z_OUT_H: int = const(0x3F)
    __ACCEL_Z_OUT_L: int = const(0x40)

    __TEMP_OUT_H: int = const(0x41)
    __TEMP_OUT_L: int = const(0x42)

    __GYRO_X_OUT_H: int = const(0x43)
    __GYRO_X_OUT_L: int = const(0x44)
    __GYRO_Y_OUT_H: int = const(0x45)
    __GYRO_Y_OUT_L: int = const(0x46)
    __GYRO_Z_OUT_H: int = const(0x47)
    __GYRO_Z_OUT_L: int = const(0x48)

[... 89 lines cut ...]
        pass

    def calibrate(self) -> None:
        pass

    def temperature(self, room_offset: float, sensitivity: float) -> float:
        # Read the raw temperature data from the sensor
        raw_temp: bytes = self.__read_from_register(mem_addr=self.__TEMP_OUT_H, num_byte=2)
        temp_out: tuple = ustruct.unpack("<h", raw_temp)
        # Calculate the temperature in degrees Celsius
        temp_c: float = ((temp_out[0] - room_offset) / sensitivity) + 21
        return temp_c

    def gyroscope(self) -> tuple:
        raw_gyro: bytes = self.__read_from_register(mem_addr=self.__GYRO_X_OUT_H, num_byte=6)
        gyro: tuple = ustruct.unpack("<hhh", raw_gyro)
        return gyro

    def accelerometer(self) -> tuple:
        raw_accel: bytes = self.__read_from_register(mem_addr=self.__ACCEL_X_OUT_H, num_byte=6)
        accel: tuple = ustruct.unpack("<hhh", raw_accel)
        return accel

```
this is my prototype. Not fully implemented.

kriswiner, 2023-10-28:
Sorry, I can't debug your code.

Just look at the raw bytes, do they make sense?

zerovijay, 2023-10-28:
the gyroscope function give a request to read_from_register function the function response to the request the response is 16bit signed integers tuple of xyz

zerovijay, 2023-10-28:
i need to know is that who am i all so signed integer is that we want to do 2's compliment for this value?

kriswiner, 2023-10-29:
No, this is a single byte that apparently should be 0x75 for your MPU925X
variant.