kriswiner/MPU9250 #450: Roll, pitch, yaw keep changing while using MPU9250

clc25600, 2021-01-02:
Hi Kris, 
How are you, thanks a lot for your code.
I am doing a project on NANO using MPU-9250.
The magnitude of the angles seems to change correctly when I tilted and spin the board.
But if I just put it still for a while, I discovered that it would keep changing its value instead of keeping the angle. 
Even if I have calibrated it, the roll pitch yaw value will just keep increasing or decreasing continuously.
I am wondering what would cause the problem, thank you in advance!
  
[image]

kriswiner, 2021-01-02:
How did you verify that all three sensors are properly calibrated?

clc25600, 2021-01-02:
I put the calibration code in the setup function and wait till it started to have detected value.
I have not really verified if the value is correct or not.
Is there any method that I can make sure that the value after calibrated is correct?

kriswiner, 2021-01-02:
Accel should read 0, 0, +1 g to within +/- 10 mg and gyro should read 0, 0,
0 dps to within +/- 0.001 dps when the sensor is flat on a table and
motionless.

Mag should read consistent with your local magnetic field.

If Mag Mz = X when the sensor faces up at rest, then it should read -X when
flipped 180 degrees.

Total magnetic field magnitude should be ~0.5 Gauss, etc.

clc25600, 2021-01-02:
Thanks a lot for the reply,
Now I have confirmed that my accel and gyro should be correct.
But the mag have some offset that make it not satisfied the -X when I flipped 180 degrees.
Should I just add the bias to it or if I should do anything else?

kriswiner, 2021-01-02:
Maybe properly calibrate the mag? Or at least do this
<https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration>

clc25600, 2021-01-03:
Thanks a lot for the help!
After properly calibration everything goes well!