kriswiner/MPU9250 #452: Self test Gyro and Accel is fail

iamnot-v, 2021-01-21:
Hi, Kriss.

I bought IC MPU9250 30 pieces, 
I soldered them into the board,
Оnly 5 chips passed the self test. Mainly gyroscope test is failed, but accel test sometimes is fail too.
Result of test: 
1.
Compass = 0x71   // whoami
Accel test: 2.113, -5.867, 6.430   // x, y, z respectively
Gyro test: 35.538, -1.239, -3.729 // x, y, z respectively
2.
Compass = 0x71
Accel test: 3.196, -1.056, 4.287
Gyro test: 54.333, 7.460, 0.413
3.
Compass = 0x71
Accel test: 10.754, -1.104, 0.790
Gyro test: 46.511, 1.630, -2.275
4.
Compass = 0x71
Accel test: -90.022, -92.128, -90.965
Gyro test: 64.147, 38.558, -11.374
5.
Compass = 0x71
Accel test: 16.004, 52.897, 63.220
Gyro test: 912.197, 17.676, -96.470
6.
Compass = 0x71
Accel test: 9.040, 3.397, -1.560
Gyro test: 85.267, 9.598, -99.190

Soldering temperature did not exceed 270 degrees Celsius.
My init algorithm is:
    resetMPU9250(); // Reset registers to default in preparation for device calibration
    Delay_us(20000);
    
    MPU9250SelfTest(selfTestResult);
    printf("Accel test: %.3f, %.3f, %.3f\r\n", selfTestResult[0], selfTestResult[1], selfTestResult[2]);
    printf("Gyro test: %.3f, %.3f, %.3f\r\n", selfTestResult[3], selfTestResult[4], selfTestResult[5]);
    
    Delay_us(20000);
    calibrateMPU9250(NULL, NULL);
    Delay_us(3000);
    initMPU9250(); 
    Delay_us(20000);
    res = initAK8963(magCalibration);

What is the problem, any considering?

kriswiner, 2021-01-21:
This code is meaningless without the method detail.

I wouldn't worry too much about the self test results since almost no one
implements this properly.

You can compare the method to the documentation
<https://github.com/kriswiner/MPU9250/blob/master/Documents/AN-MPU-9250A-03%20MPU-9250%20Accel%20Gyro%20and%20Compass%20Self-Test%20Implementation%20v1%200_062813.pdf>
.

I would simply calibrate the sensors and see if they return reasonable
values.

iamnot-v, 2021-01-25:
Thank you for your answer.

My selftest function was copyed from this project:
 https://github.com/kriswiner/MPU9250/blob/72038b040bef3cf072612cd8a8ee8f26e3c87158/Dual_MPU9250/MPU9250.cpp#L463

I worry about the self test results because this have direct correlation with bechaviour of MPU9250.
If gyroscope test is fail i have slow change yaw at fast rotate of device. 
if accelerometer test is fail then i have errors yaw at tilt of device.

kriswiner, 2021-01-25:
You linked to the calibration function, which is quite different.

If this is what you are using as your "self test" function, maybe there are
other, um, errors in your code too.

iamnot-v, 2021-01-26:
this is my error.

that is selftest function what i use.
https://github.com/kriswiner/MPU9250/blob/72038b040bef3cf072612cd8a8ee8f26e3c87158/Dual_MPU9250/MPU9250.cpp#L610

kriswiner, 2021-01-26:
But of course, this is wrong, because the self test only uses the LSByte.
Check the documentation.

In any case, if you get crappy results, lots of jitter or poor response it
doesn't matter what the self test says, the chips are bad.

If you buy only high-quality breakouts or genuine MPU9250 not fakes less
chance of these kinds of problems....

iamnot-v, 2021-01-27:
Thank, Kris. 

This is bad ICs,
I think my issue can be close.

closed 2021-01-27 by iamnot-v