kriswiner/MPU9250 #453: Two MPU9250s: read magnetometer

ChiaraBolgiani, 2021-01-22:
I am using two Grove IMU 9DOF ([https://wiki.seeedstudio.com/Grove-IMU_9DOF_v2.0/](url)) and I want to read angles from both of them. I already have one on address 0x69 and the other on the default 0x68 as explained in the Grove IMU. I can read the angles nicely when I use each IMU alone with the basicAHRS code. 
I am trying to use the dualMPU9250_ladybug code but with an Arduino nano. I know the interrupts pin on the nano are different and I changed them to 2 and 3. I still can't read any data from the sensor. I am now wondering if my wiring is correct.

For both sensor I have:

sensor | arduino
3.3V    |  VCC
GND   | GND
SDA    | SDA + 4.7k pullup resistor
SCL     | SCL   + 4.7k pullup resistor

Am I missing something? I also don't have the RTC library and therefore commented everything involving it in my code, but I don't expect it creating some issues, or does it?
Thank you in advance for your help.

kriswiner, 2021-01-22:
Are you sure you have the interrupts right? I would try simply polling the
sensor data (comment out the check for interrupt flag == true) to see if
you can read data.

ChiaraBolgiani, 2021-01-25:
Yes, without the interrupts I can get nice readings. Why are they not working though? I don't necessarily need them but it would be nice to understand what mistake I'm making.
Thanks again for your help.

kriswiner, 2021-01-25:
Why are they not working though?

Probably because you don't have them configured properly.