kriswiner/MPU9250 #457: FIFO_R_W

Beldian, 2021-03-22:
Hey!

When you read from the FIFO_R_W register 12 bites, you actually read EXT_SENS registers? If I understand correcty, there is a connection between them.

kriswiner, 2021-03-22:
not sure what you are asking here?

Beldian, 2021-03-23:
So in MPU9250.h headerfile at line 498 there is this:

"readBytes(MPU9250_ADDRESS, FIFO_R_W, 12, &data[0]);"

What is this? It says you read 12 bites from FIFO_R_W register. I would like to ask, what that makes sense.
I know that, for example when you read 6 bites from ACCEL_XOUT_H, this is auto increment and you'll get all the ACCEL data.
But FIFO_R_W is not auto increment. So I am in doubt, why did you do that. Could you answer?

kriswiner, 2021-03-23:
The FIFO_R_W is the register from which FIFO values are read per the data
sheet/register map.

In a sense, this is auto-incrementing, but better to think of FIFO_R_W as
an array 512 bytes deep that can be filled with data bytes and read
back like an array.

Again, read the data sheet and register map.

What is it that you don't understand?

Beldian, 2021-03-23:
I think I understood that. Thank you so much!