kriswiner/MPU9250 #459: calculation yaw from mpu 9250

Aturgut45, 2021-05-31:
Hi Kris
I have a problem. I have read your all comment. But ı havent solved my problem.
I calibrated my all sensor (Accel, gyro and mag). I get 9 data very well. Although I apply all your formula, I dont have good yaw angle. Couldu you help me please?

kriswiner, 2021-05-31:
How did you check your sensor calibration? And what is your MCU and fusion
rate?

Aturgut45, 2021-05-31:
First of all thank you for your reply me. I follow your study. They are very useful. 
[mpu 9250.txt](https://github.com/kriswiner/MPU9250/files/6571320/mpu.9250.txt)
This is my study.  
I use your mag calibration code. But data is very strange. After, I get mx,my and mz data. I used this data on Magneto 1.2  program. This program helps me to find bias and scale matrix. ı put my code Bias and Scale data to calibrate magnetometer. This job was very good. My mx,my and mz data is very acceptable. for exp: mx:-0,29 my:-0,37 mz:0
I think this value is good.
If you dont have time to examine my code, ı apply this
//apply offsets (bias) and scale factors from Magneto
  tempt[0] = (Mxyz[0] - M_b[0]);
  tempt[1] = (Mxyz[1] - M_b[1]);
  tempt[2] = (Mxyz[2] - M_b[2]);
  Mxyz[0] = M_Ainv[0][0] * tempt[0] + M_Ainv[0][1] * tempt[1] + M_Ainv[0][2] * tempt[2];
  Mxyz[1] = M_Ainv[1][0] * tempt[0] + M_Ainv[1][1] * tempt[1] + M_Ainv[1][2] * tempt[2];
  Mxyz[2] = M_Ainv[2][0] * tempt[0] + M_Ainv[2][1] * tempt[1] + M_Ainv[2][2] * tempt[2];

  mx = Mxyz[0];
  my = Mxyz[1];
  mz = Mxyz[2];

Thank you

Aturgut45, 2021-05-31:
Also ı dont know about MCU and fusion rate. Could you explain me please?
Thank you again.

kriswiner, 2021-05-31:
I can;t debug your code. How did you test the accuracy of your sensor
calibration? What MCU are you using and what fusion rate does it achieve?
What are you using for a sensor fusion algorithm?

kriswiner, 2021-05-31:
What microcontroller are you using to manage the sensors and perform the
sensor fusion?

Aturgut45, 2021-05-31:
Actually I didint control mag calibration. How can I do this section? How can I find fusion rate?
My sensor fusion algoritm is
#ifdef RESTRICT_PITCH 
  roll = atan2(ay, az) * RAD_TO_DEG;
  pitch = atan(-ax / sqrt(ay * ay + az * az)) * RAD_TO_DEG;
#else // Eq. 28 and 29
  roll = atan(ay / sqrt(ax * ax + az * az)) * RAD_TO_DEG;
  pitch = atan2(-ax, az) * RAD_TO_DEG;
#endif

I just use Arduino Mega and MPU 9250.

kriswiner, 2021-05-31:
This is not sensor fusion...no way you are going to get accurate YAW like
this. Maybe try something else
<https://github.com/kriswiner/MPU9250/tree/master/AK8963_as_slave>?

And if you don't calibrate your sensors you will never get good results.

Aturgut45, 2021-05-31:
Sorry Kris. My sensor fusion code is 
  // 3. deneme
  double Mx = mx * cos(pitch) + my * sin(roll) * sin(pitch) + mz * cos(roll) * sin(pitch);
  double My = my * cos(roll) - mz * sin(roll);

  /* ikinci deneme
    double My = mz * sin(roll) - my * cos(roll);
    double Mx = mx * cos(pitch) + my * sin(pitch) * sin(roll) + mz * sin(pitch) * cos(roll);
  */
  /*    //bizim ilk eklediğimiz
    float Mx = mx * cos(pitch) + mz * sin(pitch);
    float My = mx * sin(roll) * sin(pitch) + my * cos(roll) - mz * sin(roll) * cos(pitch);*/

double yaw = atan2(-My, Mx) * RAD_TO_DEG;

There are three different ways. But they are not working.
Could you check my code plz?

Aturgut45, 2021-05-31:
ı think you offer me to use quaternion code. Actually ı have tried this. But Q values are always zero. I have a big problem

kriswiner, 2021-05-31:
your fusion code will never work well.

Aturgut45, 2021-06-01:
Again Hi Kris,
Can you meet on Skype, If you have short time? I would be very pleased. I realy need your help. Thank you
My Skype ID is aykutturgut45@gmail.com

kriswiner, 2021-06-01:
Sorry no.