kriswiner/MPU9250 #460: Does it work in free fall?

Sofus199, 2021-06-05:
Does your code work in free fall or with acceleration like on a rocket? I can see in your code that you estimate gravity to get orientation, is that right? Because when it is in free air you can not detect gravity, so will the magnetometer just take over or will it make crazy yaw, pitch and roll values?

Thanks for your attention. I'm looking forward for your reply.

kriswiner, 2021-06-05:
The 9 DoF sensor certainly works in high G and free fall environments, but
the AHRS solution will likely be inaccurate when the inertial acceleration
is on the order of gravity or higher. The magnetometer calibration might
also be affected when far from the Earth's surface. I just don't
know, never having tested absolute orientation estimation sensors in this
environment. Best and cheapest way is simply to test it.

Sofus199, 2021-06-05:
Thanks for quick reply!
Do you know if it works with any airborne application like quadcopters, airplanes, etc...
Without having tried, do you think it is possible to get orientation using only the gyroscope and the magnetomter?

kriswiner, 2021-06-05:
Works great with UAVs, we use this solution all the time for headless
flight.

It works and is being used in airplanes but as I mentioned, on hard banks
where the g forces approach gravity the solution suffers. This is a
long-standing problem in need of a solution. We have some ideas but this is
R&D at the moment.

Roll and pitch can be estimated pretty well with just an accel. For
accurate heading, you need all three...or 9 DoF.

You might want to put a baro on the board as well as we usually do, so 10
DoF since modern baros estimate altitude within <10 cm

Sofus199, 2021-06-05:
Thanks again!
I'll try it out. Learn by doing.

closed 2021-06-05 by Sofus199