kriswiner/MPU9250 #461: MPU9250 Question

AbhishekJoshi, 2021-06-07:
@kriswiner 

Quick question for you, as the datasheet is not clear, and I couldn't find obvious hints.

If I powercycle an MPU9250, will the offset/registers be set to "default"/uncalibrated values?

Do I need to recalibrate my MPU everytime I turn it on?

kriswiner, 2021-06-07:
The memory in the MPU9250 is volatile, so it will return to the default
contents (i.e., 0) on power on. You can store your calibration in arrays in
your program or on-board flash, etc.

AbhishekJoshi, 2021-06-07:
Hi. Thanks for your response.

From your experience, how often are these MPU's dead from factory on arrival?

https://www.banggood.com/5Pcs-MPU-9250-GY-9250-9-Axis-Sensor-Module-I2C-SPI-Communication-Board-For-Arduino-p-1233671.html

kriswiner, 2021-06-07:
Depends where you buy from. Reputable sources, never, from China about half
the time...

kriswiner, 2021-06-07:
nCS needs to be pulled HIGH to work IIRC...

AbhishekJoshi, 2021-06-07:
Last one! Do you have any reputable sources you can recommend?

Thanks!

kriswiner, 2021-06-07:
Maybe look at Sparkfun...I also have a few
<https://www.tindie.com/products/onehorse/wearable-mpu9250-nano-board/>.